CANopenNode/CANopenNode #131: Endianess problem for helper functions

blankm, 2019-12-06:
Related #130 
Don't the following functions also need to be adjusted according to the endianess?

uint16_t CO_getUint16(const uint8_t data[]);
uint32_t CO_getUint32(const uint8_t data[]);
void CO_setUint16(uint8_t data[], const uint16_t value);
void CO_setUint32(uint8_t data[], const uint32_t value);

Since through the union access, like here
```
    CO_bytes_t b;
    b.u8[0] = data[0];
    b.u8[1] = data[1];
    return b.u16[0];
```
Endianess should be considered?

martinwag, 2019-12-09:
Those functions are for reading/writing OD variables in user code. OD variables should already have host enianess.

blankm, 2019-12-09:
Ah yeah, my mistake.

closed 2019-12-09 by blankm