CANopenNode/CANopenNode #136: uintptr_t and void pointer for the CAN device address

JuPrgn, 2019-12-09:
The PR #116  introduced : 

- use of uintptr_t : this type is undeclared for XC16 compiler

> ../CANopenNode/stack/PIC24_dsPIC33/CO_driver.c:125:28: error: 'uintptr_t' undeclared (first use in this function)

I could not find this declaration on XC16 `stdint.h`

- using a void * as CAN device address generate the following warnings : 

> warning: comparison between pointer and integer

`if(CANdriverState == ADDR_CAN1)`

> warning: initialization makes integer from pointer without a cast

`uint16_t addr = CANmodule->CANdriverState;`

Do I have to cast `CANdriverState` everywhere as `*((uint16_t*)CANdriverState)` ?

@odesenfans

JuPrgn, 2019-12-20:
Does anyone have a suggestion about this ?

odesenfans, 2019-12-21:
Hi,

The driver might need an adaptation indeed. I wasn’t able to test for this platform, my bad.

1. ˋtypedef uint16_t uintptr_t` in the driver header file.
2. The best way to use the ˋvoid *` argument is to define a struct holding all the driver specific parameters (including the base address) and then cast your pointer back to a pointer to this struct. Then dereference that pointer where you need it!

Does this help?

JuPrgn, 2020-01-09:
Thank you for your answer ! Sorry for the delay... 
Ok I will give it a try. I probably missed something from your PR you should have applied this to the driver you are using right ?

CANopenNode, 2020-01-27:
After rearrangement of the driver and update of the CANopenPIC, this is solved.

closed 2020-01-27 by CANopenNode