CANopenNode/CANopenNode #139: PDO not sent / received

qworkz11, 2019-12-12:
Hi,
I have a CANopen device which is transmitting a PDO with a single mapped value (transmission type 255):
```
can1  283   [4]  0A 3B 00 00
can1  283   [4]  FD 3C 00 00
```
My Linux CANopenNode node (using the socketCAN driver from the stack) has the corresponding RPDO with COB-ID 283, transmission type 255 and a mapped object which matches the one sent by my device (called "Position Demand Value", INTEGER32, rw access, memory type RAM, PDO COS set). `CO_OD_RAM.positionDemandValue`, however, does not contain any value, although I can see the values being sent.

Similarly for TPDOs: Writing to a `COB_OD_RAM` value that is mapped to a TPDO does not result in a TPDO being sent (observed via candump). 

Any idea why?

`CANrx_taskTmr_init()` is called once, and a realtime thread calls `CANrx_taskTmr_process()` every ~1ms and updates the timer. ROM & EEPROM are initialized as shown in CANopenSocket (although these are unnecessary for RAM I guess?). I don't (yet) have a mainline thread, but as far as I understand it I don't need one for PDOs.

Any help would be great!
Thanks

martinwag, 2019-12-13:
Take a look at CANopenSocket project (Linux). PDOs should work on their own, I think you mostly have missed to initialize someting.

JuPrgn, 2019-12-13:
Hi, 

I don't know if you are familiar with CANopen, if this is not the case, this may help you : 

PDO are processed only on NMT Operational mode (Tx and Rx devices should both be in Operational mode).

Can you show your RPDO mapping parameter ? 

- You have to set RPDO communication parameter for example 0x1400 (subindex  1 : COBID (0x283), subindex 2 : Transmission type (255))
- And you have to set RPDO mapping parameter for example 0x1600 (subindex 1 : 0x64010108) this one say to your device that 8 bits from PDO(0x283) will be copied to your object 0x6401 at subindex 01

qworkz11, 2019-12-14:
My mapping & communication parameters are as follows:

Transmitter:

- 0x60620020 (Object 0x6062, sub 0x00, i32) is mapped to the first 4 bytes of TPDO 0x1A01 (sub 0x00 is set to 1)
- Communication parameters are 0x283 COB-ID, 0xFF transmission type, and 1s inhibit time

Receiver (CANopenNode):

- 0x24210020 (Object 0x2420, sub 0x00, i32, called "Position Demand Value") is mapped to the first 4 Bytes of RPDO 0x1410
- Communication parameters are 0x283 COB-ID, 0xFF transmission type

> PDO are processed only on NMT Operational mode (Tx and Rx devices should both be in Operational mode).

This could be the issue, `CO_NMT_getInternalState()` returns 0, which means initializing I guess... How do I set my node to operational, and why do I even need to when I implement a master node? `CO_NO_NMT_MASTER` is set to 1 and I use my node to send NMT commands as well... 

> PDOs should work on their own, I think you most likely have missed to initialize someting.

`CO_init()`, `CANrx_taskTmr_init()`, `CO_CANsetNormalMode()` are called after the CAN device is set up, and `CANrx_taskTmr_process()`is called periodically (SDOs are working as well). Any idea what I could have missed?

Thanks for your help!

qworkz11, 2019-12-14:
Ok so I got it to work by simply setting `CO->NMT->operatingState = CO_NMT_OPERATIONAL`. Is there a more elegant way?

JuPrgn, 2019-12-14:
Yes your master should initialize slaves to set their operating mode. There is also a default value configured by your OD to put slaves in pre-op or op at start-up (if I remember well this is done with 1F80). I prefer initialize slaves as pre-op and use master to switch to operational after configuration.

qworkz11, 2019-12-14:
Alright, thanks a lot. I set 0x1F8 (NMT startup) to 0x5 to transition to operational automatically. I don't have a HB consumer yet, so I guess my node does not know when the slave has finished booting which means it doesn't do the transition. I'll use my hack for now and switch to the proper solution later.
Thanks a lot for your help!

closed 2019-12-14 by qworkz11

JuPrgn, 2019-12-14:
You are welcome ! =)