CANopenNode/CANopenNode #140: OD_errorStatusBits mandatory

ArminBmrt, 2019-12-13:
I have CANopenNode working using SDO. It is able to communicate with an epos2 servo controller. I generate the CO_OD.c/h files using libedssharp.

Almost everything works without changing the source files except for the  OD_errorStatusBits. The epos2 does not have such an object but the CANopenNode code depends on it. So I created the variable in CO_OS.c using:
`
OCTET_STRING OD_errorStatusBits[ODL_errorStatusBits_stringLength] = {0};
`
For SDO this is all fine but now I want to use PDO and I need to go to operational state. This fails because some of the OD_errorStatusBits are set

CO_Emergency.c line 275 in CO_EM_process()
`
 if(em->errorStatusBits[2] || em->errorStatusBits[3]){
        errorRegister |= CO_ERR_REG_COMM_ERR;
 }
`
How do I get to an enabled state? Can I disable this check without changing the CANopenNode source code?

Should this variable be mandatory? In the example IO.eds file it is mapped to 0x2100 which is a manufacturer specific object

martinwag, 2019-12-16:
This is indeed missing when creating a new device with EDS editor.
Just add a a manufacturer object "Error status bits" on address 0x2100 with:
type octet string
size >= 10 bytes
read only
ram type
pdo optional
with the EDS editor.

This element is needed by the stack for storing errors. You can't remove it.

ArminBmrt, 2019-12-18:
I have added the 'Error status bits' with the eds editor. Now I can compile without changing any of the source files, thanks!

Unfortunatly I am still getting the same error. Do you have any idea how this can happen? I have tested this with and without can device connected to the bus.

I want to put this node where CANopenNode is running in operation mode so I can use PDO's.
I want to go to operation using:

`CO_sendNMTcommand(CO, CO_NMT_ENTER_OPERATIONAL, ownNodeId); // Put ourselfs in operation mode` 

Before I go to operation I have started the CANopenNode using:

	const uint8_t ownNodeId = 1;
	void StartCanTask(void const * argument) {
		/* USER CODE BEGIN StartCanTask */
		while (!isNetworkReady) {
			osDelay(1);
		}

		//CO_errorReset();
		for (int i = 0; i < ODL_errorStatusBits_stringLength; i++) {
			OD_errorStatusBits[i] = 0;
		}
		// CAN module address, NodeID, Bitrate
		// We do not use CAN registers directly, so address here is a pointer to the CAN_HandleTypeDef object.
		volatile CO_ReturnError_t err = CO_init((uint32_t) &hcan1, ownNodeId, 1000);

		//TODO behavior in a case of the stack error. Currently not defined.
		while (err != CO_ERROR_NO) {
			Led3.on();
			osDelay(1000);
			err = CO_init((uint32_t) &hcan1, ownNodeId, 1000);
		}

		// start CAN
		CO_CANsetNormalMode(CO->CANmodule[0]);

		CO_NMT_reset_cmd_t reset = CO_RESET_NOT;
		isCanReady = true;

		for (;;) {
			// CANopen process
			reset = CO_process(CO, 1, NULL);

			if (CO->CANmodule[0]->CANnormal) {
				bool_t syncWas;

				// Process Sync and read inputs
				syncWas = CO_process_SYNC_RPDO(CO, 1000);

				// Write outputs
				CO_process_TPDO(CO, syncWas, 1000);

				CO_CANpolling_Tx(CO->CANmodule[0]);
			}

			osDelay(1);
		}
		/* USER CODE END StartCanTask */
	}

martinwag, 2019-12-20:
If you use hardware CAN bus, you need at least one other device with active CAN on that. This can be anything that does "ACK" when you send a message (see [https://www.vector.com/de/en/news/news/want-to-get-our-most-coveted-poster-for-free-cancan-fd-reference-chart/](url))

You can find the list of error bits inside CO_emergency.h, table CO_EM_errorStatusBits.

ArminBmrt, 2019-12-20:
I have 3 epos2 drives and 1 WDGA absolute encoder on the bus.

I have moved the CO_sendNMTcommand() to a bit later. I first check if two of the epos2 devices are online by reading the devicetype using SDO. Then I try to put my own device to operational.

When the CO_init() function returns the OD_errorStatusBits[3] is set to 0x04.

I have a log from candump. This is after CO_sendNMTcommand() returns.

	$ candump -ta -d can0
	(1576835802.675966)  can0  701   [1]  00
	(1576835802.677003)  can0  602   [8]  40 00 10 00 00 00 00 00
	(1576835802.677082)  can0  582   [8]  43 00 10 00 92 01 02 00
	(1576835802.684831)  can0  081   [8]  00 82 10 1A 40 00 C1 20
	(1576835802.686828)  can0  603   [8]  40 00 10 00 00 00 00 00
	(1576835802.687021)  can0  583   [8]  43 00 10 00 92 01 02 00
	(1576835802.696760)  can0  000   [2]  01 01

[image]

CANopenNode, 2020-01-27:
This is old, bu anyway:

You received an emergency object: `can0  081   [8]  00 82 10 1A 40 00 C1 20`
See example for emergency object here: https://github.com/CANopenNode/CANopenSocket

In your case fourth bit is 1A, from CO_emergency.h, CO_EM_errorStatusBits you see, you have Error with PDO mapping.
The same you can see from OD_errorStatusBits. You have OD_errorStatusBits[3] set to 0x04, if you count the bits, you get 00000100 00000000 00000000 00000000, this is 0x1A-th bit.

closed 2020-01-28 by CANopenNode

fenugrec, 2020-12-27:
> Just add a a manufacturer object "Error status bits" on address 0x2100 with:
> type octet string
> size >= 10 bytes

I have the same problem and cannot get libedssharp to generate the correct Errorstatusbits stuff. 
How did you do this ? Is the object type VAR or ARRAY ? If it's VAR and  datatype =OCTET_STRING, how do you specify size ?? 

I tried ARRAY of OCTET_STRING with 10 subindices but that's clearly not the right thing.

JuPrgn, 2020-12-27:
This object should be an octet_string and the size is set with default value "00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00"

JuPrgn, 2020-12-27:
One more thought about this I don't remember well why but the size should be 16 bytes I had issue using 10 bytes some times ago. I could look further to this if you need some more informations

fenugrec, 2020-12-27:
Aah, default value, that's the trick. Thanks. I'll have it 16 bytes long for now, I'm still just trying to get the whole thing to compile, working through errors one at a time...

JuPrgn, 2020-12-28:
After some checks on my OD errorStatusBits should be 32 bytes long : each bit correspond to an EMCY code 8 * 32 = 256 errors.

- Object Type : `VAR`
- Data Type : `OCTET_STRING`
- Default value : `00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

CANopenNode, 2020-12-28:
This nonstandard object is optional in newOD branch. For old OD this object can be inserted from `example/_project.xml` by latest libedssharp. See also https://github.com/CANopenNode/CANopenNode/issues/247#issuecomment-751703054

fenugrec, 2020-12-30:
Thanks. I noticed latest libedssharp ' 301-profile.xpd does not define object 1029 the same way as `example/_project.xml` and cannot work with current master.