CANopenNode/CANopenNode #141: [Question] What is CO_Trace?

rubenverhoef, 2019-12-13:
I'm seeing CO_Trace and i understand most of it what it does but how to display the data.
Do you guys have an example program or something to display the Trace data?

wilkinsw, 2019-12-20:
Here's the description from the docstring:
https://github.com/CANopenNode/CANopenNode/blob/e1244868db565b6df7f93a6731b17985ea44a26d/stack/CO_trace.h#L59-L79

I have not used it personally, though I might try it out at some point. Perhaps @CANopenNode can share what he used it for originally.

CANopenNode, 2020-01-15:
CO_trace is quite useful and functional, i have some GUI with CANopenSocket and Apache web server + SVG, but is is so tangled now, I can't share it. I'm organizing the things now, step by step.

There are two approaches:
- Use CO_trace on remote CANopen node (on fast microcontroller). Read SVG string via SDO. Records with very accurate time can be produced.
- Use CO_trace on CANopenSocket (rich Linux computer). Collect PDOs from multiple remote CAN(open) nodes with CO_trace. Because CANopenSocket uses SDO communication with internal Object Dictionary directly, it does not occupy CAN bus. So large amounts of data can be transferred.

I can't offer more support currently.

CANopenNode, 2020-01-20:
Some more description is now here:
https://github.com/CANopenNode/CANopenPIC

rubenverhoef, 2020-01-23:
Thanks for all the feedback.
At [CANopenPIC#monitor-variables-using-built-in-trace-functionality](https://github.com/CANopenNode/CANopenPIC#monitor-variables-using-built-in-trace-functionality) it is saved as csv. I don't really understand how to display the data.
Do you maybe have an example dump (and screenshot) of data so i can play around with visualizing the data?

CANopenNode, 2020-01-23:
`plot1.csv` from the example is basically a text file, with values separated by commas.

"A CSV file is a comma separated values file commonly used by spreadsheet programs such as Microsoft Excel or OpenOffice Calc."
https://fileinfo.com/extension/csv

rubenverhoef, 2020-01-23:
I know what an csv file is but i don't know how it is related to the SVG stream. SVG is a different type of file format.

CANopenNode, 2020-01-23:
I don't understand the question.
After trace is configured, it basically prints a variable with timestamp into internal buffer. Currently it is able to print into three different formats: csv or svg or binary, depending on the configuration. (see functions `printPointCsv()`, `printPointSvg()` and other similar). csv and svg are not related, they are just two options, whichever you prefer.

Populated internal buffer can then be read via SDO. You get one of:
- csv text string with time:value pairs
- svg text, which represents a svg path. It can be directly put into a svg file, to draw a graph for example.
- raw binary data.

closed 2020-01-28 by CANopenNode