CANopenNode/CANopenNode #146: TPDO not transmitted

qworkz11, 2020-01-07:
Hi,
what is the correct tx-type setting for a TPDO when it is supposed to be sent when the objects mapped into it change? I thought it was 255, I have the following issue though: I have an object `controlword` which is mapped into a TPDO with said setting, but `CO_OD_RAM.controlword = some_uint16t` does not result in a PDO being sent.
When an event timer is set, though, the TPDO works just fine:
```
can1  303   [3]  0F 00 00
can1  303   [3]  0F 00 00
can1  303   [3]  5F 00 01
can1  303   [3]  5F 00 01
```

Receiving RPODs works as well.

CaCtus491, 2020-01-07:
Are you using the EDS Editor in libedssharp? If so, you'll need to have the 'Change Of State' checkbox enabled for the relevant object dictionary entries under the Object Dictionary tab.

[image]

qworkz11, 2020-01-08:
Yep, that was the issue. Thanks!

closed 2020-01-08 by qworkz11