CANopenNode/CANopenNode #147: Change the license to Apache 2.0

CANopenNode, 2020-01-08:
According to discussion in #105, we decided to change the license of all code to Apache 2.0.

Before that, the license was GPL with linking exception.

Apache license is widely used. It is permissive, free software license. It is easier to use in commercial projects. See [wiki](https://en.wikipedia.org/wiki/Apache_License) for more information.

henrikbrixandersen, 2020-01-08:
@CANopenNode That's great, thank you. I think you may have missed the `CANopen.h` one location in `CANopen.h` (the doxygen intro text).

CANopenNode, 2020-01-08:
I changed the license of the files I own in this project (CANopenNode). Also updated README and forgotten CANopen.h.
I also plan to change to the Apache license in other two my projects CANopenPIC and CANopenSocket in next few days. If there are any pro or cons, please let me know.

In this project, there are some files (which I don't own) still with the old license:
1. Third-parity or old driver files: It is a plan to split those files from the main project. Author of specific driver is responsible for the license. See also #108.
2. CO_time.h/.c: @JuPrgn, you are the author. Can you change the license of your files to Apache? Or give me a permission? You also have to specify the copyright holder.
3. CO_LSS* and neuberger-socketCAN: @martinwag, will you please update your files.

JuPrgn, 2020-01-08:
@CANopenNode you have my permission please do as if this was a part of your stack/work. I did not know what to put in copyright I am not very familiar with this...

martinwag, 2020-01-14:
I've changed our part in 5ff34938597593ed5c17411b62a2ee65ed19ae4c

CANopenNode, 2020-01-15:
License for CANopenNode is now completely changed. Some files in stack/*/ directories still has the old license, but they are not relevant to the stack and they will be removed in the future. See issue #108.

License of CANopenSocket project is also changed to Apache 2.0.

I will change the license of CANopenPIC project to Apache 2.0 in near future.

closed 2020-01-15 by CANopenNode

Reissner, 2020-02-12:
Reopened discussion? 
I heared, that changing GPL like licenses requires consent of all contributors, 
not only the current ones.