CANopenNode/CANopenNode #148: Thoughts on TIME_OF_DAY

gotocoffee1, 2020-01-10:
Hello,
I want to discuss the TIME_OF_DAY struct, which is declared in the generated OD.h file.
The usage feels a bit awkward.  The problem is that `sizeof (TIME_OF_DAY)`  is 8 in most cases or at least platform specific. But the standard requires it to be 6 bytes. The TIME stack uses `#define TIME_MSG_LENGTH 6U` to solve that. But when you want a custom array of TIME_OF_DAY you are doomed, since the memory size and the expected size are diffrent.  You have to write an SDO callback for each of these objects to fix the offsets.
Futher I think the endianess support is broken too, since you cannot simply swap all bytes in the struct. I am not even sure how to correctly swap an 28 bit value alone yet in a struct.

I think the most stable and portable solution would be to define the type as an 6 byte char array and generate some getter and setter to access the ms and days. And let them also do the swap magic.

What do you guys think?

JuPrgn, 2020-01-16:
This is a good idea ! Current implementation is ok for TIME and that was enough for me but sure we can improve this. Getter and setter are ok for me. 
I am curious what are you doing with a TIME_OF_DAY array ? On my application I am using TIME_OF_DAY for logging but they are always associated with data so I have a DOMAIN callback.

gotocoffee1, 2020-01-16:
Thanks for the feedback. I am tracking life/maintenance time of some hardware components. Maybe a uint32_t array would have done the job too. But I wanted to use the standard datatype for time.
Independent from my design choice we should fix that.
I will wait a bit more before I implement a pull request, to make sure we have a solid solution.

CaCtus491, 2020-05-03:
FYI, we have a similar issue trying to use arrays of INTEGER24 and UNSINGED24.

CANopenNode, 2020-11-05:
TIME_OF_DAY, INTEGER24, UNSINGED56 are very nonstandard and I have no idea, how to handle them correctly. In latest CANopenNode they are just translated to uint8_t array with correct length.

I'm working on libedssharp now and there is also the problem, because there is no standard datatype for them in xdd file. So I decided to abandon those data types and translate data automatically to OCTET_STRING (BITSTRING in xdd file).

gotocoffee1, 2020-11-26:
This is probably the best solution without losing your mind. How about the endianess should those objects have the `CO_ODA_MB_VALUE` flag?

CANopenNode, 2020-11-26:
TIME_OF_DAY is exchanged in direct, 6bytes long, CAN message. It is not used in object dictionary, so I just didn't include it in libedssharp.

INTEGER24 and similar has no practical use in C program. Those types may be usable in PDO mappings. But CANopenNode enables mapping of only 3 bytes of int32_t variable. So those types are also not included in new, not-yet-pulled, libedssharp. Existing "exotic" types are treated as array of uint8_t, which is little-endian anyway and CO_ODA_MB_VALUE is not relevant. That is my opinion.

gotocoffee1, 2020-11-26:
Sounds great. I will close the issue then.

closed 2020-11-26 by gotocoffee1