CANopenNode/CANopenNode #149: Do not use NMTstartup in slave mode, make it optional

lukegluke, 2020-01-14:
[Here ](https://github.com/CANopenNode/CANopenNode/blob/3d41f20adb2a63737fb7f26cd33a8ae37158b98f/stack/CO_NMT_Heartbeat.c#L224) second bit of NMTstartup (0x1F80 OD element) is used to go into operational state if it zero in all cases.

But according to CANOpen CiA302_v3.0 if device is NOT the NMT Master (0 bit is not set) all other bits have to be ignored.

As for me there is no need in NMTstartup argument of CO_NMT_process at all, but I don't use master, so could someone please take a look how to fix it correctly and do not break master logic.

Also I think 0x1F80 can be optional for slave. So if NMTstartup argument stays for further master functionality, #ifdef OD_NMTStartup really should be added.

CANopenNode, 2020-01-15:
I verified the standard, we have inconsistency, but also:
- bit0=0 - "CANopen device is not NMT master. The entries of the object 1F81 h shall be ignored. All other bits of object 1F80h shall be ignored with the **exceptions** defined in Table 16."
- bit2=0 - "Shall switch into NMT state Operational in the process NMT startup"
- bit3=1 - "The NMT master shall not start the NMT slaves and the application may start the NMT slaves."
- table16 - exception1, value =  00000000 00000000 00000000 00001000b - "NMT slave that shall enter the NMT state Operational after the NMT state Initialisation autonomously (self starting)"

(DSP302, part2, V4.1.0, 2009)

So, I suggest to fix from:
https://github.com/CANopenNode/CANopenNode/blob/3d41f20adb2a63737fb7f26cd33a8ae37158b98f/stack/CO_NMT_Heartbeat.c#L224

into:
```
            if (NMTstartup == 0x00000008U) NMT->operatingState = CO_NMT_OPERATIONAL;
```

lukegluke, 2020-01-15:
Ok, but I have only DSP302 V3.2.1 09.04.2003 from here http://overpof.free.fr/schneider/CAN%20&%20CANopen/CANopen/%A9CiA%20CANCANopen%20CD%20V5.1/html/spec3.htm, could you show link where to find DSP302, part2 (?), V4.1.0, 2009.

Your suggestion only about slave, maybe it would be better this?
`if ((NMTstartup == 0x00000008U) || (NMTstartup & 0x05)) NMT->operatingState = CO_NMT_OPERATIONAL;`
for future use for master? I will correct my pull request after your answer.

CANopenNode, 2020-01-15:
It is here: https://www.can-cia.org/groups/specifications/
But I'm not sure, if it is free to download.

I think, my suggestion complies with the standard.

lukegluke, 2020-01-15:
Ok, I make code by your suggestion. But what about possible CO_NMT_OPERATIONAL for NMT master?
Or in future it will be complex logic, with turning on slaves also using 1F81h and so on,  so no need to check `NMTstartup & 0x05` now, am I right?

CANopenNode, 2020-01-16:
Probably the newer standard, which I mentioned above, just clarifies NMT automatic startup.

NMT master functionality is quite advanced feature and could be added in the future. If device is NMT master, then it takes care for starting the nodes, including itself.

If it is not the NMT master, then only relevant is object 1F80, which allows two exceptions for slaves (according to the latest standard):
- Object 0x1F80, value = **0x8** - "NMT slave shall enter the NMT state Operational after the NMT state Initialization autonomously (self starting)"
- Object 0x1F80, value = **0x2** - "NMT slave shall execute the NMT service start remote node with node-ID set to 0"

The second option can be implemented by custom application.

---
Object 1F80 is not part of CiA301 standard, which is public available. I will add some description in README.
This change will also break behavior in some devices, which has object 0x1F80 set to 0.