CANopenNode/CANopenNode #154: Inconsistent time bases used in CO_process functions

heliochronix, 2020-01-17:
There is an inconsistency in the time bases used by functions and objects within this project.

For all objects processed by `CO_process()`, the time base is milliseconds. However, all objects processed by `CO_proces_SYNC()` and `CO_process_TPDO()` are in microseconds.

There is discussion on whether the whole project should move to a consistent time base. Microseconds might be preferred, but there is a concern about if the uint32_t values used to represent microseconds might roll over.

If I'm not mistaken, stdint.h definitions for uint32_t should force even 8 or 16 bit MCUs to actually employ a full unsigned 32 bit value for a variable. This means we have 4294967295 microseconds we can track, or about 71.58278825 minutes.

Since this protocol stack is based on time deltas, with a uint32_t microsecond time base the maximum time difference a system could check would be about 71.58278825 minutes. I suspect this might be long enough for any objects in the protocol stack, especially since they are based on deltas rather than an absolute time.

What are everyone's thoughts on moving to a consistent microsecond time base across the project?

martinwag, 2020-01-17:
I'm in favour for microsecond resolution, even if I don't need it. I'm only using 32bit and 64bit systems, so I don't expect any proglems for me.

The thing with 8/16 bit is not only that you have to check all those drivers for not using "int" type (I don't have any idea about the quality those...). You also have to think about atomic access to uint32_t type timers inside the stack.

CANopenNode, 2020-01-17:
I agree with microseconds. No critical code is executed inside CO_process() and 32 bit should not be a problem.
Just one note: I'm working on driver now. CO_driver.h will be part of the stack, CO_driver_target.h and CO_driver.c will be part of the example. All files in the stack directory will be independent from CANopen.h. There are many different tastes for CANopen.h and .c files, so these two are meant more like a part of the example, same is Makefile.

heliochronix, 2020-01-17:
It is worth noting that because the stack is moving in a direction that leaves the driver portion up to developers, perhaps the concern over 8-bit and 16-bit MCU atomic operations should also be left up to the driver developer. If the goal of this project is to not dictate how a driver is implemented, but only the interface to the driver needed by the stack, then it seems like maybe mentioning the requirement for atomic operations in the example driver code (as part of the documentation and example) is all that is needed. This leaves it up to the developer utilizing the stack to ensure the driver does what is expected.

heliochronix, 2020-02-13:
I noticed the time bases have been updated as of ae48e57c3339f1d0902b50ac8b5da159effc2dc0. Is there anything left to do on this issue or is it safe to close?

CANopenNode, 2020-02-14:
I will make some tests for this.

CANopenNode, 2020-04-29:
Time base is now microsecond in all objects. It seems, it works OK.

closed 2020-04-29 by CANopenNode