CANopenNode/CANopenNode #156: EMCY consumer

qworkz11, 2020-01-18:
Hi,

how do I subscribe to EMCY messages (as a master node)? I tried using an RPDO that corresponds to the EMCY COB-ID (0x83 in my case) and mapping two variables into it (one for the error code and one for the register) - that did not work though. The only thing I can find in the stack is `OD_H1028_EMCY_CONSUMER        = 0x1028U,/**< Emergency consumer object */` in CO_SDO.h, but I am not quite sure what to do with it... I guess the emergency consumer object contains all subscribed EMCY COB_IDs? How can I access the error code and register, though?

Tanks a lot in advance!

martinwag, 2020-01-20:
Hi,

Use CO_EM_initCallbackRx() function in CO_emergency. Currently, you cannot register to specific emergency.

qworkz11, 2020-01-21:
Thanks, works perfectly.

closed 2020-01-21 by qworkz11