CANopenNode/CANopenNode #160: Linking Exception

Reissner, 2020-01-28:
Whereas the file LICENSE just contains GPL, in various haeders like CANopen.h, 
a linking exception is included. 
I think, one shall make it consistent. 
I am not an expert in legal issues, but if possible, add the linking exception also to the LICENSE file. 
Maybe one has to do research on history of CANopenNode to find out whether this is ok, 
because one may not lessen a license. 

This issue is quite urgent, as it decides whether this software is useable for me.

Reissner, 2020-01-28:
I did research. Original license was LGPL which includes linking exception.

CANopenNode, 2020-01-28:
Actually license changed to permissive Apache 20 days ago.
This is easier for commercial projects than GPL with linking exception.

There is still file LICENSE.old and some directories under the "stack" directory, which are unchanged. This will be removed soon anyway.

Reissner, 2020-02-11:
is it really that easy, making a license more admissive?? 
I researched and found out no. 

But the older sources and also readme showed it was LGPL one day. 
Why is it easier for commercial projects to have apache license? 

All in all these quick changes made things quite inconsistent.

CANopenNode, 2020-02-12:
I'm not an expert, but [wikipedia](https://en.wikipedia.org/wiki/Free_software_license) distinguishes:
- Permissive license - Apache, MIT: Grants use rights, including right to relicense (allows proprietization, license compatibility)
- Copyleft (protective license) - GPL: Grants use rights, forbids proprietization

---
People had problems with GPL, also LGPL, so we added linking exception. And then this was incompatible with Zephyr RTOS, so we simply changed to Apache. I think, that's it now. See also #147.

Reissner, 2020-02-12:
OK, Just one point: Change of license requires consent of all contributors. 
Not of the current only.

CANopenNode, 2020-02-12:
Or only those, who have copyright?

martinwag, 2020-02-12:
I've mentioned this here before... (https://github.com/CANopenNode/CANopenNode/issues/105#issuecomment-561117683). That's why I only changed licence on files I own.

You have copyright on **everything** you created, except for "obvious" stuff. A good example for this problem is the discussion on who has copyright for snippets on stackoverflow.
So, strictly speaking, you need written permission of everyone who ever contributed a few lines to chane license. Only exceptions are
- you remove the parts that lack permission (not practical)
- licence change is covered by current license (e.g. GPL2 -> GPL3) (doesn't apply in our case)
- you ask every contributor for written permission before accepting any contribution (already too late)

CANopenNode, 2020-02-12:
It seems, a wasn't very pedantic, when I changed the license. But it is hard to find a contact to all, who ever contributed to the project.

Anyway, most of the contributions are mine and in the last time from martinwag. Files for all microcontrollers (subdirectories of stack) are removed anyway (see branch split-driver and #108).

All other contributions are various fixes. I'm not sure, how is with that. If someone found a bug and wrote an issue, is he a contributor? Are contributors only those with pull requests? If one small contributor will complain about the Apache license, may I remove his contribution and fix the bug myself?

heliochronix, 2020-02-12:
The legal world is so convoluted at times... Sorry you have to figure all this out, I'm sure it's a headache.

I do agree with the path you are taking on licensing. If there's anything I need to do to ensure my small contributions are properly licensed, let me know. If it's simply giving approval, you have it.

CANopenNode, 2020-02-13:
I don't know all the details of these licensing issues and I also can't find all answers to my questions. According to my experience, important to users (including myself) is freedom of usage and so I'm happy with Apache license. I don't expect problems, since there is agreement for all major parts of the project. If someone won't agree, I'm sure, we will find a solution.

jmakip, 2020-03-31:
All the users who have contributed into GPL licenced project own copyright to their submissions. Therefore changing licence can be tricky. Proper thing to do is to contact all the authors are they willing to relicence their submission into Apache licence. If not their commits are still under GPL. 

Then there is option to drop their submission out of relicenced version. At least I have heard some other projects have done it. (not sure if its legal or not) Or write the whole thing from scratch.

Git log shows that project only has 12 contributors and most of them have submitted only few bug fixes, so sending emails and asking nicely is still good option.

CANopenNode, 2020-06-09:
Project has a long history from file releases on Sourceforge, to SVN and git. There were quite a lot of contributors and it is impossible to find them all. Probably i violated legal rules multiple time, without knowing it. Project started with GPL, then changed to LGPL, then GPL with linking exception, now Apache licence. Majority of the code was mine, so I didn't worry much. I also didn't hear opinion, opposing to my efforts, to make licenses more permissive.

I promise, I will be more serious about licensing from now on. Anyway, I hope, Apache license is OK now and we won't have any more licensing issues.

I think, CANopenNode can not have potential legal problems, because of my naive licensing changes in the past. Recent key contributors agreed with license change. If some unknown contributor, who disagree with Apache license, will require removal of his part, there can not be much damage to the project.

closed 2020-06-09 by CANopenNode