CANopenNode/CANopenNode #162: New Object Dictionary

CANopenNode, 2020-01-28:
Object dictionary is the oldest part of this project. It is time, to make some improvements. I was thinking about multiple different principles.

I think, that general principle of the current implementation is quite good. But implementation could be much better.

Here are my ideas:
- GUI tool [libedssharp](https://github.com/robincornelius/libedssharp) will be used for generating CO_OD files. Object dictionary can be generated directly from standard xdd (or EDS) file, no specific options or special project file needed. (I hope so).
- Remove RAM, ROM, EEPROM. Objects are initialized globally (as now), but each index separately. No need for defines in CO_OD.h. Storing of the variables should be handled by application.
- Structure definitions as specified in CANopen, for example, OD entry 0x0020 - DEFSTRUCT - PDO_COMMUNICATION_PARAMETER, etc. Not for each object of type RECORD separately.
- I would like implementation of some kind of grouping in libedssharp (modules). I would like to get for example:
```
typedef struct {
  OD_RPDOCommunicationParameter_t communicationParameter;
  OD_RPDOMappingParameter_t mappingParameter;
} OD_rpdo_t
OD_rpdo_t OD_rpdo[NO_RPDO] = {... initial values ...};
```
- Object dictionary array (`CO_OD[]`) is collection of objects, one entry for each index implemented, as it is now. But it should not be `const`, but configurable by application via function calls. No need for extra `CO_OD_extension_t`.
- SDO callback function for OD variable reviewed, should be also used for PDO.
- Application may have some parameters attached to specific OD variable: It can verify, if variable is mapped into PDO, in that case it can request PDO send or check, if PDO was transmitted. No ChangeOfState flag in Object Dictionary Editor, no CO_TPDOisCOS() function.
- uint16_t instead of UNSIGNED16, etc. usage in CO_OD.c.h

CANopenNode, 2020-01-28:
There are some related issues, for which I suppose, will be fixed by new Object Dictionary:
- [Suggestion] SDO/PDO interface #145 
- TPDO Calls Extension use of ODF_arg doesn't match SDO OD function ODF_arg behaviour #100
- inconsistent error behavior #75
- Feature - PDO serial data streaming #70
- Variable-Length Strings #71
- Conflict type name and std macro #68

I will close them now.

CANopenNode, 2020-01-28:
I have one question about libedssharp. Is it possible to run it in Linux? It seems it can't be run with mono.
I would very like for it to be portable, because I currently rarely use Windows.

Bartimaeus-, 2020-01-28:
What are you suggesting as a replacement for the current object storage scheme? The current implementation for RAM,  ROM, and EEPROM in CO_OD.h allows you to easily define the properties for storing/not storing each object using the libedssharp tool, which makes it very easy to add new objects and define how they are stored. The application already decides how to handle storage of variables by writing your own EEPROM driver. The persistence/non-volatility of some objects are already defined as part of the CANopen standard, as well as the commands to save changes to non-volatile memory. You can also define your own group besides RAM, ROM and EEPROM using libedsssharp.

Bartimaeus-, 2020-01-28:
> I have one question about libedssharp. Is it possible to run it in Linux? It seems it can't be run with mono.
> I would very like for it to be portable, because I currently rarely use Windows.

The libedssharp GUI is designed as a Windows forms app using Visual Studio. It may be possible to run it using Wine, but I haven't tested it myself. I agree that portability is ideal, however there currently isn't an alternate tool available that I know of.

robincornelius, 2020-01-28:
On Tue, 28 Jan 2020 at 18:30, Bartimaeus- <notifications@github.com> wrote:

> I have one question about libedssharp. Is it possible to run it in Linux?
> It seems it can't be run with mono.
> I would very like for it to be portable, because I currently rarely use
> Windows.
>
>   Hi,

Its a CLR .net app and it happily runs under mono (so linux or osx are
fine), no need for wine. Its also split into a library and GUI. The library
can be run with a little extension program as a standalone command line
tool to convert xdd/eds to the c/h files that canopennode currently uses.
Some of the validation logic and PDO stuff is currently in the GUI and this
needs moving to the lib. The core goal is to have a 100% implementation as
a library/standalone tool with a standalone GUI.

If there is any mono breakages please let me know on the github issue
tracker.

I already use the library portion with my canopen monitor app so that i can
load in an EDS file from a 3rd party vendor and use this to directly
read/write a remote OD and know what the names/defaults/types are and also
save an entire OD as a dcf file (a companion of the EDS) which contains
actual values ideal for deployment as appose to default values that the eds
uses.

Robin

>

geoffrey-vl, 2020-01-28:
For me it is a bit weird though to see ROM and EEPROM mentioned in the OD, where my devices don't have a EEPROM for example but some other storage type, and that term ROM is used for storing const variables.

> The libedssharp GUI is designed as a Windows forms app using Visual Studio. It may be possible to run it using Wine, but I haven't tested it myself. I agree that portability is ideal, however there [...]

It's working for my on Ubuntu 18.04. I can compile it using MonoDevelop, only the Unit Tests project is failing for me.

robincornelius, 2020-01-28:
On Tue, 28 Jan 2020 at 18:17, Bartimaeus- <notifications@github.com> wrote:

> What are you suggesting as a replacement for the current object storage
> scheme? The current implementation for RAM, ROM, and EEPROM in CO_OD.h
> allows you to easily define the properties for [...]
One problem with the RAM/ROM/EEPROM scheme is its incompatible with the
CanOpen standards for EDS and XDD  i've already cheated and added a comment
field to EDS to specify this within libedssharp. The PDO COS flag is the
other "cheat" i've had to add a comment for as that is also not specified
in the standard for eds/xdd storage.

That said it works really well and i make a lot of use of it with my PIC
projects using CanOpenNode, and i also have an EEPROM driver so i use all 3
schemes. So i would be sorry to see it go completely. But i don't have a
better suggestion at the moment.

 Robin

martinwag, 2020-01-29:
XDD also doesn't support "disabled" entries, for obvious reasons. I think, there will always be a few internal variables that are not represented in XDD interface file.

@janez I'm running entire workflow on linux. Only thing not working is the CiA conformance tester...

martinwag, 2020-01-29:
Some more thoughts...

> For me it is a bit weird though to see ROM and EEPROM mentioned in the OD, where my devices don't have a EEPROM

Yes, I think we should change names to something like "volatile", "persistent", ...
I also need to be able to add memory regions (calibration data, serial number, LSS data, ...), which you can use independent via 0x1010 and 0x1011 objects.

Another thing I miss in current implementation of the OD is that there is no encapsulation for OD data (getter/setter). Access is done either directly to variables or multiple function calls like this.

```
int Canopen::od_get(u16 index, u8 subindex, u16* retval)
{
  u16 entry;
  u8 length;
  u16 *p;

  entry = CO_OD_find(CO->SDO[0], index);
  if (entry == 0xffff) {
    *retval= 0;
    return 1;
  }

  length = CO_OD_getLength(CO->SDO[0], entry, subindex);
  if (length != sizeof(*p)) {
    *retval= 0;
    return 1;
  }
  p = CO_OD_getDataPointer(CO->SDO[0], entry, subindex);
  if (p == NULL) {
    *retval= 0;
    return 1;
  }
  *retval= *p;
  return 0;
}

/* use */
auto ret = co.od_get(OD_2220_energySave, OD_2220_1_energySave_LEDs, &es);
```
I have a function like this for every data type I use. However, this is very inefficient, as every access goes to searching the variable inside the OD table...

CANopenNode, 2020-01-30:
First of all, I don't want to drop any useful functionality here. I just want to (I will borrow famous cite here) make things as simple as possible, but not simpler. However, If there is a complex procedure, which could be simpler, I think, it is better to drop the complex procedure and make some revisions in our projects. I don't want to overdo simplification, but I think we can't keep full compatibility.

#### ROM, RAM, EEPROM
It is from the old times, where microcontrollers has enough flash but very little RAM. For that reason many global variables were `const`, so they can be hardcoded in flash. (And microcontroller has self-programming functionality, so ROM variables were also configurable.) Old 16-bit dsPIC30F4011 with 2kb RAM already has problems with current stack, but newer 16-bit dsPIC33 has not. 32-bit PIC30 and (I suppose) arm-s have no problems.

So, we will have:
```
CO_OD_entry_t CO_OD[CO_OD_NoOfElements] = { {0x1000, ... }, ... }
typedef struct {
    uint16_t index;
    uint8_t maxSubIndex;
    uint8_t storageGroup;
    uint16_t attribute;
    void* variable;
    ...
} CO_OD_entry_t;
```
Storage function can go through CO_OD[]. `storageGroup` specifies if current object is relevant for current storage and `variable` points to data location... However, `storageGroup` can be specified by libedssharp or by application. But no need for structures, based on storage location.

It's no problem, if libedssharp is cheating. I'm not even sure, if it is cheating, does standard prohibit extra parameters in eds or xdd file? However, for storage I recommend data field for `storageGroup` with values from 0x02 to 0x7F, which will suit objects 0x1010 and 0x1011.

#### Organizing of OD variables
I would keep organizing of variables into groups. If anybody needs group ROM, he can use it. I would just rename "Memory Type" into "Group" in libedssharp. However, I have some more ambitions about grouping, its just idea for now.

#### Scope of OD variables
There is a question, how to apply default values, those from factory-reset and those from storage. I think, initialized variables should stay global. Some say, [global is evil](https://softwareengineering.stackexchange.com/questions/148108/why-is-global-state-so-evil), but anyway. Globals are easy for simple programs and are also easy in relation to storage. So, after program starts, OD variables are there, initialized with factory settings. Then, if set so, storage function read the saved data and populates OD variable with them. Then CANopen gets initialized. So is now and so may stay.

#### Access of OD variables from application
OD variables are global, so simple applications just use them. But in object oriented approach I'm keeping objects separated from Object dictionary. OD variables and objects (like CO_TPDO) are bound together at initialization (CANopen.c file). I would like to make this part more clear.
From objects I never use direct access to OD variables via index/subindex as in example above: `int Canopen::od_get(u16 index, u8 subindex, u16* retval)`. I prefer binding at the object initialization.

#### Getter / Setter from SDO, PDO
OD variables are not always calculated, they sometimes depends on internal state of the object. Currently SDO may call object specific callback, which gets/sets the actual value of the OD variable. Implementation is somehow clumsy. PDO has even more clumsy solution `TPDO_CALLS_EXTENSION`, which I never use. Getter would also be necessary for storage. Getters/setters should be configurable only by application at startup by configuring two additional variables:
```
typedef struct {
    ...
    void* extension; 
    uint16_t extensionType;
} CO_OD_entry_t;
```
`extension` will be something similar as current `CO_OD_extension_t`, but multiple types of extension will be possible. It could contain getter/setter callbacks, flags, etc. CANopen object would have some information about specific OD variable like: is it mapped to TPDO, to RPDO, was TPDO sent, request TPDO send (on change of state), etc. It is only an idea for now.

CANopenNode, 2020-01-31:
I would like to discuss accessing OD variables a little more. I have some approach, but I'm not sure, if it is optimal. I would like to hear possible better solutions, how to implement OD and access to its variables in complex projects. In object-oriented way, in 'C' or in 'C++'. I currently don't use C++, but some advanced device profile may be easier to implemented in it?

Simple application can `#include CANopen.h` or `#include CO_OD.h` and operate directly on OD variables, which are global.
But if someone wish to implement for example DS402 for 8 virtual devices in one microcontroller, using of globals directly is not a good solution.

Current approach in all CANopen objects in `stack` directory is something, what I'm talking about. No file includes CANopen.h or CO_OD.h. I will try to explain my current idea in example of TPDO, communication parameters:
- CO_OD.h has structure OD_TPDOCommunicationParameter_t and CO_OD.c initializes one member with initial values.
- CO_PDO.h declares structure CO_TPDOCommPar_t, which is used for argument in CO_TPDO_init(). As a pointer to initial values.
- Both structures must be the same, but I can't verify that 100%. Verification is made in CANopen.c: `sizeof(OD_TPDOCommunicationParameter_t) != sizeof(CO_TPDOCommPar_t)`
- CO_TPDO_init() is called with `(CO_TPDOCommPar_t*) &OD_TPDOCommunicationParameter[i]` argument.
- CO_TPDO_init() initializes some internal variables in `CO_TPDO_t *TPDO` "object". In general, pointer to initial data from CO_TPDOCommPar_t not needs to be remembered.
- CO_TPDO_init() also has an argument `uint16_t idx_TPDOCommPar`, which specifies OD index of OD_TPDOCommunicationParameter in Object dictionary. CO_TPDO_init() then configures object dictionary with its own callbacks.
- When SDO or PDO or storage wants to access OD variable, it scans object dictionary and finds specific `CO_OD_entry`. For example entry for TPDO_comm_par was configured by CO_TPDO_init() so, that access to variable is now via callbacks.

----

This is actually very similar to current implementation for SDO, except I would like to simplify it a little. And add PDO and storage.

Is anybody using similar approach for own objects? Are there any other suggestions? Is there another way for implementing Object Dictionary?

CANopenNode, 2020-02-03:
Approach from my previous post is currently used for CANopen objects. But I'm clearing my idea now.

Implement getters/setters in two directions use from application or use from SDO, PDO, storage.

#### getters/setters for usage from application
Something as recommended by @martinwag, for example:
`uint16_t OD_readUint16(CO_OD_entry_t *entry, uint8_t subIdx, returnError_t *err);`
Function would verify type and return data from OD_entry. Call like this:
`uint16_t myvar = OD_readUint16(OD_find(SDO, 0x1017), 0x00, &err);`
Or optimized, if libedssharp will define correct `CO_OD_INDEX_1017`.
`uint16_t myvar = OD_readUint16(CO_OD[CO_OD_INDEX_1017], subIdx, &err);`
This could also be used inside initialization of CANopen objects for fetching default values. So there would be no need for mess with two types  `OD_TPDOCommunicationParameter_t` and `CO_TPDOCommPar_t` as in the comment above and call to `CO_TPDO_init()` would be much simpler.

#### getters/setters for usage from SDO, PDO or storage:
This is advanced functionality. CANopen object or user object can change specific `CO_OD_entry_t *entry` - register own getters/setters for data. Then data won't be read from structure defined by libedssharp anymore. For example, when SDO server will try to read data value from OD, it will call getter, which will calculate the data value from internal state of the specified object.

jdormand, 2020-02-22:
Something I've used before for a similar situation my apply here for access to the OD from the application side:

Define a struct of functions pointers that mean all access to all OD entries are via a function (pointer) call. The advantages of this approach are:

So you end up with something like: (pseudocode)

```
struct
{
   uint16_t(*countslave_tiltdata_get);
   void(*countslave_tiltdata_set)(uint16_t);
   ...
} OD;
```

In my example `countslave_tiltdata` comes from the EDS editor - it's my object name. The type and `_get` and `_set` come from the code generator, a bit like how C# generates getters and setter for properties.

Calling in this becomes:
```
myOD.countslave_tiledata_set(56);
```

To make this all work, the code gen would also have to generate stub functions, that fit the _get and _set prototypes, and initialise the structure of function pointers with pointers to the stub functions. The stub functions can then follow a basic template, very similar to about e,g. for a uint16_t setter:

```
uint_16 countslave_tiltdata_get(void)
{
  return OD_readUint16(CO_OD_t *od, 0x2026, 2);
}
```

This approach has the following advantages:

  * No direct access (unless you really wanted to work around the API) to underlying memory, so can't (easily) bypass systems that need to know about change
  * Can ensure interested parties are notified of changes, simplying other areas of the CO stack e.g. PDO COS, heart beat times etc where changes to the OD need to be reflected into the stack, which quite often takes copies of OD data.
  * Can be auto-generated from the EDS file - you don't have to know whether to call CO_readUint16 or CO_readUint32 for example - easier not to get it wrong
  * Changes to types in EDS, without changes to app code would result in compiler warnings / errors - which is great - other solutions can fail silently
  * Can enforce read only - just don't generate the setter
  * Can support default initialisation - generate an _init with a write one implementation, which is called from CO_Init or similar with default values

If there is a concern about memory usage with all the function pointers requiring storage, and the runtime overheads of calling them, then it *could* be done with a whole bunch of macros using the pre-processor and still end up with type correctness, but it'd be less elegant imo.

An improvement on this might be to add another layer. Instead of having _get, _set and _init versions of all the object access functions, instead we could define types of (common) behaviour. e.g.

```
struct
{
  uint16_t(*get)();
} readonly_uint16;

struct
{
  uint32_t(*get)();
  void(*set)(uint32_t);
} rw_uint32;
```

This way, the OD and calling it is cleaner:

```
struct
{
  rw_uint32_t myvar1;
  readonly_uint16t myvar2;
} OD;

uint16_t value = OD.myvar1.get(); // Compiler warning uint32_t truncation
OD.myvar2.set(45);  // Compiler error - set doesn't exist
```

Also a nice to have in my opinion is something similar to the existing OD_extension function callbacks / property change system:

```
struct
{
  uint8_t(*get)();
  void(*onupdate)(uint8_t value);
} readonly_sync_uint8_t;
```

Which would allow the application to set an onupdate() handler, which would then be automatically called (if set) when the OD entry was changed by the stack... IT's limited to one interested party, so not quite a true event system, but could be extended possibly.

Just some ideas - hope it makes sense... They're not massively far off what we have now but if the EDS compiler/code gen could do some of the work for us, it'd make coding the app side of things to the OD simpler and quicker.

CANopenNode, 2020-02-24:
Thank you for your ideas.
I didn't start with new OD yet, so any ideas are welcome. Best approach may be a combination of multiple ideas.

I'm not very familiar with advantages of using getters and setters in the application. I may need some hints here.

----

    struct
    {
      rw_uint32_t myvar1;
      readonly_uint16t myvar2;
    } OD;

This idea seems interesting from application side, but how to access OD from network side, with only known index and subindex? Maybe use a linked list?

----

My current idea is this:

    typedef struct {
        uint16_t index;
        uint8_t no_of_entries;
        CO_OD_entry_t *entry;
    } CO_OD_object_t;

    typedef struct {
        const uint8_t subindex;
        const uint8_t dataType_or_something;
        const uint16_t attribute;
        uint8_t dataAccessType;
        void *pData;
    } CO_OD_entry_t;

    const CO_OD_object_t CO_OD[CO_OD_NoOfElements] = {
        {0x1000, 1, &OD_entry_1000},
        ...
    }

    OD_entry_1000[1] = { 0, 4, 0x1234, 0, (void*)&CO_OD_ROM.deviceType};
    ...

Something like this will be generated by Object Dictionary editor (libedssharp). It is very similar, what we have now. `dataAccessType` will always be 0 in the beginning, that will mean, `pData` is a simple pointer to initialized data.

Application will have the possibility to change `dataAccessType` and `pData`. For example, it can configure `dataAccessType` to 1 and set `pData` to preconfigured structure like this:

    struct {
      void *object;
      uint32_t(*get)(void *object);
      void(*set)(void *object, uint32_t);
    } OD_access_1_rw_uint32;

There will be getters and setters for the data access and they will behave differently for different `dataAccessType`.

That is a rough idea for now.

jdormand, 2020-02-24:
```
This idea seems interesting from application side, but how to access OD from network side
```

Yes, this approach is only helpful from the application side. It has no benefit and indeed makes it harder to access the OD from the network side. I would propose the underlying storage had 2 APIs, one for direct access (most useful for the application, but *may*have some use from the stack side for accessing known locations) and the (index/subindex) method for stack access.

The key here is allowing any access directly to the underlying data is potential unwanted because it allows data to change without anyone knowing about it. I note the COS system currently compares data buffers to see if anything has changed - the getter/setter system would allow that to bee removed as we could simply flag up the change whenever the application sets it.

I think the underlying storage approach for the OD is sound, the downsides are really only fairly minor in the application side i.e. that the user (of the API) has know things about the internals of the OD and remember things like the type, length and access properties of the OD members... which the getter/setter approach I suggested attempts to hide away and encapsulate with techniques that take advantage of compiler time type checking.

Regarding the network access - yes, you have to have some sort of reverse lookup system or search facility to convert index/subindex pairs into an OD entry. I don't know of any better way of doing this other than searching an array / list. Basically a dictionary pattern. I don't think we can assume there is a dictionary construct in the libraries that CANOpenNode currently depends on as it's dependencies on 3rd parts libs is minimal (which is good), so an implementation of a dictionary may be required to facilitate this. I don't think that is a bad thing however, it means that the exact mechanism of finding an OD entry from an index/sub-index is abstracted away from the SDO/PDO stack code and allows for 3rd parties to more easily evaluate, optimise or replace it for their chosen platform if for whatever reason there was deemed a need to do so. I think the current implementation relies on OD_find() or similar (sorry working from memory) so we're not a million miles away from this type of solution now, it just depends on whether you wanted to formalise it a bit...

jdormand, 2020-02-24:
How easy is it to extend / add to the code generation from libedssharp - is it a fairly closed black box or is it something we can extend? I have no familiarity with this lib development at all...

CANopenNode, 2020-02-24:
https://github.com/robincornelius/libedssharp basically reads standard CANopen device description file (with few extra properties) and generates pair of CO_OD.h/.c files. Don't want to put much extra stuff there. (I plan to move COS functionality away from there.)

libedssharp generates default values as initialized variables. This is useful for "factory reset". It is also easy to populate the variables from storage before CANopen objects are initialized. CANopen objects then have default values at initialization and CANopen objects can then specify own getters/setters for network side(SDO, PDO, storage). CANopen objects operate on own data and provide getters/setters for network access to own data.

With this approach there will also be no need for deeper change of the existing code, but existing code will be simplified.

Simple application can use OD variables directly - no need for getters/setters.

jdormand, 2020-02-24:
Sounds good, I'll be interested in how the COS stuff changes.

I see regarding libedssharp - it would be (a lot) more work to extend the code gen side... still I might have a play, see if there is any interest in new features in the libedssharp github project

Simple is always good :)

jdormand, 2020-02-24:
Actually, whilst we're on the subject - can I ask for some advice?? It's related to application development / accessing OD variables from directly on the master side (not a device/slave/server)

To make my motor move, requires a PDO, currently I do this:

```
OD_Motor_PDO_Data.pdo4_driveCtrl = 0x02;
OD_Motor_PDO_Data.pdo4_modeCtrl = 0x03;
OD_Motor_PDO_Data.pdo4_Ref16 = 250; //velocity;
OD_Motor_PDO_Data.pdo4_Ref32 = -98425; //position;
CO->TPDO[1]->sendRequest = 1;
```

Assuming COS is not set for these entries, is this the only / best way to send the PDO down to my device? Or have I missed something?

CANopenNode, 2020-02-25:
Yes, for now it is the only way to send PDO manually.

However, more usual approach to send time sensitive (motor data) is by using CANopen SYNC object and send synchronous PDOs. Intensive use of ChangeOfState mechanism can overload CAN in critical moments. With synchronous PDOs there is more traffic on CAN, but it is deterministic.

For new OD I plan a callback for PDO request (If variable will be mapped to some TPDO, then application will have an option to trigger it)

jdormand, 2020-02-25:
Ok cool - thanks v much

geoffrey-vl, 2020-02-26:
J1939 is supported by Linux kernel nowadays: https://www.kernel.org/doc/html/latest/networking/j1939.html. I'm not sure how well CANopen compares to J1939, but maybe that could give some extra insights?

FreddieChopin, 2020-04-23:
I just glanced through the posts briefly and I want to add something important about getters/setters. I use C++ in my embedded applications exclusively, so I have some experience with object-oriented design. Part of my experience is also that C projects frequently approach object-oriented design in a suboptimal way (sorry...).

If you want to have getters/setters, then these functions must (and I really mean it - they _MUST_!) also get a pointer to some user provided state. This would be the equivalent of C++'s `this` and in C land it's just a standard `void* context` or maybe a pointer to the base `OD_t*` object, assuming that the user can easily "extend" or "inherit" it (this is tricky to do correctly in C, especially if most of the code would be auto-generated).

I'm writing this because the proposal from @jdormand from https://github.com/CANopenNode/CANopenNode/issues/162#issuecomment-589973899 does not have such pointers. While these pointers may seem useless in some cases, just imagine that the user wants to have the most trivial OD possible - each object is mapped to an element of an array (or a global variable). The only "twist" is that the OD is huge, say 1000 objects. If you have the extra pointer with the context, you can implement it entriely with two functions - one common setter and one common getter. Without this pointer you end up needing 2000 separate functions, which can never be inlined or optimized away (because the application stores the pointer to these functions) - 1000 setters and 1000 getters, each being basically identical.

As I've read @CANopenNode replies, CANopenNode code currently accesses underlying data of the object dictionary directly. Someone above proposed two APIs - one for the app, second one (raw API) for use by CANopenNode (I think having these 2 APIs would not be a perfect situation, but it's actually not that bad). But does CANopenNode really have to have direct access to the binary representation of the data? Let's say that there's a getter which returns `uint32_t` but CANopenNode just wants to get one byte from the middle. True, in this case the getter returns  4 times more data than required, but is this really a problem? The same can be said about setters - if there's a `uint32_t` setter and CANopenNode just wants to change one byte, why can't it read the whole entry, change this single byte and then write the whole entry again? API based on getters/setters indeed nicely solves the problem of notifying the application that the entry was changed by the network (but this can be also done with an user-supplied callback) and notifying the CANopenNode stack that the entry was changed by the application.

One last thing - I think it would be very hard to auto-generate generic code with custom getters/setters, while also making them object-oriented. This is probably much easier when doing like @jdormand proposed (just name the function like the object and expect the application to provide it), but how would that work in case of functions which require some additional state? No idea... It is probably much easier to just have optional, user-supplied callbacks executed when the entry is changed by the network (but please - also here the `void* context` is essential!). As these callbacks would be optional, the generator would not need to know anything about them - it would just add support for them in cases of objects which are writable from the network, but all callbacks would be disabled by default. In such scenario, the raw representation of the OD data can actually be hidden entirely from the application.

jdormand, 2020-04-23:
@FreddieChopin Thanks for the comments, I entirely agree that in a general case, implementation of OO in a non-OO language will require some manual implementation of the this pointer or equivalent, (context etc). However I don't see how that applies to this specific case in particular I don't see how extra context is appropriate to the code snippets I proposed:

`myOD.countslave_tiltdata_set(56);`

and

`OD.myvar1.get()`

as they already have context by virtue of the fact that they are members of a struct.

In the first case, which is a little clunky, the function pointer `countslave_tiltdata_set` is a member of the root OD struct. In the second, which is an improvement I think, the `get` is a member of the struct of which myvar1 is a type. In my example this was:

`
struct
{
  uint16_t(*get)();
} readonly_uint16;
`

Assuming this was what you were commenting on, could you clarify for me? Have I missed something?

The comment about overhead is absolutely fair, the downside of this approach is that every object dictionary entry would have (by the eds code generator) and additional function pointer for a ro or wo type, and 2 for a rw. In addition, there would be a lot of single line functions, in exactly the same way that C++ and C# would have, although thankfully in the latest C# it's all built in so a lot of the laborious typing is no longer required.

The question really becomes, do the benefits outweigh the costs - by forcing OB Dict access through a function, whatever the form is, the stack can react more effectively and efficiently to changes originating from the application/user code, and crucially could be auto-generated in. And conversely, changes to the OB Dict originating from the stack (network) could be effectively communicated (on an as needed basis) back to the user code, and the crucially the support for this also auto-generated.

That said @CANopenNode later comments on extending the code-generator are valid and sadly it may be difficult to extend the code generator in a way that the community in general would accept, so what then, a CANopenNode specific code generator?? Probably not :/

FreddieChopin, 2020-04-24:
> as they already have context by virtue of the fact that they are members of a struct.

This solves nothing, as there is absolutely no way to get back to this context from the function - in a generic way. If you would want to have a generic function - say "get_motor_speed" and your application would have 10 motors, it would not work. You would need 10 separate functions, each being basically identical. If you have just one function anyway, then you indeed don't need the context, but please don't forget about cases where there are multiple instances of similar functionality. We don't need to look very far - the example object dictionary of this repository has 8 digital inputs, 8 digital outputs, 12 analog inputs and 8 analog outputs. If you would have to implement it with a context-less functions, you would need 52 functions for this alone (I'm counting getters and setters). If the functions would get the context as an argument, all it would take is 6 functions.

> In addition, there would be a lot of single line functions, in exactly the same way that C++ and C# would have

That's my main point. If the functions don't get the context you need as many functions as there are OD entries. If they get the context, in some cases one function can be used as a handler for many OD entries. You don't always need this context, but when you do there's no workaround for the case where this pointer is just not given to the function.

> The question really becomes, do the benefits outweigh the costs - by forcing OB Dict access through a function, whatever the form is, the stack can react more effectively and efficiently to changes [...]

I guess this can be solved in a slightly less user-friendly way, which is much more code-generator-friendly. How about having each entry in the following form:

```
struct OD_entry
{
... // whatever CANopenNode needs...
void (*notifyApplication)(struct OD_entry* entry);
}
```

By default the pointers to `notifyApplication` would be all `NULL`, and I'm assuming the user can replace them with whatever he/she likes (I have also a solution for a case where it would be preferrable for this pointer to be stored in flash memory). Whenever stack changes such entry, it would call this callback to notify the application about that event.

For the opposite case, I propose to have a global (there's no point in having pointers in each entry) function in a form:

`void CO_OD_notifyStack(CO_t* co, CO_OD_t* od, struct CO_entry* entry);`

(the list of arguments is obviously just an example, the `entry` is obligatory, the rest can be improved)

Whenever user changes one entry, it should call this function to notify the stack about that fact. This is slightly less convenient for the user (however there is no tragedy, it's quite good and easy to use), but I guess this requires zero tricky features in the object dictionary generator.

> That said @CANopenNode later comments on extending the code-generator are valid and sadly it may be difficult to extend the code generator in a way that the community in general would accept, so what then, a CANopenNode specific code generator?? Probably not :/

Why not? Assuming that there's a separate tool for editing and creating EDS files, making a generator is not that hard really. libedssharp can be still used just to make EDS files, but we would have CANopenNode-specific tool just to convert an EDS file into source code. It would just be a simple python script which would read the EDS and use one of many template engines for python to output whatever sources you like. Personally I use [Jinja2](https://jinja.palletsprojects.com/en/2.11.x/) and I think it's very good (I'm using it to generate board sources for my [C++ RTOS](https://github.com/DISTORTEC/distortos), given the input in YAML file, similar in idea to what "devicetree" does for Linux). For the "parse EDS" part I hope that packages like [canopen](https://canopen.readthedocs.io/en/latest/) already have all the necessary functionality (and even if they don't, it's just a simple ini-like file anyway). Such generator would be also much easier to modify by the user, because the template files are just simple text files, pretty easy to understand (assuming you have some very basic knowledge about python - things like if-else, for loops, ...).

CANopenNode, 2020-04-24:
Thank you for your comments.

I'm talking about newOD for quite a long time and I still didn't start with it. However, I have some ideas and I'm also very thankful for your comments as they help to get better solution.
The thing is, I don't have 100% time for CANopen, I also have other things to do. And second thing, I'm currently consolidating some other parts of CANopenNode (split-driver branch, move some parts from CANopenSocket, etc.). As my time enables, I will start with newOD soon.

I can't comment precisely your last posts now, I have to start diging into newOD first.

Janez

martinwag, 2020-04-30:
Another thing as it seems you are changing quite a bit of code now: Add **unit tests**. We all make mistakes, and implementing test cases, especially for error cases, helps find them. After you have the tests, you can make changes to code with some degree of certainty not to break another thing.
Protocol stacks are normally a prime use case for unit testing as you have very clear input and output interfaces :-)

For simplifying tests you can use a test framework. This leaves you with writing expectations, like this:
```
#include "path/to/mock-turtle.h"
#include "gmock/gmock.h"
#include "gtest/gtest.h"

using ::testing::AtLeast;                         // #1

TEST(PainterTest, CanDrawSomething) {
  MockTurtle turtle;                              // #2
  EXPECT_CALL(turtle, PenDown())                  // #3
      .Times(AtLeast(1));

  Painter painter(&turtle);                       // #4

  EXPECT_TRUE(painter.DrawCircle(0, 0, 10));      // #5
}
```
(example from here: https://github.com/google/googletest/blob/master/googlemock/docs/for_dummies.md).

Can someone reccomend a "C" test framework? The example is google mock / C++, and that is a pain in the ... to use for C code.

JuPrgn, 2020-04-30:
Ceedling is the right tool for this job it uses unity and cmock. https://github.com/ThrowTheSwitch/Ceedling

CANopenNode, 2020-04-30:
I have currently no experience with unit testing. I make own tests when writing new code, but it is not the same.

There was an effort for new OD by someone else, but it stalled. There is also unit testing, but unfortunately stack is very changed. I didn't go into details there: https://github.com/IzidorM/CANopenNode_OD

Ceedling seems nice, is there any example for some part of CANopenNode or something similar?

JuPrgn, 2020-04-30:
There are some "Embedded code" examples on Ceedling in folder "examples". 

The test code from your link is using Unity which is  used by Ceedling and part of the same open source test framework. You can look at unity and cmock doc to get what Ceedling do.

martinwag, 2020-04-30:
But please keep default build tool at "make", this is what comes with even the most obscure toolchain...

CANopenNode, 2020-05-12:
I just pushed quite a lot of new code (CiA309-3 command interface, CO_fifo and rewritten CO_SDOclient). I have been testing it during the creation and in the finish. It works fine so far.
However I think, unit testing could be quite useful, especially for some parts of the code.

If unit testing will be implemented in CANopenNode, I don't like to add it to base git repo. 
I would rather add CANopenNode testing code into CANopenSocket repo, which can be used also for some other Linux examples.

gotocoffee1, 2020-06-15:
Hello, I have started with the implementation of the CANopen safety protocol (EN 50325-5 / CiA 304) I will also be making a pull request soon. One problem I stumbled upon is the mapping. Basically it is similar to PDO mapping, with the strange difference that the mapping object is not a record but an array.

The thing is that you have to change the maximum subindex of the array to support variable mapping. But this is not possible in the current OD implementation. Currently, I treat it like a record as a workaround. Perhaps we will find a more suitable solution in the new OD implementation.

CANopenNode, 2020-06-15:
Yes, this is currently a limitation. It will be solved with new OD.

I have just finished other updates to CANopenNode and I'm starting with new OD now.

You safety protocol contribution will be very welcome.

CANopenNode, 2020-07-13:
FINALLY - new object dictionary interface is here.

It is specified by two files: `301/CO_ODinterface.h` and `301/CO_ODinterface.c`.
Currently there are only those two files, compilable and full documented, no integration into the stack yet.
Basic principle and all good things from previous implementation stays the same. Everything is simplified, optimized for low memory devices and missing features are added. SDOserver, PDO, application, etc., all will use the same read/write access. Simple global access is still possible. Object dictionary will be generated with libedsshap as before, standard xsd files will be used. I think, I found quite a nice solution. Please view the files and add comments or suggestions.

Next, it is necessary to update libedssharp and SDOserver, PDO and other objects, which use OD variables.

gotocoffee1, 2020-07-15:
Very cool interface! I haven't tested it but what I've seen from the code looks promising.

One thing I notice in the new files but also in some of the already existing once is the C90 code conformance. There are some constructs that do not conform with Ansi C (e.g. [flexible array member](https://github.com/CANopenNode/CANopenNode/blob/master/301/CO_ODinterface.h#L517), usage of stdint.h, etc.). In my opinion we can drop the C90 support and move on to C99 (which is already 20 years old) since its already kinda broken.l

martinwag, 2020-07-15:
We've discussed this a few times ([https://github.com/CANopenNode/CANopenNode/issues?q=c99](https://github.com/CANopenNode/CANopenNode/issues?q=c99)), but never made a desicion. Only thing against is that some compiler manufacturers _cough cough Microchip cough cough _ are still not implementing it.

CANopenNode, 2020-07-15:
You scared me. I somehow still use Microchip based hardware.
So I tried to compile latest master with CO_ODinterface.c with Microchip XC32 and XC16 - and it worked. That is good enough for me.

I think, we can move to C99. Is there something, what could make problems to Microchip compilers?

CANopenNode, 2020-08-13:
Some news:
New OD interface is advancing steady. Already pushed CO_ODinterface seems OK. SDO server is rewritten. There is also some work with transferring "OD access functions" for each CANopen object. Then comes libedssharp with OD generator and finally testing.

CANopenNode, 2020-10-07:
News:
newOD branch is pushed. It compiles well, but doesn't work, because there is only blank object dictionary.

Current changelog:
- New Object dictionary interface. It has very similar principles as before. All parts of CANopenNode objects, which works with OD entries, are rewritten.
- New OD.h and OD.c files, replaces CO_OD files.
- CANopen.c and CANopen.h files redesigned. Integration of "OD.h" is optional. Configuration of multiple object dictionaries is possible with one CANopen device. Interface is the same, with some changes to function arguments.
- Rewritten SDO server. Object dictionary part is removed.
- CO_Emergency is mostly rewritten. Now is much easier customization.
- CO_NMT_Heartbeat is redesigned.

Now follows libedssharp. testing, and other objects: HB consumer, sync, PDO, time, SDO, trace

ddario, 2020-10-26:
I saw the new CO_ODinterface.h/c implementation but didn't understand how it's actually used in the stack:
in CO_ODinterface.h I see OD_entry_t which compose the OD_t but in all the other parts of the stack objects are structures of type CO_OD_entry_t and the two are not simple casts.

Is it planned to switch the CO_OD_xx_t types to OD_xx_t or did I totally misunderstand the goal of this OD change?

CANopenNode, 2020-10-26:
CO_ODinterface.h/c in master branch is currently only for preview. Was.

But in branch newOD, there is a working and tested part of the stack with fresh CO_ODinterface.h/c.

So, anyone interested may view and test the new OD principles:
- https://github.com/CANopenNode/CANopenNode/tree/newOD (SDO, NMT, HB producer)
- https://github.com/robincornelius/libedssharp/raw/gh-pages/build/OpenEDSEditor-latest.zip
  - preferences -> CANOPENNODE_V3
- there is also a brand new [CANopen test suite](https://github.com/CANopenNode/CANopenSocket/tree/master/test/running_canopen) and extended `cocomm` interface in CANopenSocket.

ddario, 2020-12-11:
Hi Janez, everyone and sorry for the delay in replying but I took some
time to organize some ideas.

> CO_ODinterface.h/c in master branch is currently only for preview.
> Was.
> But in branch newOD, there is a working and tested part of the stack
> with fresh CO_ODinterface.h/c.
> So, anyone interested may [...]
https://github.com/robincornelius/libedssharp/raw/gh-pages/build/OpenEDSEditor-latest.zip
> preferences -> CANOPENNODE_V3
> there is also a brand new CANopen test suite and extended cocomm
> interface in CANopenSocket.

I gave it a brief look and the new OD interface looks much better than
the old one, at least from coding perspective.

The entire project looks great and impressive.

I know that the CANOpen best seller devices still are small uC that in
a way or the other have strong requirements on RAM/FLASH size or
processing speed and usually don't use OSs so all kind of possible
optimizations are welcome (defines to short circuit parts of the code,
defines to easily access well known objects in OD ...).

But I think that, slowly and silently, this is changing: building
tailored custom distros for many uP is now easy and RAM/FLASH
availability increases and ...

Another consideration I'd put on the table is that the stack is only
one part (even if the great one) of the CANOpen application and, at
least to me, the most annoying part of coding is doing the same thing
many times. I mean that there are lots of devices where the only
difference is in the OD contents but the application that handles it
depends on a .c/.h file that has to be derived from a .eds and than
requires to rebuild the application.

This makes me think that it would be great to develop code that
directly eats a .eds file and dynamically creates the OD.

Is this somehow interesting for you?

If anyone wants to discuss about it please let me know.

With best regards,
Daniele.

martinwag, 2020-12-11:
Hi Daniele,

having OD build from a description file (eds, xml, ...) at run-time is a great thing for powerful controllers, most likely master/managers. This also makes software development much easier.
This will, however, also make your software impossible to run on controllers with less than about a megabyte of ram/flash. And on a system of that size I will make serious thoughts about using ethernet/mqtt or similar instead of can/canopen... 
With CAN-XL 2048 Byte frame on the horizon, in my opinion using etherent protocols will be the way to go anyway :-)

best regards,
Martin

ddario, 2020-12-11:
Hi Martin

yeah, of course CAN is not the latest media/protocol to use when you're
dealing with fast uPs with lots of RAM/flash.

The problem is that in many places CAN is largely used and
gateways/converters are required to let user replace their networks
with newer devices.

This is why I'm rising the question: convert from CANOpen to  another
media is not a big deal, but every device has it's own OD and I'd avoid
creating tons of applications just to deal with them when the
difference is so little. Many applications means lots of trouble IMO.

.eds files are not a big deal to parse so I'd like to know if someone
is interested in this to share the work load.

Of course, basing it on CANopenNode stack requires that you guys are
interested in it too and that's why I'm asking.

Cheers,
Daniele.

geoffrey-vl, 2020-12-11:
> Hi Janez, everyone and sorry for the delay in replying but I took some time to organize some ideas.
> On Mon, 2020-10-26 at 11:49 -0700, Janez wrote: CO_ODinterface.h/c in master branch is currently only for preview. Was. But in branch newOD, there is a working and tested part of the stack with fresh CO_ODinterface.h/c. So, anyone interested may view and test the new OD principles: https://github.com/CANopenNode/CANopenNode/tree/newOD (SDO, NMT, HB producer)
> https://github.com/robincornelius/libedssharp/raw/gh-pages/build/OpenEDSEditor-latest.zip
> I gave it a brief look and the new OD interface looks much better than the old one, at least from coding perspective. The entire project looks great and impressive. I know that the CANOpen best seller devices still are small uC that in a way or the other have strong requirements on RAM/FLASH size or processing speed and usually don't use OSs so all kind of possible optimizations are welcome (defines to short circuit parts of the code, defines to easily access well known objects in OD ...). But I think that, slowly and silently, this is changing: building tailored custom distros for many uP is now easy and RAM/FLASH availability increases and ... Another consideration I'd put on the table is that the stack is only one part (even if the great one) of the CANOpen application and, at least to me, the most annoying part of coding is doing the same thing many times. I mean that there are lots of devices where the only difference is in the OD contents but the application that handles it depends on a .c/.h file that has to be derived from a .eds and than requires to rebuild the application. This makes me think that it would be great to develop code that directly eats a .eds file and dynamically creates the OD. Is this somehow interesting for you? If anyone wants to discuss about it please let me know. With best regards, Daniele.

You'd need some storage to read that OD file from: ROM, Flash memory. And a platform independent way to accomplish that.
Relying on platform specific code is actually not something we want, hence why the various drivers where removed from the project last year. So in the end you'd have to implement that yourselves too, unless you rely on someone else's driver.
At best we'd have a platform abstraction layer that works with any kind of uController, but currently the choice is to let that not be part of stack but rely on any kind of PAL within the driver code.

martinwag, 2020-12-11:
> but every device has it's own OD and I'd avoid
> creating tons of applications just to deal with them when the
> difference is so little. Many applications means lots of trouble IMO.

Most likely, if your OD is that different between two devices, they actually are two different devices with different functionallity and different hardware. In that case, you will most likely have many differences in firmware anyway. 
For mostly similar devices, you can have a OD implemented that fits multiple devices, and switch  the entry implementation and EDS for each device in runtime. Only problem is that CiA tool checks if there are more entries implemented in OD than EDS.

> .eds files are not a big deal to parse so I'd like to know if someone
> is interested in this to share the work load.

.eds is outdated, current format is XML CiA311. As @geoffrey-vl mentioned, this is not a problem of parsing, but of memory and functionallity for file storing.  You also need to transfer the file onto the CANopen device somehow in the first place.

Please don't misunderstand me. I think this functionallity would be great. I really missed it when building our CANopen manager software on top of CANopenNode. But I prefer to have to only introduce myselve into one stack that fits all instead of one small and one large one, even if that brings some compromises...

ddario, 2020-12-11:
> Most likely, if your OD is that different between two devices, they
> actually are two different devices with different functionallity and
> different hardware. In that case, you will most likely have [...]

I'll try to better explain myself: let's assume we want to use a well
tried CANOpen device but the network where it has to be connected is of
different kind (e.g. ethernet or other dedicated busses).

Obvious answer is the device doesn't fit the requirements so find out
one that works on the network in use but if this was the case I
wouldn't have asked.

So we need a gateway/converter. This "in-the-middle" device will have
an OD designed to listen for TPDOs sent by the end device, with SDO
client/server or whatever needed to configure and talk with the end
device. Let's forget for now the other side and how communication has
to be handled.

Now, it comes up that another end device that does something different
from the first one and so has a different OD can't be replaced: we need
another "in-the-middle" device to connect it to the network.

Assuming that the hardware for the "in-the-middle" devices will be the
same, the OS distro also, is the only available approach to create
different applications to handle this?

> For mostly similar devices, you can have a OD implemented that fits
> multiple devices, and switch the entry implementation and EDS for
> each device in runtime. Only problem is that CiA tool checks if there
> are more entries implemented in OD than EDS.

Of course: that's why only what's in eds/xml has to be exposed by the
application.

> .eds is outdated, current format is XML CiA311. As @geoffrey-vl
> mentioned, this is not a problem of parsing, but of memory and
> functionallity for file storing. You also need to transfer the file [...]

As you said, xml or eds, is not a problem when you have ram/flash,
speed and an OS that can help. Also, being this not an end device but a
"pc like" device, the idea is to load it as a configuration file saved
in flash. Instead of building an application, you configure it.

> Please don't misunderstand me. I think this functionallity would be
> great. I really missed it when building our CANopen manager software
> on top of CANopenNode. But I prefer to have to only introduce [...]

No problem: I'm trying to collect information and every suggestion is
welcome even if it is a "you can't do it that way".

So is this idea so wrong?

Regards,
Daniele.

CANopenNode, 2020-12-11:
Hello Danielle,

I think, with new OD it is perfectly possible, what you suggest.
Object dictionary is basically an array of OD_entry_t objects and it can safely be generated on program start, from whatever data source. The only requirement by OD_find() function is, that array is sorted by index.
However, OD_find() is not used by the stack directly, it is used by application, so also this function can be customized. For example, configuration of object dictionary as linked list or even dynamic adding/removing of OD entries in run time may be possible.
CANopen.h and .c files works in two modes: with or without '#include OD.h'. In later case CO_config_t structure has to be filled with pointers to multiple OD_entry_t objects, which can be defined anywhere. Of course, even CANopen.h and .c files can be customized.

If a system with more advanced object dictionary usage will be made, for example generating object dictionary from eds on program start, it is probably best to be own project, with own git repo, built on top of CANopenNode.

Regards,
Janez

ddario, 2020-12-11:
Helly Janez

> Hello Danielle,
> I think, with new OD it is perfectly possible, what you suggest.
> Object dictionary is basically an array of OD_entry_t objects and it
> can safely be generated on program start, from whatever data source.
> The only requirement by OD_find() function is, that array is sorted
> by index.

Yeah, that's what I intended with "from a coding point of view".

> However, OD_find() is not used by the stack directly, it is used by
> application, so also this function can be customized. For example,
> configuration of object dictionary as linked list or even [...]

Basically this is what I started doing.

First thing I'd like to know was if I was going nuts, and from your
answer it seems it's feasible.

Than I'd know if this could be of any interest also for you guys, at
least to share ideas.

> Regards,
> Janez

Thanks everybody for replaying and you're welcome if you have hints or
thoughts to share.

Best regards,
Daniele.

CANopenNode, 2020-12-11:
For me it is not actual at the moment. But I would certainly like to add references (into the readme.md) to related projects, especially, if they may be usable for others. Sharing ideas about the tools, we use in our projects, is the beauty of open source.

geoffrey-vl, 2020-12-15:
> Also, being this not an end device but a
"pc like" device, the idea is to load it as a configuration file saved
in flash. Instead of building an application, you configure it

Come to think about it, CANopenNode is really not exclusively useful for uController kind of projects. With the stack supporting Linux devices, and basic master functionality, I bet more and more people will look to extend its purpose from a demo tool to something that is running on a daily basis. I think we should take that into account. I think the stack should not _enforce_ having a statically compiled OD. Maybe its best for your project, but it stands correct that even a medior Cortex M4 has enough RAM,ROM,Flash to pull the trick of dynamically loading the EDS file that is for example 200kb big. I don't think at this stage many people will do that, but at least now its not possible because everything has to be pre-allocated with zero values.
On linux you'd be writing mostly dynamic application code, because it's running on a generic OS. You may generate the OD from EDS, I for example was thinking of loading a custom description from SQL(lite), and some other people may decide for their application they're OK with a fixed machine/network config.

The most important thing to tackle first is allowing to support dynamically generation of OD. I believe this is already possible with newOD, but I haven't looked at it myself. From their on, custom extensions and demo application code can be shared, for example within CANopenSocket.

> Can someone reccomend a "C" test framework? The example is google mock / C++, and that is a pain in the ... to use for C code.

I've been using mbed greentea in the past. You may not like it since it is bound to mbed ecosystem. But, it would allow to kick an entire CI/CD flow whenever a PR appears. This flow could build the code, deploy it to a uController device, and run unit tests on the device itself. 
Here is picture of some server room at some ARM office showing that in action:
[image]

FreddieChopin, 2020-12-15:
> Maybe its best for your project, but it stands correct that even a medior Cortex M4 has enough RAM,ROM,Flash to pull the trick of dynamically loading the EDS file that is for example 200kb big.

That's true. There are chips that could load such file dynamically. And do nothing more than that...

Please don't blur this issue, which is about supporting multiple instances of CAN on a single, statically linked application. There are dozens of options available for Linux and I don't think anyone is going to choose CANopenNode over - say - canopen python module, which does all that you want right now, with no need to extend it at all.

geoffrey-vl, 2020-12-15:
> > Maybe its best for your project, but it stands correct that even a medior Cortex M4 has enough RAM,ROM,Flash to pull the trick of dynamically loading the EDS file that is for example 200kb big.
> 
> That's true. There are chips that could load such file dynamically. And do nothing more than that...
> 
> Please don't blur this issue or this project, which is meant to be used on microcontrollers. There are dozens of options available for Linux and PCs and I don't think anyone is going to choose CANopenNode over - say - canopen python module, which does all that you want right now, with no need to extend it at all.

I did :) When sharing codebases between applications and platforms it's easier to choose for any kind of C/C++ stack instead of higher level python based implementations. It's not only about features... Furthermore, if linux is not one of the target platforms then why is its driver supported in mainline CANopenNode?

martinwag, 2020-12-16:
> That's true. There are chips that could load such file dynamically. And do nothing more than that

We're using a cortex M4, but not with 200k+ RAM. We have 32k, which also includes the application stuff. On linux / PC like system you can use as much memory as you like compared to that :-) .

>  Furthermore, if linux is not one of the target platforms then why is its driver supported in mainline CANopenNode?

Linux is one of the target platforms. It's also the easiest target platform to obtain, which makes it a good candidate for the "default target".

On both the [law of the minimum](https://en.wikipedia.org/wiki/Liebig's_law_of_the_minimum) applies.

FreddieChopin, 2020-12-16:
> Furthermore, if linux is not one of the target platforms then why is its driver supported in mainline CANopenNode?

I'm not the author, so I can only guess, but I would say that mainly for testing.

geoffrey-vl, 2020-12-16:
> We're using a cortex M4, but not with 200k+ RAM. We have 32k, which also includes the application stuff. On linux / PC like system you can use as much memory as you like compared to that :-) .

To clear things up : there is no need to load that entire EDS file into RAM. You'd still only allocate only the things that you've put into the EDS, so in the end the same OD is generated and allocated in RAM. During parsing there is no need to load the entire ASCII EDS file to RAM, you can parse it line by line I'd say. That's 100bytes?
I've applied such thing in the past on CortexM4, it's not impossible. I'm not sure that dynamically loaded OD doesn't consume _that much_ more memory than statically allocated. It's mostly just differently allocated.
And on some systems you'd opt for dynamic allocation (linux, medium and high-end CortexM4 and up), where on others you'd opt for least possible memory footprint and the predictability and ease of static allocation (for example yours), or PIC32, or ...).

I do understand the extra complexity, we don't want yet more PRECOMPILER statements in the code, there are already much to consider.  I'm not asking "loading of EDS" from source. I'm questioning if we couldn't make an interface that allows the developer to choice how the OD is allocated? Instead of having the stack poking into the OD directly: go through the OD interface... More features will follow as from that point on the application developer is free in deciding how to allocate things. 
It's not about that the stack is not working for linux, because it is, and linux is also clearly an important target platform to consider. From the README:

```
...
CANopenNode is written in ANSI C in object-oriented way. It runs on different microcontrollers, as standalone application or with RTOS. Linux implementation with CANopen master functionalities is included.
```

So it's really about: do we want this 'static/dynamic allocation' feature, or not? Next, do we want it to be included _at this state_, _now_ that the OD is being reworked? Obviously adding it afterwards will be a bigger pain when considering backwards compatibility. But maybe we're fine with that... and the outcome may well be that it is not possible to support this anytime soon due to lack of manpower, or that it adds too much complexity for the majority of the current users at this moment. We may also decide to consider it a feature that may come in CANopenNode 5 or later...  As @ddario says, IT evolves, and you must take into account that the projects you've started years ago will be different than the projects in few years from now.

CANopenNode, 2020-12-16:
> Linux is one of the target platforms. It's also the easiest target platform to obtain, which makes it a good candidate for the "default target".

That is correct. Linux is also very useful for testing and for developing. And Linux have one of the most standardized programming interfaces.
There have to be a "default target". There have to be a working reference.
However, Linux "driver" may not be most suitable as example, because it is quite complex. But there are other examples.

martinwag, 2020-12-16:
> I'm not the author, so I can only guess, but I would say that mainly for testing.

I've written that driver for 1.3 stable branch to be used in productive environment, and we are using it for exactly that purpose :-)

> To clear things up : there is no need to load that entire EDS file into RAM. You'd still only allocate only the things that you've put into the EDS, so in the end the same OD is generated and [...]

As said before, this is not a problem of parsing. It's a problem of transfering and storing the EDS. Of course, that can be solved, even on small controllers, but at what cost in parts and engineering.

CANopenNode, 2020-12-16:
I think, "dynamic allocation ability" is not a question any more, since newOD. It is available.
OD can be switched in running program at the communication reset (not tested, but should work).
OD of type OD_t is passed to SDO server, SDO client (and application) init functions.

[34 more replies]

closed 2021-05-17 by CANopenNode