CANopenNode/CANopenNode #165: Support for large SDO upload on memory restricted SDO clients

heliochronix, 2020-02-12:
Hello everyone,

I'm opening this issue mostly to start a discussion on how to add support for large SDO uploads when the client is resource limited in memory. To give some context, I'm currently working on a system that uses an STM32F4 with limited RAM as the NMT and SDO client for the overall system. As it currently stands, I need to transfer large blocks of data at a time via the DOMAIN data type.

The problem is, with the current implementation of `CO_SDOclientUploadInitiate()` and `CO_SDOclientUpload()` it just takes a simple buffer pointer and maximum size, and then starts writing as it receives. The limitation here is if I can only have a buffer of (for example) 1kB due to available RAM, but have several GB of non-volatile storage and need to transfer 5MB via an SDO client upload to the non-volatile storage, then there's no way to accomplish this. This is also a restriction going the other direction as well I believe (SDO client download, with data in non-volatile storage on the client and limited RAM for a buffer)

I was trying to think of a good way to solve this, and one option I was thinking was to provide an option for the SDO client transfers to configure not just the maximum TX or RX data size (like the current implementation), but also a maximum buffer size with the option of invoking a user defined callback function to "empty" or "fill" the buffer. This would effectively allow for processing of large blocks of data with limited RAM resources. This might be analogous to the `CO_SDO_OD_function` system used for DOMAIN data types, where the callbacks are used to handle the data on the SDO servers.

The limitation I see is when you're actively receiving data (SDO client upload), the SDO server will keep sending its data even if you are in the middle of processing the current block of data in your buffer via your callback. I don't recall if the protocol has an ability to have the client signal a wait condition or not, or if the internal buffering of CANopenNode's SDO code will take care of holding data until the user provided callback returns and it can start writing over the user provided buffer again.

What are your thoughts on this?

martinwag, 2020-02-13:
The protocol doesn't have "flow control" like functions. However, you can either use SDO expedited transfer or block transfer with small block size and delay your confirmation for a few ms.

heliochronix, 2020-02-13:
That's roughly what I was thinking. Perhaps the whole idea builds around SDO_BUFFER_SIZE, where on sending or receiving a callback can be called. On SDO client upload, it calls it immediately before doing a SDO_BUFFER_SIZE byte transfer. On SDO client download, it calls it after transferring SDO_BUFFER_SIZE bytes but before sending the confirmation as you mentioned.

With something like that in place, then perhaps the callback can have access to the buffer offset used within the SDO transfer to optionally reset it back to 0, in case the buffer is limited in size? It would still need to track the overall amount transferred though I think. Does this sound reasonable to you?

heliochronix, 2020-02-13:
Actually, SDO_BUFFER_SIZE may differ between devices, so I think that's not the proper thing to use. Perhaps the negotiated SDO->blksize is the boundary to use?

CANopenNode, 2020-02-13:
It is some time since I study SDO block transfer, but I think, it is no problem here. SDO block transfer executes in sub-blocks, where each sub-block can transfer up to 889 bytes of data (SDO_BUFFER_SIZE).
And every sub-block is confirmed. So before sending the confirmation one can empty the buffer, on the client or on the server side.

heliochronix, 2020-02-19:
Sounds good. If you don't mind, I can look into implementing this in a branch off split-driver (in anticipation of split-driver being merged into master at some point in the future).

jdormand, 2020-02-21:
Hi - sorry for hijacking this forum post, but I couldn't find another easy way to drop you a message. I noticed you are working on CANOpenNode on an STM32F4 - I am also, and have a port of the CO_driver for this target based off some old STM32F1/bxCAN code from another project. I wondered if you had an STM32F4 specific driver and whether you were using the HAL or the StdPeriphLib in your projects? I thought it might be nice to collaborate on an STM32F4 driver to upload to the CANOpenNode project?

heliochronix, 2020-02-21:
I followed up via your email from your profile.

CANopenNode, 2020-02-23:
@jdormand, please see split-driver branch of CANopenNode. Read `doc/deviceSupport.md`.
You are welcome to open an issue here to start a collaboration. There are some attempts for STM32F4, but I currently don't use it, so I can't support it much, but I may offer some hints.

Driver for STM32F4 should be a separate git repository. If there will be a collaboration I can also open a project here and set a maintainer.

martinwag, 2020-02-24:
We're using an STM32F3 device, and after evaluating quality of ST drivers decided to drop them entierely (in 2016). In my opinion, they are a nice examples for the reference manual and first tests, but evaluating them for production use is as much effort as writing your own.

CANopenNode, 2020-06-09:
SDO client is now "rewritten", especially part, which transfers data to/from application. Now it uses very small fifo buffer and should be possible to (block) transfer very large ammount of data. However, it is not much tested with large data transfer yet.

CANopenNode, 2020-12-28:
In my latest tests, I correctly transferred (SDO block, both directions) 10GB large file (tested on Linux, virtual CAN). With just few kb large buffers. See https://github.com/CANopenNode/CANopenSocket/tree/master/examples/basicDevice
So I think, this issue is solved now.

closed 2020-12-28 by CANopenNode