CANopenNode/CANopenNode #168: Possible BUG in CO_HBconsumer.c

jdormand, 2020-02-21:
I believe the idx checking in methods such as:
```
CO_HBconsumer_initCallbackHeartbeatStarted
CO_HBconsumer_initCallbackHeartbeatStarted
CO_HBconsumer_initCallbackRemoteReset
CO_HBconsumer_getState
CO_HBconsumer_getNmtState
```
is incorrect. The code in question is part of the bounds check of the `idx` parameter passed into these functions:
```
idx>HBcons->numberOfMonitoredNodes
```
I believe that `idx` is zero based and the check should therefore be `>=`

This really comes to what the property `numberOfMonitoredNodes` means in relation to the array `monitoredNodes`. Conventional wisdom suggest that if the array was declared [4] then `numberOf...` should be 4 and valid indices would be 0-3, but just in case I misses some subtlety, I have checked further:

These properties as setup in `CO_HBconsumer_init()` which in my code is currently called with
`monitoredNodes` = `CO_HBcons_monitoredNodes`
`numberOfMonitoredNodes` = `CO_NO_HB_CONS`

`CO_HBcons_monitoredNodes` comes from the main CO object ultimately references:
```
static CO_HBconsNode_t      COO_HBcons_monitoredNodes[CO_NO_HB_CONS];
```

CO_NO_HB_CONS comes from the EDSeditor created CO_OD.h/.c and ultimate resolves to
```
#define ODL_consumerHeartbeatTime_arrayLength               7
```

Either way, `numberOfMonitoredNodes` does appear to relate directly to the array length, so I believe that the `idx` checking is incorrect.

This also seems to be supported by code in `CO_HBcons_monitoredNodeConfig`:

```
if(idx >= HBcons->numberOfMonitoredNodes) return;
```

which indicates that only values of `idx` unto n-1 (inclusive) are value.

and numerous places in the code which loop the array using:

```
for(i=0; i<HBcons->numberOfMonitoredNodes; i++)
{
...
```

jdormand, 2020-02-21:
Just found this in comments which seems to support my theory
```
 * @param HBconsTime Pointer to _Consumer Heartbeat Time_ array
 * from Object Dictionary (index 0x1016). Size of array is equal to numberOfMonitoredNodes.
 * @param monitoredNodes Pointer to the externaly defined array of the same size
 * as numberOfMonitoredNodes.
 * @param numberOfMonitoredNodes Total size of the above arrays.
```

CANopenNode, 2020-02-24:
You are right, there is a bug.
Fixed in split-driver branch.
Thank you.

closed 2020-02-24 by CANopenNode