CANopenNode/CANopenNode #172: Implementing CANFD

evsevsevs, 2020-03-26:
Hi Janez,

What are you thought on implementing CAN FD?
How much impact does this have on the implementation? My guess is that the most imported change is the data size from 8 to 64 bytes which is hardcoded in the implementation.
Do you thing there are more important changed needed?

Thanks,
EVS

martinwag, 2020-03-26:
This has a few more implications:
- your hardware will be incompatible to non-FD hardware
- we need to implement the CiA 1301 estensions like USDO if we want more functionallity than just 64 byte PDOs.
- we need to extend the PDO stuff to support 64 byte wide mappings
- we need to change the driver layer

evsevsevs, 2020-03-27:
Thanks for your comments. That is a bit more complicated than I was hoping for.  There is not that much detailed information available.
Do you know if CANFD improve bus load or only bus speed? Does a lot of small message from different devices also benefit from CANFD? 
Thanks

martinwag, 2020-03-27:
There is quite some info about FD, you only have to find it :-)
- [https://assets.vector.com/cms/content/know-how/can/Graphics/CAN_FD_Poster_V2.2.jpg](https://assets.vector.com/cms/content/know-how/can/Graphics/CAN_FD_Poster_V2.2.jpg) (you can order that as poster for free too, I think it's worth the wait)
- [https://www.vector.com/de/en/know-how/technologies/networks/can/](https://www.vector.com/de/en/know-how/technologies/networks/can/)

FD currently only has a real advantage if you use messages larger than 8 byte. Most of the CANopen FD extensions should be possible to implement for standard CAN too. Encryption most likely won't because that makes use of larger messages.
If you have the choise, select hardware that is at least _FD tolerant_, because otherwise your device can't be integrated in an FD network.

CANopenNode, 2020-04-29:
CAN-FD may be a challenge for the future. CANopenNode should be flexible enough for its integration.

CANopenNode, 2020-12-28:
I'm closing this for now, until it will become more relevant.

closed 2020-12-28 by CANopenNode