CANopenNode/CANopenNode #173: Multiple instances?

FreddieChopin, 2020-04-14:
Hello!

In some CANopen profiles there is a need to have many independent CANopen instances on one device. For example "CANopen Application Profile for Lift Control Systems" (DSP417) describes several configurations where something like that is required. most important ones:
- devices working as gateways/bridges/extenders,
- central controller devices which need to access two/three separate networks (for example one network is for all "call" buttons next to lift doors, second network is for communication with the lift itself, third network for controlling motor drives).
[image]
[image]
[image]

I'm pasting this to show what's the real-world use case for a multi-instance scenario.

Let's say I'm mostly interested in a scenario with two completely separate object dictionaries and two separate CAN interfaces in a single device. Let's also say I'm using a MCU with flat address space, not an embedded Linux with processes and virtual addressing. Am I right to assume that such things cannot be done with CANopenNode at the moment? Or maybe doing something like this would be possible, but very awkward? Maybe it just requires some minor edits in the source files?

Thanks in advance for any clarification!

CANopenNode, 2020-04-15:
Hello,
The problem is Object Dictionary, where currently are fixed global variables. This is oldest part of CANopenNode and is planned to be rewritten in next few months. For now it could be possible to manualy rename global variables in second object dictionary.
Next thing are CANopen.h/c files, which are fixed to one CANopen device. For complex configuration they have to be customized.
Other CANopen objects are independent from Object Dictionary and from CANopen.h/c files. I think, there should not be much effort to implement two CANopen interfaces.

Please note also, there is a branch split-driver, which will soon be merged into master and is recommended for new designs.

martinwag, 2020-04-20:
Hi,

I think the most difficult problem with this is to get the driver right, as you now have multiple instances using one driver. This means you have to handle calls independently to some degree. This also means you have to analyse received messages to select a device instance (or all of them) a message is sent to.

FreddieChopin, 2020-04-20:
> I think the most difficult problem with this is to get the driver right, as you now have multiple instances using one driver.

Well, there will be single driver source code, but there will be definitely multiple driver objects.

Generally now I'm only gathering info and planning my actions, for now I have the single interface working (or at least it seems so). I also have to learn a lot about CANopen (; When my application will be using more features I'll then move on to implementing the second instance.

In my opinion the most difficult part will - indeed - be the object dictionary _AND_ all the constant macros defined along with it (like the number of SDOs, number of PDOs, whether or not the device is a SDO master or slave). I think it would also be convenient if all of the "lock" macros (like `CO_LOCK_CAN_SEND()` / `CO_UNLOCK_CAN_SEND()`) would take the associated object as their first argument. This way it would be possible to lock sending (and the two other things) only for one instance instead of globally. In this case you would obviously have to move the mutex / lock into appropriate objects - for example - into the `CO_CANmodule_t` struct. But it would be required to also extend the definition of the `CO_EM_t` (for `CO_LOCK_EMCY()` / `CO_UNLOCK_EMCY()`) and one more into the object dictionary itself (for `CO_LOCK_OD()` / `CO_UNLOCK_OD()`).

cfr-mir, 2020-08-03:
Hello,
I am actually working on using multiple busses and OD's. Check branches "Support_Multiple_ODs" in my forks of libedssharp and CANopenNode.

My approach are to make libedssharp  generate code with a prefix on all of the constants in the *.h file .In the *.c file I initialize a struct which contains the constants for the OD. like CO_NO_SDO_CLIENT.

A pointer to the const struct is added to the CO_t struct, which is passed down in all the initialization file in CANopen.c, there are also a few references to the constants in the stack code.

This is a work in progress and it WILL break the API of CANopenNode, as some functions have added a CO_t pointer parameter. Further more you must initialize som new members of the CO_t struct before calling CO_init. Like below:

  COO.consts=&MYPREFIX_CO_Consts; // Tell Where the constants are found.
 // StatusBits this could be a global variable if you don't want it to be accessible in the OD.
  COO.errorStatusBits=MYPREFIX_OD_errorStatusBits; 
  COO.errorStatusBitsSize=MYPREFIX_ODL_errorStatusBits_stringLength;
  COO.em_syncronizationWindow=100;
  CO_init(&COO,&MYPREFIX_CO_OD[0],driverstatus,nodeid,1000);

Freddie you have a point about the locks. I havent addresed this in my implementation. I just disable/enable interrupts.  That always works.

The impact on the API can be minimized. I just didn't reach that point yet. Additionally there are some impact on RAM/ROM usage compared to the single OD version of CANopenNode.

jcatsong, 2020-08-19:
Just want to clarify after reading this thread, does CANOpenNode currently support a single master with multiple slaves on the same CAN bus (CO_NO_SDO_CLIENT seems to indicate so)?  And this thread is more about adding support for a single master and multiple slaves on different CAN buses (networks), or multiple masters each controlling a different bus with one more more slaves on it?

FreddieChopin, 2020-08-19:
This thread is about having multiple separate instances of CANopenNode, each interfacing with its separate CAN controller (possibly connected to separate CAN networks).

CANopenNode, 2020-10-07:
It seems, it will be possible to make flexible OD configurations with next release, currently available in newOD branch.
See https://github.com/CANopenNode/CANopenNode/issues/162#issuecomment-705035805

geoffrey-vl, 2020-12-01:
Although not really related to having multiple instances, the socketcan driver should allow for opening multiple socketCAN sockets. So on embedded linux Figure2 is possible.
Another way to roughly get to that results is to use the can-gw module. This will allow you to bridge your networks. Messages on can0 will also appear on can1 and vice versa. Now you need to open only one socketcan socket, which means you don't need to edit any of the code within CANopenNode to have your master control nodes on the 2 physical CAN networks. On my embedded linux I found latency to be <1ms, measured with candump tool. So if that's OK for you, here is how to do that.

First you need the load the can-gw module for the linux kernel. Probable you need to enable it first in your kernel build.
Once you have the module build into your kernel, you can:

```
$ modprobe can-gw
```

After enabling the module, setup your can interfaces:

```
$ ip link set can0 type can bitrate 250000
$ ip link set can1 type can bitrate 250000
$ ip link set can0 up
$ ip link set can1 up
```

Now you can bridge your 2 networks:

```
$ cangw -A -s can0 -d can1 -e
$ cangw -A -s can1 -d can0 -e
```

In one terminal, start monitoring you both CAN busses:

```
$ candump -t A can0&
$ candump -t A can1
```

In another terminal start your NMT master:

```
$ ./canopend can0 -i 2
```

Notice how in your candump session you now see all nodes messages in both physical networks:

```
 (2020-12-01 09:41:29.107717)  can0  082   [8]  00 50 01 2F F3 FF FF FF
 (2020-12-01 09:41:29.108300)  can1  082   [8]  00 50 01 2F F3 FF FF FF
 (2020-12-01 09:41:30.097539)  can0  702   [1]  7F
 (2020-12-01 09:41:30.097814)  can1  702   [1]  7F
 (2020-12-01 09:41:30.718237)  can1  705   [1]  05
 (2020-12-01 09:41:30.718564)  can0  705   [1]  05
 (2020-12-01 09:41:31.097640)  can0  702   [1]  7F
 (2020-12-01 09:41:31.097918)  can1  702   [1]  7F
 (2020-12-01 09:41:31.408718)  can0  703   [1]  7F
 (2020-12-01 09:41:31.408989)  can1  703   [1]  7F
```

FreddieChopin, 2020-12-01:
> So on embedded linux...

Unfortunately (or not (; ) I'm using a microcontroller (STM32) with custom RTOS (https://github.com/DISTORTEC/distortos), so I cannot use the options you describe.

> $ ip link set can0 type can bitrate 250000
$ ip link set can1 type can bitrate 250000

BTW in these applications the bitrates for the interfaces are usually different, one is 125k, the other is 250k.

CANopenNode, 2020-12-28:
Since newOD https://github.com/CANopenNode/CANopenNode/issues/162 this issue is solved, I think.

closed 2020-12-28 by CANopenNode