CANopenNode/CANopenNode #177: Source organization of complex CiA modules

martinwag, 2020-04-27:
Hi,

This continues off-topic from https://github.com/CANopenNode/CANopenNode/pull/175#issuecomment-619995977

> 
> I certainly don't want CANopenNode to become fat or interlaced. Actually I'm trying to keep it configurable also for thin configuration. For example enable only expedited SDO transfer, etc.
> 
> There are two options:
> 
>     One project, some optional modules, like 302(CANopen manager), 309 (gateway), but no device profiles or project or microcontroller specific code. Everything well organized and independent as possible. Only modules independent from microcontroller included (except basic Linux driver)
>     Separate projects for each module, like CiA309, CiA302, etc.
> 
> I prefer first option, because everything is nice on one place and everybody use only, what he needs. There may be some extra code, which can be safely ignored. Otherwise, I'm afraid, things can become too partitioned. For example, SDO client should not be part of the basic stack.
> 
> Sorry for off-topic.

You should look at this from viewpoint of network layers, not CiA specs. They are a mess in that regard...
- 303 Hardware stuff, wiring
- CAN transmitting, receiving (not part of CiA)
- 301 CANopen, 305 LSS and 306 Object Dict, 303-3 (LED blinky), which are what currently is part of the stack.
- 302-1 ... 302-9 management / PLC stuff and network interface parts of the 400 application profiles. Those specs rely on the stuff from 301, 305, ...
- 400 application profiles combinatory logic and 309 gateways, those again rely on the lower layers
- user application

martinwag, 2020-04-27:
In my opinion, we should separate those layers. That doesn't mean those can't be inside a single git repo :-)

CANopenNode, 2020-04-29:
Hi,
I introduced some of my ideas of source organization in #161. I think, it will be better organized, if things will be added to CANopenNode.

CANopenNode must certainly not become bloated. But I want to keep standardized stuff inside single repo.

CANopen itself is useful for many very different devices, so CANopenNode must be configurable for many different configurations.

For example, most tiny configuration should include only few source files:
- CO_driver.h with other driver files (which are customizable anyway)
- SDO server with only expedited transfer enabled. Small object dictionary with mandatory CANopen objects.
- NMT slave and Heartbeat producer, which are also mandatory by CANopen.
- main and CANopen.h/.c files necessary to connect everything.

It should be possible to add other independent modules to build a rich CANopen device.

There should be an example for tiny configuration and also example for rich configuration, where all source files can compile and run out-of-the-box (in Linux environment).

----
I introduced 301/CO_config.h file with some macros defined. They enable functionalities inside different parts of the stack. Default values are inside CO_driver.h. Those values can be overridden inside CO_driver_target.h file, which is specific for microcontroller. And latter can further be overridden inside CO_driver_custom.h file, if present.

For example, I made SYNC functionality optional inside PDOs. See file 301/CO_PDO.c. for lines with `#if (CO_CONFIG_PDO) & CO_CONFIG_PDO_SYNC_ENABLE`. If sync is not necessary for the custom project, then files CO_SYNC.h/.c can be safely removed from it.

Many `#if` statements inside code may reduce readability, but I think, it is not critical. If there are some other ideas, please share.

CANopenNode, 2020-06-09:
I finished transferring of CANopen master command interface (CiA309-3) from CANopenSocket project into CANopenNode. If this part was prototype there, here is more mature and independent from OS.

Project has file structure reorganized. It is more configurable now (see CO_config.h). Driver files was removed from the project, because they were unable to maintain. Linux example is fully added to the project. There are quite a lot of lines changed, some things are improved or extended. But project is almost the same, as it was before. There should not be much effort to update existing applications or drivers. `canopend` from CANopenNode works almost the same as old `canopend` from CANopenSocket.

Object dictionary interface did not change yet, this is plan for next and final major step.

CANopenNode, 2020-06-09:
I updated also LSS, original author is @martinwag.

LSS master is used from CANopen master command interface. 305/CO_LSSmaster.c is mostly unchanged. File 'CANopenSocket/canopend/src/CO_LSS_master.c' (the one with do{}while loops) is not needed anymore. I think, usage with `canopend` is as before.

LSS slave has some rearrangement. First, CO_LSSslave_receive() function now works similar as in other objects. It it can safely be used also inside fast interrupts now. Time consuming functions are moved to mainline CO_LSSslave_process(). Otherwise functionality is the same, code was just moved between the functions.

LSS slave in now also more integrated into CANopen.c file. Handling of pending node-id and bitrate is a little different.

CANopenNode, 2020-12-28:
Source is now organized in versions 2.0 and 4.0. Usage of separate sections of the code is very configurable through macros. From very light devices with SDO expedited transfer only to rich devices with quite complex gateway. With new Object Dictionary also multiple instances of it are possible.

closed 2020-12-28 by CANopenNode