CANopenNode/CANopenNode #187: How can I configure the dict for PDO communication?

linghuix, 2020-05-23:
if I have PDO producer whose node is 1 and PDO comsumer whose node is 2.

TPDO of node 1:
Index 0x1800 ,subIndex 0x1, 0x181; subIndex 0x2, Transmission_type 1
1A00 mapped = 0x60410010;

RPDO of node2:
0x1400 ,subIndex 0x1, 0x202; subIndex 0x2, Transmission_type 1
0x1600 mapped 0x60410010;

Is it correct? Thanks.

geoffrey-vl, 2020-05-26:
I've made a schematic overview some time ago that shows how you can map one device (node 0A) its general purpose input on TPDO1 onto another device's (node 17) its general purpose output on RPDO1.

Follow the green lines from node 0A to node 17:
[image]

linghuix, 2020-05-26:
> You probably want node2 to receive PDO from node1. You have Transmission_type 1, so you have to enable sync producer on one of the nodes.
> 
> You can keep node1 as is. RPDO on node2 should receive PDO from node1, so RPDO on node2 must have the same COB-ID as PDO transmitted from node1, in your case 0x181. That is it.
> 
> Usually you can keep default COB-IDs on TPDOs and set RPDOs to matching COB-IDs, you want to receive. In this case you don't need a master.
> 
> You can also map RPDO to some other variable in case 6041 is already used for something else on node2.

“RPDO on node2 should receive PDO from node1, so RPDO on node2 must have the same COB-ID as PDO transmitted from node1”.  But why standard CAN frame requires COB-ID of TPDO between 181h-1FFh and  RPDO1  between 201h - 27Fh ? It is impossible.

CANopenNode, 2020-05-26:
It is just pre-defined, it is not required by standard.
But some devices may not allow other than pre-defined on first four R or TPDOs.

linghuix, 2020-05-30:
> It is just pre-defined, it is not required by standard.
> But some devices may not allow other than pre-defined on first four R or TPDOs.

I have built the PDO communication according to your instructions. Thank you very much!

CANopenNode, 2020-06-09:
Thank you for your feedback.

closed 2020-06-09 by CANopenNode