CANopenNode/CANopenNode #190: errorRegister calculation issue

pobingxiaoxiao, 2020-05-28:
I tried to use the CO_errorReport function to send mergency messages. It works ok but the errorRegister(1001h object) send together with emergency message is not correct.

For example, I test it with CO_errorReport(CO->em, CO_EM_RXMSG_WRONG_LENGTH, CO_EMC_COMMUNICATION, 0x00000001); which is a communication error, the bit 4 of errorRegister(object 1001h should be set) but I got 0x00. In the CO_errorReport function, it use higher 5 bits as errorStatusBits index, and in CO_EM_process function, if byte 2 or byte 3 is not equal to zero, it will set the CO_ERR_REG_COMM_ERR bit of errorRegister. 
For CO_EM_RXMSG_WRONG_LENGTH(0x02), following the rules, the errorRegister bit 4 will not be set.

I think we need to redefine the CO_EM_errorStatusBits macros in CO_Emergency.h to make it correct, am I right?

CANopenNode, 2020-06-09:
CO_EM_RXMSG_WRONG_LENGTH is currently set as 'info' priority, not 'critical'. So error register is not set. I'm not 100% sure, what is correct, because standard is not very specific about error register. We may need some discussion.

CANopenNode, 2020-12-28:
Emergency module is rewritten in newOD branch and is more configurable.

closed 2020-12-28 by CANopenNode