CANopenNode/CANopenNode #192: Change rx, tx buffers ident on the run for additional SDO

lukegluke, 2020-06-01:
Hi @CANopenNode!

For now CANopenNode supports multiply SDO servers, but only with predefined  parameters. One must set client to server and server to client COB-IDs in 0x1201 (and up to 0x127F according to number of servers) before SDO initialization otherwise additional SDO servers remain invalid.

I would like to implement handling dynamically changes of parameters of additionally supported SDO servers. CO_CANtxBufferInit returns pointer to CO_CANtx_t * buffer and that's good and useful in any ways, and particulary for my case I could use it to set different ident (for example adding CO_CANrxBufferSetIdent to CO_driver).
But I wonder why CO_CANrxBufferInit doesn't return a pointer to buffer? It's just a legacy, because it was needed anywhere?
Changing CO_CANrxBufferInit is quite a big thing that brakes quite a lot, you will not merge such changes (in old master)?

CANopenNode, 2020-06-02:
There is actually no serious reason, why CO_CANrxBufferInit doesn't return a pointer to buffer.

I think, it is not such a big thing to change that. Stack is now in the middle of the change between major releases anyway.

Currently CO_CANrxBufferInit returns CO_ReturnError_t, but the only information needed is, if init succeeds or fails. So, if it returns NULL pointer in case of failure, it is just fine. It will be necessary to update multiple files in CANopenNode (only).

If you will change old-master, I will change master too.

lukegluke, 2020-06-02:
Ok, good! I will try my approach.

lukegluke, 2020-06-02:
In fact returned CO_ReturnError_t was used only in PDO and SDO.
I've implement dynamic SDO, I make pull request let's continue there.

closed 2020-06-02 by lukegluke