CANopenNode/CANopenNode #198: ESP32 Driver

guillermo-ruffino, 2020-06-19:
Is there any reason ESP32 is not done already? (Other than nobody did it I mean)

I might need this so I can try, I'm pretty new to can bus and can open in general, so quite lost.

I've been looking what is needed to support it, and I guess I could try to adapt the [SDK given by espressif](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/can.html) for the driver.

This thing uses FreeRTOS, so for thread management, etc, is there some driver which uses this already I could look into? I only found https://github.com/CANopenNode/CANopenNode/pull/28#issuecomment-365392867 regarding FreeRTOS so far.

I'm building my own network of nodes, so no compatibility with other systems is needed so far, so I still have to learn how to setup the nodes, if I understand properly, I have to use the example and somehow configure them differently so they are different nodes?

I have ESP32's and sn65hvd230 transceivers to connect them. I've already got them communicating in plain can bus.

I don't expect full feedback of this broad question, just any tips you think are worth mentioning I'll appreciate.

martinwag, 2020-06-22:
Hi,

no specific reason, it's just that nobody needed it yet :-). Didn't even know that ESP32 has CAN bus integrated...

We're having a FreeRTOS port of the driver that works with an self-written hardware layer driver. That's why I didn't add it to main repo. 
If you're intrested I can push it, you then have to adopt the hardware layer driver to the ESP32 stuff.

guillermo-ruffino, 2020-06-22:
Yes! I guess that will be great if I can reuse the sync macros for FreeRTOS, I was about to start digging in Zephyr as they support ESP32. But I'd like to learn FreeRTOS more

martinwag, 2020-06-23:
https://github.com/martinwag/CANopenNode/tree/neuberger-freertos/stack/neuberger-FreeRTOS

This is based on v1.3-master repo. Please be aware that this was not made to be added to main stack.
- You should be able to use the thread stuff, which does all the timing things. 
- The driver port only adds canopen specific stuff to the hardware layer driver. The hardware layer driver is not included in this commit!
- You can remove the log / LED stuff, it's not needed for general use.

guillermo-ruffino, 2020-06-30:
Well thanks, I'm doing some progress here, got the library kind of working, still need many details.
Ended up using latest version, not 1.3 and the threading will need to dig deeper later.

I am using PlatformIO, so project structure has changed, now all .h files are in an include folder, and c files in other. 

The esp32 has a custom FreeRTOS, modified mostly because the esp32 is a dual core mcu, it is based in FreeRTOS 8, and the code you gave me is using stuff which is not available in that version, so wasn't a plug and play solution.
Anyway, esp32 has other constructs like real time timers which I'm using, I still hasn't pay much attention to sync stuff so race conditions are there waiting to fail.
Heartbeat and NMT seems to work as intended, I can change the preop / start / stop modes, also SDO for changing the heartbeat speed works.

I really don't know where to start for PDO stuff, as I'm very new to CANopen and even CAN Bus, I still need to figure out how to setup PDO communication, unfortunately this is not in the great readme guide of CANopenNode which I'm using successfully with a Raspberry PI and I connected an MCP2515 to it.

I see there are at least 2 ways of sending PDOs,
1. unsolicited, i.e. I believe my app logic just wants to send a PDO, should I call a function of CANopenNode? 
2. via the Sync mechanism, again, how I'm suppoused to make a fresh value available to the lib?

CANopenNode, 2020-06-30:
You setup PDOs in running CANopen network by changing PDO mapping and communication parameters.

Your application must only use some variable from OD and "fill" it with data. This variable is then ready to be mapped into TPDO, for example.

guillermo-ruffino, 2020-07-01:
Thank you! I got it working now, it sends data when it changes, still need to learn how the SYNC option works

Quick question, is the library comparing memory to check if there is a need to send a TPDO?

martinwag, 2020-07-02:
Yes.

Bascy, 2020-07-10:
Im really interested in running CanOpen on ESP32, our project will have several boards with ESP32 SOCs communicating with eachother, connected through a CANbus. I already have a propriety protocol running, but looking to expand and standardize the communication so we can create a more generic solution.

@guillermo-ruffino Are you willing to share your results so far?

CANopenNode, 2020-07-13:
I added a link to Neuberger-FreeRTOS into `/doc/deviceSupport.md` file.
If there will be something new, feel free to send a pull request or just write a note about update in deviceSupport.md file.

glmnet, 2020-07-14:
@Bascy yes, I haven't worked much on this after my last comment, but I can send you what I have done which is not much, but will allow you to start working from something that communicates data. The easiest for me will be to zip the thing and send it to you or post it here, I'm using platformio to build the project on vscode, so I had to modify the structure, all .h files are in an `include` folder, etc.

guillermo-ruffino, 2020-07-14:
@Bascy yes, I haven't worked much on this after my last comment, but I can send you what I have done which is not much, but will allow you to start working from something that communicates data. The easiest for me will be to zip the thing and send it to you or post it here, I'm using platformio to build the project on vscode, so I had to modify the structure, all .h files are in an include folder, etc.

Bascy, 2020-07-14:
@guillermo-ruffino Thanks, a zip to basschouten65(AT)gmail.com would be great
Im using the same set of tools so no problem there

guillermo-ruffino, 2020-07-14:
[esp32_CANopenNode.zip](https://github.com/CANopenNode/CANopenNode/files/4921689/esp32_CANopenNode.zip)

hope this helps!

nathanRamaNoodles, 2020-07-20:
I've looked over the web, and I'm surprised to see how few examples there are for CANopen and CANopenNode.  I made a repo for my findings and tests: https://github.com/nathanRamaNoodles/CANopen-ESP32-nodes

Currently, I've forked @guillermo-ruffino and [@xXAM22Xx](https://github.com/xXAM22Xx/CANopenESP32) into an ESP-idf format.  So far I've gotten both to transmit properly, but I have no idea what these Object dictionaries and SDO/PDOs mean.  Hopefully open-sourcing this will get us somewhere.

-Nathan Ramanathan

CANopenNode, 2020-12-28:
@guillermo-ruffino, thank you, link to your file is now in deviceSupport.md

closed 2020-12-28 by CANopenNode

jblanz, 2021-01-04:
As I am also interested to use CANopenNode on ESP32: Has anyone tried to integrate the work of @guillermo-ruffino into the Eclipse IDE? 

@guillermo-ruffino: Thanks for your very kind contribution! Did you make any progress since you provided the link to the zip archive? I am wondering if PDO Tx/Rx is working and if Sync objects can be processed?

Thanks a lot for your feedback!

Torgano, 2021-03-18:
I am also interested in CANopen for ESP32.
My question: is 
> [esp32_CANopenNode.zip](https://github.com/CANopenNode/CANopenNode/files/4921689/esp32_CANopenNode.zip)
still the most recent work to start from / continue with?

mkuznecovs, 2022-05-06:
> I am also interested in CANopen for ESP32. My question: is
> 
> > [esp32_CANopenNode.zip](https://github.com/CANopenNode/CANopenNode/files/4921689/esp32_CANopenNode.zip)
> 
> still the most recent work to start from / continue with?

Hi guys. Currently I work on the ESP32 project translating OPC UA commands into CANopen commands for a motor control. I was wondering if any of you made  progress on getting TPDO, RPDOS, Sync, etc. working on ESP32 (esp-idf framework)?

jblanz, 2022-05-09:
Hi Maksims,

I have not done anything for getting CANopenNode running on ESP32, sorry.

Josef

notiflux, 2023-03-11:
For anyone interested, here's a version of the object dictionary file that works with the current version of [CanOpenEditor](https://github.com/CANopenNode/CANopenEditor). Note that, as of writing this comment, the latest binary release contains a bug in which the logic is reversed for RPDO/TPDO mappings (see [this PR](https://github.com/CANopenNode/CANopenEditor/pull/24)), so either build from the bugfix branch or apply the patch from the PR manually. Don't forget to switch the exporter to CANOPEN_LEGACY.

Also, remember to set `CO_VERSION_MAJOR=3` in your build flags in `platformio.ini`.

[deviceFile_barebones.xdd.txt](https://github.com/CANopenNode/CANopenNode/files/10949009/deviceFile_barebones.xdd.txt) (remove .txt from filename)