CANopenNode/CANopenNode #199: newlib warnings

guillermo-ruffino, 2020-06-22:
I'm trying to compile this on esp32 which uses the [newlib](https://sourceware.org/newlib/) library and getting the following warnings:

````
In file included from src\CANopenNode\301\CO_fifo.c:28:0:
src\CANopenNode\301\CO_fifo.c: In function 'CO_fifo_cpyTok2Vs':
[...]/framework-espidf/components/newlib/include/ctype.h:57:54: error: array subscript has type 'char' [-Werror=char-subscripts]
 #define __ctype_lookup(__c) ((__ctype_ptr__+sizeof(""[__c]))[(int)(__c)])
                                                      ^
[...]/framework-espidf/components/newlib/include/ctype.h:68:23: note: in expansion of macro '__ctype_lookup'
 #define isgraph(__c) (__ctype_lookup(__c)&(_P|_U|_L|_N))
                       ^
src\CANopenNode\301\CO_fifo.c:1098:11: note: in expansion of macro 'isgraph'
       if (isgraph(c) == 0 || c == DELIM_COMMENT) {
           ^
````

Looks like `isdigit(c)` gives a warning when `c` is char, which sounds a bit odd, but indeed.

I'm casting them to `unsigned int` which is a bit ugly

fenugrec, 2020-12-12:
should be fixed in 03df5f65b803e1c9a95736afb9043a46f0ef7c92 , no ?

CANopenNode, 2020-12-28:
It is fixed there.

closed 2020-12-28 by CANopenNode