CANopenNode/CANopenNode #208: delay between gateway commands

gotocoffee1, 2020-07-08:
After the gateway has completed a command, it returns with the `timerNext_us` variable unaffected, which can cause a delay of up to 50 ms before the next command is processed. This adds up, of course, if you want to poll data frequently.

CANopenNode, 2020-07-08:
I'm not sure, if I understand the problem.
CO processing is triggered by interval timer (which can be shortened by timerNext_us). But it is also triggered by other events, so it should also be triggered when new command is received from socket. So there should be no delay. 
Or there is special case, where this is not true?

CANopenNode, 2020-07-08:
I checked again CO_GTWA_process() function and there may be a problem, as you describe. If there are multiple commands in the buffer (which is possible), then unnecessary delay will be there. I will fix this now, please verify, if it will work better.

gotocoffee1, 2020-07-08:
Yes indeed this only applies when multiple commands are in the buffer.

gotocoffee1, 2020-07-08:
Thank you works perfectly.

closed 2020-07-08 by gotocoffee1