CANopenNode/CANopenNode #209: Gateway timeout for small values

gotocoffee1, 2020-07-13:
Hello, I have found another small problem in the gateway code.
If you set the gateway timeout values to a value smaller than your sleep interval in the main loop, you will get a timeout error without sending the SDO message at all. The reason for this is that the variable `timeDifference_us` (value is 50000 or higher because there was nothing to do in the last iteration) is passed to the function 'CO_SDOclientUpload', but no time has passed since the call of `CO_SDOclientUploadInitiate`. This only happens when the command queue is empty since we fixed the unnessaray wait time in my last issue.

My first thought was to reset the variable `timeDifference_us` when a read/write command is detected, but I'm not sure if this has any effect on other commands.

CANopenNode, 2020-07-14:
I did, as you say in last sentence.
`timeDifference_us` argument is used later in the function, but in other states. So it is no problem.
I didn't change `timeDifference_us` for LSS functions, because there it is not a problem.

closed 2020-07-14 by CANopenNode

gotocoffee1, 2020-07-15:
Thanks again, works great now.