CANopenNode/CANopenNode #211: Make command fails, CentOS 7

garrett928, 2020-07-14:
Hi, I'm attempting to install CANopenNode for a project but am running into build issues. After cloning the repo and running the make command the build instantly fails with this error: 
[image]

I haven't seen others report this issue so I'm not sure if it's an issue with my machine or maybe something introduced in a recent commit. Should I be using the 1.3 branch as opposed to master? 

Note: I am running CentOS 7. I don't know if this would affect CANopenNode? Thank you for any help!

gotocoffee1, 2020-07-15:
My hottest guess is that your compiler version is too old. __has_include is a gcc extension introduced with gcc 5.X.

CANopenNode, 2020-07-15:
You can safely remove that line.

garrett928, 2020-07-15:
> You can safely remove that line.

after removing that line I receive a new error: 
[image]

should I modify this block 
```
#if __has_include("CO_error_custom.h")
    #include "CO_error_custom.h"
#else
    #include "CO_error_msgs.h"
#endif
```
to be just `#include "CO_error_msgs.h"`?

CANopenNode, 2020-07-16:
`CAN_ERR_CRTL_ACTIVE` is from Linux system file, in Ubuntu `/usr/include/linux/can/error.h`. It is included with `#include <linux/can/error.h>`

Can you upgrade your Linux version?

You can also disable that "else if" statement, it is not so important.

garrett928, 2020-07-16:
> Can you upgrade your Linux version?

The CentOS VM is provided to me from my work. I might be able to upgrade it? 

Interestingly though, I complied and built fine following the CANopenSocket repo. I've been using the tools the past two days. I'm a little bit confused by this since part of building CANopenSocket is building CANopenNode. 

Also, looking back I might have been running candump or another can tool while trying to build. If this matters?

closed 2020-07-21 by garrett928