CANopenNode/CANopenNode #213: using custom OD with canopend

garrett928, 2020-07-21:
Hi, I've been using `canopend` and `canopencomm` the past few days and they have been working well. I would like to start using my own object dictionary when creating a `canopend` device. However, I can't really tell from the documentation or the source code how to go about this. 

I have generated an EDS file, and CO_OD.c and CO_OD.h using [libedssharp](https://github.com/robincornelius/libedssharp). However, I'm not sure what to do with them now. The doc pages show the `-s` and `-a`  flags for `canopend` to store od storage and od automatic storage. But I don't think this is what I want? 

In the source for `canopend` I see the CO_OD.c and CO_OD.h . But, I don't think replacing those and recompiling is the best way to do this. Am I missing something?

CANopenNode, 2020-07-22:
Object Dictionary part is now in the middle of renovation, see #162  
CANopenSocket currently uses outdated code for `canopend`. Please use CANopenNode only. Files in `example` directory should be replaced with your own (CO_OD.c and CO_OD.h generated by libedssharp.)
Storage is additional feature, which can be used to preserve some variables from Object Dictionary.

If you need custom behavior of CANopen device in Linux, you should make your own `socketCAN/CO_main_basic.c` file.
You can also communicate with current canopend via socket, similar as `canopencomm` do.

martinwag, 2020-07-22:
>  But, I don't think replacing those and recompiling is the best way to do this. Am I missing something?

CANopenNode is made to work on small embedded targets. There is no way to do dynamic loading on small devices, so files are compiled in.

CANopenNode, 2020-07-22:
> There is no way to do dynamic loading on small devices, so files are compiled in.

This may change in new OD, because there will be separated OD interface.

garrett928, 2020-07-22:
>Object Dictionary part is now in the middle of renovation, see #162

Yeah, I read through that post. That was part of my confusion. I wasn't sure what state currently things are in.

>CANopenSocket currently uses outdated code for canopend. Please use CANopenNode only.

Ok, got it! Should CANopenSocket just be used for `canopencomm` then?

> CANopenNode is made to work on small embedded targets. There is no way to do dynamic loading on small devices, so files are compiled in.

Yeah this makes sense. I knew if i wanted to use this on a PIC or something the OD would be compiled in. But I didnt know if canopend allowed the loading of different ODs though the terminal.

Thank you guys for the clarification!

CANopenNode, 2020-12-28:
CANopenNode  and libedssharp were updated quite a lot recently.

closed 2020-12-28 by CANopenNode