CANopenNode/CANopenNode #214: Reset communication will result into resetting everything

rubenverhoef, 2020-07-29:
I want to implement the reset communication for my platform.
What I have done is to re-initialize the CANopen stack with `CO_CANopenInit();`.

The problem is that this will result in deleting all configuration of the callback functions and I think more like that. I have configured these things from within my application, on several places.

I expected that there was an method to only re-initialize things like state-machines, buffers, hardware, (static configuration). And let the dynamic configuration unchanged.

CANopenNode, 2020-08-02:
I agree. Currently all those "CO_xyz_init()" functions initializes or set to zero everything. And that is actually not necessary, because memory for all those objects is allocated with "calloc()", which sets them to zero anyway.

I think, we can change that paradigm and leave some variables or callbacks unchanged by "CO_xyz_init()" functions. I think, it is only necessary to remove some "set-to-null" lines from "init" functions and document that.

Can you prepare a pull request for what you think it is necessary?

CANopenNode, 2020-12-28:
Current approach in CO_CANopenInit() is to reconfigure everything, it clears also all callback functions. If there is no strong reason, I have no plans to change this behavior.

closed 2020-12-28 by CANopenNode