CANopenNode/CANopenNode #216: Failed compiling in Qt C++ 11

reyanvaldes, 2020-08-03:
Hello thanks for nice work.
I am trying to use it in a very simple Qt C++ 11 program, but it failed when compiling. I am not sure what I am missing, is it compiler directive? 

Please could you help me to figure out. Thanks

T
[image]
Thanks for any support

seclorum, 2020-09-08:
"301/CO_driver.h" pulls in the header "CO_driver_target.h" - which is duplicated in the CANOpenNode sources, in the "example/" directly and the "socketCAN" directory as well.  

It seems you are hitting the "example/" version of this "CO_driver_target.h" file, which does not declare this field in the CO_CANmodule_t struct - *as it is just an example*, you have to provide your own driver - and you can easily do this, if you are indeed using the SocketCAN module, by simply removing "examples/" from your projects include path and ensuring that you *only* get the files from the "SocketCAN/" directory - which builds clean for me, out of the box, and should for you as well.

The other errors are happening for the same reason. Do NOT include the example/ folder in your project...

CANopenNode, 2020-12-28:
I think, this was answered. Otherwise see also https://github.com/CANopenNode/CANopenNode/blob/master/doc/deviceSupport.md

closed 2020-12-28 by CANopenNode