CANopenNode/CANopenNode #222: PDO buffers init problem

Norcio, 2020-08-13:
Some ID's in TPDO, and RPDO are not initialized properly (set to 0) in buffer because of datalenght=0, please check debug output. For tests I use [CO_OD.zip](https://github.com/CANopenNode/CANopenNode/files/5071004/CO_OD.zip) generated from [xdd.zip](https://github.com/CANopenNode/CANopenNode/files/5071020/xdd.zip) with latest libedssharp [image]
It looks like parsing problem but I couldn't find any issues in generated CO_OD files. Any suggestions? Thanks

[image]

CANopenNode, 2020-08-13:
I think, mapped variables in your PDOs are not enabled for PDO mapping. You will have to re-edit CO_OD with libedssharp and enable PDO mapping for object 0x6100, etc.

Norcio, 2020-08-14:
Thank you. It was the cause.

closed 2020-08-14 by Norcio