CANopenNode/CANopenNode #223: RPDO receive problem

Norcio, 2020-08-19:
I'm trying to run simple communication example with PDO. I have two nodes 0xA and 0xB. 0xA sending TPDO 0x18A with two bytes of random data every 1s. Node 0xB receive it on RPDO 0x1400 but there is no data in mapped variables OD_writeOutput8Bit[0]  and OD_writeOutput8Bit[1]. I'm using sligty modified IO.eds from example. I checked so far:
- RPDO transmission type is 255 - data should be processed imidiattely  
- Both nodes are in operational state = 5
- PDO map is enabled for Write output data 8 bit object,it's located in RAM, access=rw
- 0x1600 I think is mapped properly, 0x62000108 and 0x62000208
- CO_process_RPDO runs cyclically every 1ms
- In CANreceive I get msgMatched=True and callback should be fired

What I'm missing ? Thank you for any suggestions.
[IOxdd.zip](https://github.com/CANopenNode/CANopenNode/files/5096917/IOxdd.zip)

[image]

CANopenNode, 2020-08-21:
I don't know, what can be the problem.

Can you try "CANopen command interface" with your network, as described in https://github.com/CANopenNode/CANopenNode/blob/master/doc/gettingStarted.md ? Or some other CANopen configuration tool?
You can try to read RPDO communication parameter COB-ID, to see, if RPDO is enabled (if there is an error, RPDO is disabled). You can also try to configure PDOs dynamically in running network. In case of writing wrong value you get SDO abort message.

Norcio, 2020-08-24:
Problem was in receive driver function. CO_CANrxMsg_t was passed to callback instead of can_message_t. Thank you.

closed 2020-08-24 by Norcio