CANopenNode/CANopenNode #224: SDO client upload size check and abort

heliochronix, 2020-08-25:
Hello,

I've been working on trying to get the new SDO client working, but ran into some things I'm unclear about.

As far as I can tell, there doesn't seem to be an effective way of aborting an SDO client upload in progress. Specifically, in the context of a DOMAIN data type, the size indicated by an SDO server isn't known until `CO_SDOclientUpload()` is called. However, if the size indicated cannot be handled by the SDO client, there's no way to initiate an abort without simply letting the SDO transfer time out.

Am I missing something here, or should we add a `bool_t abort` parameter to the `CO_SDOclientUpload()` to allow a client upload to initiate an abort as well?

On another note, is there some place I can ask other questions I have regarding SDO client operation? There are several other things that I'm not clear about, but I don't want to spam GitHub Issues with potentially dumb questions.

CANopenNode, 2020-08-27:
Hello,

I think, abort argument should be added to CO_SDOclientUpload() as it is in CO_SDOclientDownload().
There may still be some similar inconsistencies in SDO client.

Don't worry about asking questions here, even if they are potentially dumb. Maybe documentation needs some additional clarification.

There is also a fresh slack workspace, not much there yet, but anyway:
https://join.slack.com/t/canopennode/shared_invite/zt-ghjhfg12-qGQNWvSEmhqcSedhFaLpuA

heliochronix, 2020-08-27:
That seems logical. Should I keep this issue open as a means of tracking that issue?

Awesome, a slack workspace is a great way to discuss some of this stuff. I've joined it.

CANopenNode, 2020-12-28:
I think, this issue with SDO client is solved in version 2.0 and new 4.0.

closed 2020-12-28 by CANopenNode