CANopenNode/CANopenNode #226: Time has a direct dependency to OD

gotocoffee1, 2020-09-18:
The time header contains the OD header for defining TIME_OF_DAY, which is not optimal if you want to compile the stack without a specific OD.  I would suggest to place the TIME_OF_DAY definition somewhere else (with a little wink to #148) to remove this dependency.

CANopenNode, 2020-09-29:
I didn't put much effort in CO_time yet. I plan to do the updates with new OD.

geoffrey-vl, 2020-11-25:
I found the same problem in current master (V2).

What about added the content underneath to a separate header file, specific for data types?

**CO_datatypes.h**
```
#ifndef CO_DATATYPES_H
#define CO_DATATYPES_H
/*******************************************************************************
   CANopen DATA DYPES
*******************************************************************************/
   typedef uint8_t      UNSIGNED8;
   typedef uint16_t     UNSIGNED16;
   typedef uint32_t     UNSIGNED32;
   typedef uint64_t     UNSIGNED64;
   typedef int8_t       INTEGER8;
   typedef int16_t      INTEGER16;
   typedef int32_t      INTEGER32;
   typedef int64_t      INTEGER64;
   typedef float32_t    REAL32;
   typedef float64_t    REAL64;
   typedef char_t       VISIBLE_STRING;
   typedef oChar_t      OCTET_STRING;
   typedef domain_t     DOMAIN;

#ifndef timeOfDay_t
    typedef union {
        unsigned long long ullValue;
        struct {
            unsigned long ms:28;
            unsigned reserved:4;
            unsigned days:16;
            unsigned reserved2:16;
        };
    }timeOfDay_t;
#endif

    typedef timeOfDay_t TIME_OF_DAY;
    typedef timeOfDay_t TIME_DIFFERENCE;
#endif
```

From what I can tell, the datatypes are not part of the EDS file, and neither does libedssharp allow to set those values.
So, at first sight for me it doesn't make much sense to have it be generated. It seems rather platform specific. We could include a default CO_datatypes.h easily because mostly people don't edit that part I'd guess? And if they do, at least when the datatypes are outside the OD updating the OD from lededssharp will not undo their platform specific changes.
Or am I missing the point entirely? 😨

CANopenNode, 2020-11-25:
I don't know, if TIME_OF_DAY is actually used and working somewhere in OD. Probably timeOfDay_t can just be removed.

In newOD UNSIGNED8 etc. are not used. For 6-byte variables there is uint8_t array.

geoffrey-vl, 2020-11-25:
If I remove `timeOfDay_t ` entirely I get `CO_TIME.h:119:5: error: unknown type name ‘TIME_OF_DAY’`.

So it is required but only for CO_TIME, what approach do you recommend for V2?
A. have `timeOfDay_t ` defined in `CO_TIME.h`
B. have a `CO_datatypes.h` as demonstrated above?

CANopenNode, 2020-11-25:
I missed that. I think, option A is OK.
I think, timeOfDay_t is used exclusively in CO_TIME.h and CO_TIME.c and not in Object Dictionary anyway.

gotocoffee1, 2020-11-26:
I think this is a good solution aspecally with the new handling of the uncommon types as generic byte arrays.

closed 2020-11-26 by gotocoffee1