CANopenNode/CANopenNode #228: CANopenNode CO_OD provided example source

AnanKSDM, 2020-09-24:
Hello,
Apologies for posting these questions here, but not sure where I can make it.
I'm new to CANopenNode and to CANOpen in general and was assigned a task to replace some 3rd party Master Controller (no source code) to control existing Slave Nodes (multiple nodes) per our needs. I came across this promising project and would like few clarifications on usage please.
1. Where can I find the source input file (EDS/XML...) that was used to generate the example/CO_OD.[h/c] files. Not sure that whatever EDS/XML provided in example folder were the latest source. I'm using OD Editor libedssharp per the suggested in this project. Tried to re-generate these CO_OD(c/h) files, but always I get different one from what is already provided. Doing so, I was trying to learn and feel comfortable with it, before trying to implement the required.
2. If I have only the OD of the slave, what is the best way to define the Master OD if I wish to implement the Master side mainly?

I really appreciate the help in this as we are short in time.

Thanks you very much for the support,
A.

martinwag, 2020-09-24:
1. example/_project.xml. Output might be different as some changes in export where introduced during libedssharp development.
2. check master device manufacturer homepage for eds or xdd description file, use example from 1. as starting point and then add what you need.

AnanKSDM, 2020-09-24:
Thanks for the prompt answer. Regarding the Master Device Manufacturer, I couldn't find it, so I presume that I need to invent it.
Is there any documentation that can help me understand what options in the libedssharp tool defines the macro "#define CO_NO_NMT_MASTER"

martinwag, 2020-09-24:
You can of course also get some inspiration from looking at other master device manufacturers ODs...
NMT master should be enabled by setting OD 0x1f80 bit0 "is NMT master" to TRUE. For master you should also clear bit 2 (master is normally self starting).

AnanKSDM, 2020-09-24:
Thanks for the help. I consider this issue closed.
Much appreciated.

closed 2020-09-24 by AnanKSDM