CANopenNode/CANopenNode #236: TIME object cycle Period

jdascenzio, 2020-10-08:
I want to use TIME object as producer but I don't see any way to set cycle period.
`CO_CANopenInit` call `CO_TIME_init` and set `TIMECyclePeriod` to 0. Currently, I need to change the call of `CO_TIME_init` to set the cycle period I want. Is there a way to do this that I haven't seen?

L0PiTaL, 2020-11-26:
I am in the same situation as you. The library does not change TIME->periodTime nor TIME->periodTimeoutTime from other place than CO_TIME_init.
My guess is that this setting is application specific, and should be set up by the application outside the library, by accessing directly the CO object. I.e. setting a timeout of 1sec:

CO->TIME->periodTime = 1000
CO->TIME->periodTimeoutTime = 1000 * 1.5

And the library will begin producing TIME objects

JuPrgn, 2020-11-26:
No I implemented a basic Time producer and tested this as I am not using it on my applications. Feel free to propose pull request if you have some improvments.

JuPrgn, 2020-11-26:
Sorry for the delay I missed your issue notification ...

L0PiTaL, 2020-11-26:
> No I implemented a basic Time producer and tested this as I am not using it on my applications.

Sorry, I don't understand your sentence.
Do you tested it and it did not work?
Or do you tested it and it did work?

Just enabling the OD object 0x1012 to be a producer does not generate any TIME message. This was the original issue.

The solution I proposed is to have the application initialise the CO->TIME variables to a timeout suitable for the application purposes, which is why I think it may be left untouched, so the applicaiton can set it up.
This can be done after CO_CANopenInit.

With this, it correctly generates TIME messages.

I don't know if this behaviour was intended or was a bug. My feeling is that it was intended, so no pull request would be needed.

JuPrgn, 2020-11-26:
I tested TIME producer and consumer on v1.3 branch (when I added this functionnality) and it was working pretty well. But as I don't use TIME producer on my applications you may have some improvments to add if you need to change period outisde initialisations... 
Please note that a lot of changes to CANopenNode since that contribution could have been affected TIME, I did not tested it again with master branch but I am currently starting a new project with this branch and will re test this soon.

L0PiTaL, 2020-11-26:
Ok.
I was talking about the master branch. Don't know about v1.3, since I have not used it.

In the master branch, the app has to do the changes I stated. Otherwise it will not generate TIME messages.
I thought it was intended, but from your comment, it seems to be actually a bug.

CANopenNode, 2021-01-25:
I have just pushed CO_time.h/c updated to newOD branch. Time producer in Linux, if enabled, sends correct time-stamp message (from system RTC). Internal variables `ms` and `days` are precisely updated on each CANopen_process() pass. Consumer also updates them on reception of the time-stamp message.

Enabling/disabling producer/consumer works dynamically.

Take a look in the commit https://github.com/CANopenNode/CANopenNode/commit/464a856a3f13417769f0f36eba3ce2901df381f1

closed 2021-02-08 by CANopenNode