CANopenNode/CANopenNode #239: F28069M (C2000), what does it take to have it working?

riscy00, 2020-10-12:
TMS320F28069M (C2000), what does it take to have it working?

geoffrey-vl, 2020-10-12:
You'd need to write your own driver that brings support for these TI chips. Read the documents for info about porting this library for your board.

rubenverhoef, 2020-10-28:
You also have to change the memory related functions, like `CO_memcpySwap2`. This because the C2000 processor doesn't support `uint8_t`, it's smallest value is 16 bit.

CANopenNode, 2020-10-28:
> This because the C2000 processor doesn't support uint8_t, it's **smallest value is 16 bit**.

There may be other incompatibilities in the stack for such chip. Probably they can be fixed, but there must be someone with knowledge about "smallest value is 16 bit" problem and testing capabilities.

martinwag, 2020-10-28:
We've had this some time ago... https://github.com/CANopenNode/CANopenNode/pull/186

CANopenNode, 2020-12-28:
I'm closing this issue for now. If there will be a real interest for this chips, we can try to support them too.

closed 2020-12-28 by CANopenNode

kaushel, 2023-06-02:
> 

Do you have any updates on this? I've found a demo for 28335 but it's being done with SYS BIOS (TI's RTOS). I'm looking into a bare metal implementation.