CANopenNode/CANopenNode #242: SDO Block Transfer

gotocoffee1, 2020-10-28:
Hallo,
I have a problem. I want to to Object 0x1021 to store my eds file directly in the firmware device. To upload the file I use the follwing code:
```c
const unsigned long eds_file_length = 0x00010A32;
const unsigned char eds_file[0x00010A32] = {};
static CO_SDO_abortCode_t CO_ODF_eds_file(CO_ODF_arg_t *ODF_arg)
{
    if (ODF_arg->reading)
    {
        if (ODF_arg->firstSegment)
            ODF_arg->dataLengthTotal = eds_file_length;

    	uint32_t remaining = ODF_arg->dataLengthTotal - ODF_arg->offset;

        if (remaining <= ODF_arg->dataLength)
        {
            ODF_arg->dataLength = remaining;
            ODF_arg->lastSegment = true;
        }
        else
            ODF_arg->lastSegment = false;

        

        memcpy(ODF_arg->data, &eds_file[ODF_arg->offset], ODF_arg->dataLength);

    }
    return CO_SDO_AB_NONE;
}
```
In segmented mode this works fine but in block transfer I get either :
`[0] ERROR:0x06070012 #Data type does not match, length of service parameter too high.`
or an CRC error when I do not set the explicit size. Could someone please check if the code is theoretically suitable for block transfer and if so whether the same error occurs in practice as with me?

martinwag, 2020-10-28:
This is what I use (but download...):

```
CO_SDO_abortCode_t Canopen::program_data_callback(CO_ODF_arg_t *p_odf_arg)
{
  ...

  if (p_odf_arg->firstSegment == true) {
    flash_status = flash.start_block(p_odf_arg->dataLengthTotal);
    ...
  }

  if (p_odf_arg->dataLength != 0) {
    flash_status = flash.add_chunk(p_odf_arg->data, p_odf_arg->dataLength);
   ...
  }

  if (p_odf_arg->lastSegment == true) {
    ....
  }

  return CO_SDO_AB_NONE;
}
```

JuPrgn, 2020-10-28:
Hi I think you should not change dataLength as this is the reader length but you should fill "dataLength" bytes on ODF_arg->data[].

CANopenNode, 2020-10-29:
I don't use block transfer much, so don't have real experiences. Few years ago I was testing it and it seemed OK. Here is the example code, which should be used for domain:

https://github.com/CANopenNode/CANopenPIC/blob/master/example_PIC32_ex16_IO.X/appl_PIC32_explorer16_IO.c#L190

I plan to do heavy tests on SDO in the near future. Latest SDO server and client are quite renewed. But I will run tests also on older versions.

geoffrey-vl, 2020-10-29:
I've used it within the 1.x branch and it was working using the CANopenSocket tools. I was testing firmware updates over CANopen this way.

gotocoffee1, 2020-10-29:
Hello erveryone,
thanks for the feedback 👍 . I tested a bit more and also tried your @CANopenNode implementaion code. It seems that the error does not always occure. The frequency depends on the data size. The more data the more often it fails. e.g.

  * size 0x500 kinda never fails
  * size  0x1800 sometimes it fails somethimes not (<-- spooky)
  * size 0x10A32 **almost** always fail

Maybe an connection to #123?

The culprit does not have to be the server I am use the SDO client (as Gateway ) to upload the file.

CANopenNode, 2020-10-29:
Can you attach here some files with candump? For example with size 0x1800, one OK and one failed.

gotocoffee1, 2020-10-29:
I attached the can dump + gtw dump. Line 772 in the failing CANdump is suspicious. Seems like the client is sending an additional message...

[gtw_fail.txt](https://github.com/CANopenNode/CANopenNode/files/5457813/gtw_fail.txt)
[gtw_working.txt](https://github.com/CANopenNode/CANopenNode/files/5457814/gtw_working.txt)
[candump_fail.txt](https://github.com/CANopenNode/CANopenNode/files/5457815/candump_fail.txt)
[candump_working.txt](https://github.com/CANopenNode/CANopenNode/files/5457817/candump_working.txt)

CANopenNode, 2020-10-30:
Sorry for my late answer.
I will try to find the bug. It is probably an old one. 
Can you please tell, which version of the SDO server are you using (commit number). And which gateway?

gotocoffee1, 2020-10-30:
No problem. I used this versions:

server: https://github.com/CANopenNode/CANopenNode/tree/c075bdff3c289107eb03bfca375764b66ef143ad
gtw: https://github.com/CANopenNode/CANopenNode/tree/9446aa5792a2d8e0023558c6191a48675cefae44

They are relatively up-to-date. I tried to debug it, but could not isolate the problem. Thank you for your efforts.

CANopenNode, 2020-10-30:
Yes, server is the last one. Can you also update the gtw. Between commits I corrected thread and epoll related stuff, quite a large internal change. Maybe there could be the problem.

gotocoffee1, 2020-11-26:
Sorry for my late response on this issue. I haven't got the time to test issue. I will update you as soon as I can test it.

CANopenNode, 2020-11-26:
And now I'm finished with libedssharp and next will prepare tests for heavy load on SDO communication.

CANopenNode, 2020-12-28:
I recently pushed updates to master and newOD branches, see https://github.com/CANopenNode/CANopenNode/issues/162#issuecomment-751724920

SDO client, gateway and socketCAN driver had some problems, which I fixed now. I successfully and properly transferred 10GB large file over virtual CAN in both directions simultaneously. See https://github.com/CANopenNode/CANopenSocket/tree/master/examples/basicDevice

SDO client and gateway were tested in V2.0 and new V4.0.
But SDO server was tested only in newOD branch (v4.0). Currently I don't have appropriate testing setup for old SDO server and old Object Dictionary.

So currently we have:
- new Object Dictionary approach, rewritten and tested SDO server in CANopenNode version 4.0
- rewritten and tested SDO client in CANopenNode version 2.0 and above
- no additional information for old Object Dictionary and old SDO server.

If you will make any further tests, please let me know.

CANopenNode, 2021-01-06:
This is the problem of Linux driver. Kernel don't have enough space for additional CAN messages. (Block transfer sends max 127 messages, then waits for response.) I didn't see this error yet and I will try to handle it properly. Maybe @martinwag has some additional idea. The problem happens in socketCAN/CO_driver.c:
```
n = send(interface->fd, buffer, CAN_MTU, MSG_DONTWAIT);
...
else if (errno == ENOBUFS) {
```

Can you check message from your program startup, size of buffer. I have:
`./canopend[360211]: CAN Interface "vcan0" Buffer set to 477 messages (212992 Bytes)`

One solution could be, if some delay is added. For example in CO_SDOserver.c inside
block `case CO_SDO_ST_UPLOAD_BLK_SUBBLOCK_SREQ: {`
line (1574): `*timerNext_us = 0;`
change to something larger than zero.

However, I will try to find solution to handle this error properly, without breaking the communication.

gotocoffee1, 2021-01-06:
I have the same `CAN Interface "can0" Buffer set to 477 messages (212992 bytes)`.

I am not sure if this error is the cause of the problem on my device, because there I send messages in Tx interrupt when the buffer is already full. I also don't get a Tx overflow error.
But there I am still on V2.0 branch....

CANopenNode, 2021-01-18:
I made some quick tests, with different CAN interfaces, including PCAN. There was the same problem, but not always and not with all interfaces. I have plan to make a systematic tests with different combinations of hardware with Linux and microcontrollers.

But I would like to finish newOD first, at least HB consumer and PDO, which are still missing. Then I plan systematic communication tests with all versions, probably v1.3 also.

martinwag, 2021-01-25:
Hi,

> CAN Interface "can0" Buffer set to 477 messages (212992 bytes)

is the size of **socket** receive buffer. You should get the transmit **socket** buffer size with

```
    getsockopt(interface->fd, SOL_SOCKET, SO_SNDBUF, (void *)&bytes, &sLen);
    if (sLen == sizeof(bytes)) {
        log_printf(LOG_INFO, CAN_SOCKET_BUF_SIZE, interface->ifName,
                   bytes / 446, bytes);
    }
```
where the other message is printed.

However, if you already have the printout enabled, the driver should print a line everytime it drops a tx message. At least in v1.3 release, I didn't check the newer ones. Be aware that the epoll() ... stuff is very fiddely.

I have a few of those PCAN interfaces too, and while being great for debugging on the PC, the USB one seems to be a bit buggy. So, just re-test on the hardware that you are actually intend to use for production. If that is PC-PCAN, get the PCI-E plug in card.

CANopenNode, 2021-02-08:
Finally SDO block transfer was systematically tested with different interfaces on real wires.

There was actually a problem with Linux socketCAN driver with CO_CANsend() function. When there was transmitted sequence of data (127 CAN messages in sequence), it happend, that Linux system socket for CAN tx was full and message was dropped. SDO block transfer is robust, so this was not always the problem. Linux driver was unclear at this point from the beginning.

CANopenNode uses CO_CANsend() without checking it's return status, which is intended. If CO_CANsend() fails in the moment of call, missed message is marked with bufferFull flag (for each CANopen object separately). Missed message is sent later, after CANtx becomes available. In microcontrollers this is usually implemented with interrupts, but this was not possible in Linux.

However, now CO_CANsend() in Linux works according to the original principle: missed CANtx messages are marked with bufferFull flag. Then CO_CANmodule_process() tries to resend them automatically.

SDO server and SDO client wait, if bufferFull flag is set, so transmitting SDO sub-block now works without errors and without time delays.

I tested different CAN interfaces: PCAN-USB and CPC-USB works without detected errors. But cheap USBtin (slcand) doesn't work well with block transfer (segmented transfer works always).

I tested newOD branch in all aspects. Current master (v2.0) has almost the same driver, epoll interface (which was widely renewed in v2.0) and SDO client and that was tested too. SDO server in v2.0 is quite the same as in v1.3 and was not tested.

There are automatic tests for running CANopen devices, for test results see: https://github.com/CANopenNode/CANopenSocket/blob/master/test/test_report.md

gotocoffee1, 2021-02-11:
I can confirm that. I think my initial problem was caused by a problem in the V2.0 server implementation. But since the V4.0 version is working properly, this issue is solved.
Thank you very much for your effort.

closed 2021-02-11 by gotocoffee1