CANopenNode/CANopenNode #244: Examples (adding to show different functionalities)

edesta-be, 2020-11-23:
Good evening,
I was looking through your stack, and realised it would be really useful if there were integration examples that are easy to follow that shows the different aspects of the system. 
I.E: an NMT master, an SDO upload/download, a PDO mapping. I think just going through this and the docs on doxygen it would be a nice starting place to see something basic of that sort. I noticed you had a whole different repo for a board, so even just adding to that would be awesome.
But just generally I think I am more interested in this node trying to send commands to the other nodes. It might already exist if so great please point me to it. Otherwise i think it would be a great add.

martinwag, 2020-11-24:
CANopen master implementation: https://github.com/CANopenNode/CANopenSocket

CANopenNode, 2020-11-24:
I have to update Martin's information: CANopenSocket is currently a little obsolete. I have plans with it in a way, suggested by edesta-be. But currently there is nothing more, than it is in CANopenNode/doc/gettingStarted.md.

However, for getting started with CANopen and getting familiar with NMT, SDO, PDO, etc I recommend a little old-fashioned approach: reading a book, for example https://www.canopenbook.com/. CiA301 standard also has all the information, but it is easier to read and understand the book.

edesta-be, 2020-11-24:
Sweet thank you! Ill take a look through the book!

CANopenNode, 2020-12-28:
Here is example, which may be quite useful:
https://github.com/CANopenNode/CANopenSocket/tree/master/examples/basicDevice

More examples will follow.

closed 2020-12-28 by CANopenNode