CANopenNode/CANopenNode #245: State of current master (V2)

geoffrey-vl, 2020-11-25:
Hi, I've been using the V1.3 version of this stack on some microcontrollers boards.
However, a lot of work on V2 has been in done over  the past few months.
Is there going to be release of V2? Is it "production ready"?

I'm planning to use it in a new application, and I prefer to jump aboard the V2 release to avoid extensive refactoring afterwards when newOD becomes production ready. However, newOD still needs to some development so I guess it is still few months from production ready.

CANopenNode, 2020-11-25:
NewOD has quite final shape in parts, which are finished. Unfinished are: HB consumer, time, sync, PDO, trace and SRDO.
SDO and Object dictionary has final shape, also libedssharp is finished (I sent pull request few hours ago). Object dictionary is in the foundation of everything, so newOD is quite a big step. OD usage should me more clear, flexible and powerful now. Example usage is here: https://github.com/CANopenNode/CANopenNode/blob/master/doc/objectDictionary.md#example-usage . I will try to make at least missing PDO soon.

I think, V2 is production ready. I have to make some more tests of block transfer in all versions. I think, there is no much difference from V1.3, as it may seems. I will make a release for V2. You can compare differences between v1.3 and V2.0 in CANopenPIC, for example.

geoffrey-vl, 2020-11-25:
Thx for elaborating things. Regarding newOD: I'll be using SDO and PDO so it may come a bit late, however I'm looking forward for the V2 release.

CANopenNode, 2020-12-28:
Version 2 has final shape now, but it is interim version somehow. I don't plan to make a release with it, but will probably have own branch. See also https://github.com/CANopenNode/CANopenNode/issues/162#issuecomment-751724920

geoffrey-vl, 2020-12-28:
> Version 2 has final shape now, but it is interim version somehow. I don't plan to make a release with it, but will probably have own branch. See also [#162 (comment)](https://github.com/CANopenNode/CANopenNode/issues/162#issuecomment-751724920)

Well, previously you stated that "I think, V2 is production ready", and "I will make a release for V2".
So, I guess people took that for granted and started using current v2, at least I did.
If you ask me, I'd say that there are already 2 releases of V2:

```
$ git tag 2.0 774cc6ee1f966b741c51afaf3b72c9691d9e8fb3
```

```
$ git tag 2.1 0c000b59242e006630fe326eb3ef5a8b78ada38f
```

I think it's okay to use regular tagging, it doesn't need to be 'github release'.

Furthermore, I'm convinced that V4 is not yet finished/completed and certainly not production ready. People looking into adapting CANopenNode will not (and have not) sit out and wait indefinitely. I know "some feature branch" involved a lot of work, and it will be great *once it's there*. But I think it also not a bad thing to recognize current master is at a good starting point for any new project started anywhere between 1.3 branch and newOD branch.

I can also encourage branching V2 once you start merging newOD into master. It will allow for bugfixes to land for anyone using that release.

CANopenNode, 2020-12-29:
Thank you for your opinion. I certainly like people to have as low problems as possible.

With "not planning a release" I was thinking "not planning a github release". 

I'm not sure where to put the plain tag for v2.0. There is some mess in git history and is not clear, when v2 started.

Some explanation about v4.0:
v4.0 tag is at the first commit, splitting from master branch (pre-release). I used v4.0, because in some very old editions there is already CANopenNode v3.

This "production ready" may be a risky term. I try to make code reliable, but I can not offer guarantee. Latest commit in current master branch is certainly more "production ready" than all previous, especially for large data transfer. But I still can not guarantee for parts, which originate from old versions, like SDO server. This is old code, not much changed, but also without additional tests. 

I'm not sure, how is with "production readiness"  in v1.3.  @martinwag made quite a detailed code review, so there is some degree of "production readiness". In specific commit he use inside his specific application. After that (inside v1.3) there were some additional commits, some bugixes and some improvements. And with that there is always possibility for breakage.

My philosophy is: v1.3 is outdated, especially Object dictionary implementation. But it is not a good idea to put a new engine in an old car. So I started with re-organizing all the rest of the code, except object dictionary. I'm ambitious, so I wanted to use everything good from the code developed so far. It is almost a year, since a I started. Gateway was hard and the rest of the project is also quite extensive. And it must still be possible to put CANopenNode intoto a small microcontroller.

Alternative would be to start with a new project, but this is not good for opensource projects. We have git anyway.
For v2.0 (or all commits inside master branch) I tried to be always as operational as v1.3. But v2 will always be interim version.

I will certainly keep own branch for v2. I don't like to put too much tags to v2, because they won't mean much. I would put tag v2.0 by latest commit in the current master.

geoffrey-vl, 2020-12-29:
Ok, that sounds reasonable enough for me. Regarding V2.0 tagging, for me its also OK of latest master commit marks 2.0.
For information: I referred to '774cc6ee1f966b741c51afaf3b72c9691d9e8fb3' because that's the one I'm currently working with, I haven't been testing with the latest commit in master.

There are some other topics that you highlight here, such as "proving code quality", "testing production readiness", "devotion and time spend to this project" that certainly deserve their own Issue to discuss. I've also been thinking about how the stack could benefit from commercial support, maybe that also deserves its own spot because you could go very diverse: support plan, integration services, commercial license, etc. But lets keep that a apart from this issue.

Thanks!

CANopenNode, 2020-12-31:
Please read also this issue: #259

CANopenNode, 2021-02-08:
Current master (v2) received some updates after testing SDO block transfer. There was problem with Linux driver, which is fixed now in both branches (v2 and v4). See also https://github.com/CANopenNode/CANopenNode/issues/242#issuecomment-775292186

hab-0, 2021-02-16:
Which is the maximum data size the SDO block transfer can handle/transfer in CANopenNode?
Is the CANopen block transfer mode supporting transfer sizes in the range of KB/MB/GB?

Thanks!

CANopenNode, 2021-02-18:
In newOD branch block transfer has unlimited data size (4GB, if data size is indicated). I tested 10GB random data on virtual CAN - few hours, it worked correctly.

For v2 (current master branch), I can't say anything for SDO server, but SDO client should work the same as in newOD branch.

CANopenNode, 2021-03-18:
I will keep current "master" branch for some time (I plan one month) with version 2.0. I have pushed branch "v2.0-master", which is for now the same as "master".

After that I plan to move "master" (default) branch to the place of "newOD" branch. So, I think, there will be nothing broken. To keep the old branch just use `git checkout v2.0-master`. Pull requests to V2.0 will still be possible.

CANopenNode, 2021-05-17:
Master branch has been moved to the latest v4.0.

To keep the v2.0 branch:

    git fetch
    git checkout v2.0-master

If you are on newOD branch and want to update to the latest v4.0, delete old master branch first:

    git fetch
    git checkout newOD
    git branch -D master
    git checkout master

CANopenNode, 2023-03-28:
V2.0 is something between the v1.3 and last version, I think, it is not used much. V1.3 is still used by some older devices. For new design I recommend the latest version.

closed 2023-03-28 by CANopenNode