CANopenNode/CANopenNode #247: Compilation error because unable to locate CO_NO_TIME in CO_OD.h

rohitsay89, 2020-11-25:
Hi, I just created a new xdd, eds and CO_OD.c, CO_OD.h file from libedssharp Object Dictionary Editor v0.8-0-gb60f4eb. When I try to compile the application with CANopenNode stack, I am getting the following error:

`../Drivers/CANopenNode/CANopen.c:91:27: error: 'CO_NO_TIME' undeclared here (not in a function); did you mean 'CO_NO_NMT'?
                           CO_NO_TIME        + \
                           ^`

In my previous compilations I was using the old CO_OD.c/h files which are present in the example folder of the repository and that was working fine. 

Please suggest me any fix 
Thanks,
Rohit

geoffrey-vl, 2020-11-26:
Most possible related to #246 and #226

fenugrec, 2020-12-27:
I've added a line
`#define CO_NO_TIME CO_NO_TS`
in my CO_driver_target.h, since my current settings in libedssharp define `CO_NO_TS` with (I think) the same meaning as CO_NO_TIME, i.e. `Associated objects: 1012, 1013` 

The naming seems to have changed in bba140f .
Note, I still haven't been able to compile after many hours work, I still have an unknown number of errors to look at, so this "fix" may be invalid.

JuPrgn, 2020-12-27:
@fenugrec  which Libedssharp version are you using ? You should get the latest auto build linked on the readme page. Which CANopenNode version are you using too ?

fenugrec, 2020-12-27:
@JuPrgn , I've tried many, many permutations, all with different failures. Currently I'm wrestling with libedssharp-0.8 (compiled from the 0.8 release archive).
Same thing with CANopenNode, I tried V4.0 , newOD, and currently 1.3 which seemed promising but I'm not sure anymore.

JuPrgn, 2020-12-27:
On my last project I am using CANopenNode V2 current master branch and latest auto build from Libedssharp without issue. You may have some things to configure but open an issue and I could help you from this point.

fenugrec, 2020-12-27:
@JuPrgn thanks, I already opened #257 where you already replied­. Unless the autobuild is compiled from a different branch / commit... what commit of CANopenNode are you refering to ? I can't see any tag or branch named "V2"  ?

JuPrgn, 2020-12-27:
You are right there is no V2 tag but the latest commit should be fine with Libedssharp latest auto build. Tomorrow I can post the different versions commit numbers I am using. On my side I commented some things related to SYNC but I could share and OD example.

fenugrec, 2020-12-27:
>  latest commit should be fine with Libedssharp latest auto build.
Last time I tried a few commits ago, that was absolutely not the case. Either CO_OD_entryRecord_t or  OD_obj_record_t was undefined depending on the selected exporter.

CANopenNode, 2020-12-28:
I have recently pushed quite large commit to master branch:
> Testing and multiple fixes were made in newOD branch. This commit includes
> relevant code from newOD branch, synchronized to this branch.
> Following modules were updated:
> - SDO client, CO_gateway_ascii, CO_fifo
> - socketCAN : CO_epoll_interface, CO_main_basic
> SDO client (with gateway and fifo) which was rewritten after v1.3 is now
> tested and works the same as in newOD branch. Same is socketCAN. Works as here:
> https://github.com/CANopenNode/CANopenSocket/tree/master/examples/basicDevice
> 
> SDO server is quite the same as in v1.3. I made no additional testing at
> this point.
> 
> There are some other updates:
> - CO_driver.h, CO_config.h: additional CO_CONFIG_DEBUG macro.
> - CO_driver.h: fixed some typos, added optional CO_errinfo() macro.
> - CO_NMT_Heartbeat: added CO_NMT_setInternalState().
> - documentation synchronized.
> - example/_project.xml updated, can be used as template with libedssharp.

README was updated. I tested also latest libedssharp. `example/_project.xml` is updated too and can be used as a reference or separate objects from it can be inserted into the custom project.

There is a strange behavior in libedssharp: if XDD(v1.1) project file is opened, it is not exported correctly into CANopenNode_Legacy source files (CO_OD.h/.c). But if you export the project file into old xml  format first and then into CO_OD.h/.c), everything works fine.

CANopenNode, 2020-12-31:
latest libedssharp is fixed now, CANopenNode_Legacy works well and XDD(v1.1) may be used for storing any project. It is not necessary to use old xml format for storage.

`example/_project.xml` should be used as a template for projects based on CANopenNode (<= v2). Libedssharp produces equivalent CO_OD.h/.c files as in example directory.

CO_NO_TIME works properly now.

CANopenNode, 2021-02-01:
I think, this issue is solved now. CO_time object has also been renewed in newOD branch.

closed 2021-02-01 by CANopenNode