CANopenNode/CANopenNode #252: NMT master auto operational

geoffrey-vl, 2020-12-03:
After reading about https://github.com/CANopenNode/CANopenNode/pull/231, I'm still wondering if there isn't something missing within the NMT master implementation. I notice that when the NMT master is HB consumer, and one of the guarded nodes fails and goes into pre-op state, the NMT master also goes into pre-op but stays there even if I manually upgrade the client back to operational state. The NMT master went to operational initially through setting bit 4 of OD 1F80.
It seems HB consumer only does part of the job, although it may well be what the standard defines (I don't own a copy of CIA302).

CANopenNode, 2020-12-04:
All information about 0x1F80 object, relevant for slaves are here: https://github.com/CANopenNode/CANopenNode#self-start.

Otherwise CiA302 contains many functionalities about NMT master. But none of them is currently included in CANopenNode, which only has capabilities for sending NMT master message.

NMT (and EM) in master branch is old and contains some non-standard parts (error behaviour, higher sub-indexes). NMT and EM in newOd branch are rewritten and allows more flexibility (they does not use "error behaviour" and 0x1F80 objects, manufacturer may implement them.)

CiA302 is quite complex and it would require own directory in CANopenNode. It is a missing feature. Currently is possible to send "manufacturer specific" NMT commands.

geoffrey-vl, 2020-12-04:
If I'm understanding well:
CiA302 is definitely something that can be improved in current master AND newOD, but doesn't get any treatments real soon. So I'd need to implement this myself, in manufacturer specific way.

What I don't understand is: _NMT and EM in newOd branch are rewritten and allows more flexibility (they does not use "error behaviour" and 0x1F80 objects, manufacturer may implement them)_.
Is that because you don't want anything complex Cia302 related to be within CANopenNode?

CANopenNode, 2020-12-04:
Yes, 302 is quite useful for masters and it really is missing in CANopenNode. I don't need it in my projects so currently I have no plans with it. Another important module, missing in the past, LSS 405, was contributed by mw. There were also contributions for 304, 306 (libedssharp), etc. My goal is to have things standardized, flexible(selectable, configurable) and maintained. 

Actually I didn't drop anything related to 302, because nothing related to it was actually implemented. That small part of 0x1F80 can easily be used from custom code. "error behavior" is mostly manufacturer specific anyway. Past custom solutions made only trouble to users, who had different requirements.

I have some little experience with real NMT master. It automatically sends resets, starts, based on some rules and state machine. As I know, some CANopen controllers implement it.

CANopenNode, 2020-12-04:
Here is a link to CO_NMT_control_t, which is passed to new CO_CANopenInit() and configures NMT slave:
https://github.com/CANopenNode/CANopenNode/blob/newOD/301/CO_NMT_Heartbeat.h#L125

NMT master is currently used only from gateway.
For example, it depends on application, NMT master message "reset comm" can be send, if operator presses the reset button (NMT slaves are set for self-starting).
Or some (simplified) procedure, based on bootups and heartbeats, can be implemented for automatic NMT commands...

geoffrey-vl, 2020-12-04:
At this stage I'm thinking of using HB node guarding combined with NMT master remote state control for individual specified nodes. I'd need some sort of in-memory table that lists all required nodes and their "wanted" vs "reported" states to manage that. There is no requirement for me to be 100% CANopen compatible, it's just that relying on services that have proven their duty is often the best choice. But I guess my question came one year too early  :)
I think that is clear now, that info should suffice to get me going. Thanks!

closed 2020-12-04 by geoffrey-vl

robincornelius, 2020-12-04:
What i routinely end up doing is something like :-

  switch(main_state)
    {
        case APP_STATE_BOOT:
            if(boottimer >= OD_timings.can_rst_comms_delay)
            {
                boottimer = 0;
                CO_sendNMTcommand(CO,CO_NMT_RESET_COMMUNICATION,0);
                main_state = APP_STATE_RST_COMMS;
            }
            break;

        case APP_STATE_RST_COMMS:
            if(boottimer >= OD_timings.can_start_me_delay)
            {
                //CO_sendNMTcommand(CO,CO_NMT_ENTER_OPERATIONAL,
CO->NMT->nodeId);
                main_state = APP_STATE_START_ME;
            }
            break;

        case APP_STATE_START_ME:

            if(boottimer >= OD_timings.can_start_slaves_delay)
            {
                boottimer = 0;
                main_state = APP_STATE_START_NODES;
            }
            break;

        case APP_STATE_START_NODES:
            if(boottimer >= OD_timings.can_app_run_delay)
            {
                CO_sendNMTcommand(CO,CO_NMT_ENTER_OPERATIONAL, 0);
                main_state = APP_STATE_SDO_GUARD_CONFIG;

            }
            break;
}

You can see there are a few unused states in there, but this is generally
useful for most networks i use. I set ALL nodes to start up in
preoperational. So after initial switch on all nodes sent their boot
messages, at this point there can be a race condition for the first node to
send its message so you often see a communications error, this is fine as
the very next thing we do is COMMUNICATION_RESET, this puts everything back
in a nice state, then after a delay i fire off  CO_NMT_ENTER_OPERATIONAL
which causes all nodes to exchange PDOs and start running. What i have
found is if you don't set things to preoperational its really hard to
control the initial flow of PDOs and things can lose state as they miss a
PDO they needed to see.

The above state machine could also be expanded to before starting the nodes
to check the node is present etc, but i've not found any of the other NMT
start up options useful for my cases

Robin