CANopenNode/CANopenNode #257: clarification of CANopenNode versions vs libedssharp code generator

fenugrec, 2020-12-23:
Hi,
I've been having serious problems trying to find a combination of versions that works and still haven't managed.

Here's a list of my attempts with Libedssharp-0.8 @ g6f67539
```
*** CANopenNode V4.0 (f73025e)

_LEGACY exporter , compile with -DCO_VERSION_MAJOR=4 is  #if CO_VERSION_MAJOR < 4
_LEGACY exporter , compile with -DCO_VERSION_MAJOR=2 : fails because CO_OD_entryRecord_t is undefined
_V4 exporter, compile with -DCO_VERSION_MAJOR=4 : fails because OD_obj_record_t is undefine
_V4 exporter, compile with -DCO_VERSION_MAJOR=2 : not allowed due to #if CO_VERSION_MAJOR < 4

*** CANopenNode newOD (d224de0)
_V4 exporter, compile with -DCO_VERSION_MAJOR=4 : fails because ODA_STR not declared

*** CANopenNode v1.3 (b92b4bb)
_LEGACY exporter , compile with -DCO_VERSION_MAJOR=1 : fails because CO_OD_PERSIST_COMM is undefined
_V4 exporter, compile with -DCO_VERSION_MAJOR=4	: fails because #include "301/CO_ODinterface.h" (no such file)
```

Would it be possible to mention at least 1 working combination of any version of CANopenNode an  libedssharp ? I stopped manually fuzzing this, it was getting ridiculous...

JuPrgn, 2020-12-24:
I would recommend latest Libedssharp autobuild configured for legacy export and CANopenNode V2 current master or V1.3.

CANopenNode, 2020-12-24:
You were one day to early. Latest libedssharp will work with latest CANopenNode V4 (newOD). CANopenNode V2 will be fixed in next days.

fenugrec, 2020-12-24:
Thanks, in that case I will try again in a few days. Do I understand that 
- V2 is "feature-complete", i.e. comparable to 1.3 ;
- V4 is newOD, but as you mentioned in #245 is still missing some functionality

I understand this stack is constantly evolving especially now with the newOD refactoring, but it would be really, really helpful to summarize in the README / docs which versions work with which libedssharp builds + exporters.

JuPrgn, 2020-12-28:
@fenugrec  As a starting point I will describe what I am currently using. You should change configuration if you plan to use LSS or SYNC.

- CANopenNode (V2) current master :  aada195b
- Libedssharp latest auto build : 0.8-109-gb3f85db 
https://github.com/robincornelius/libedssharp/raw/gh-pages/build/OpenEDSEditor-latest.zip

On CANopenNode I configured the following defines : 

File | Line | Original | Modified
-- | -- | -- | --
301/CO_config.h | 457 | #define CO_CONFIG_PDO_SYNC_ENABLE 0x04 | #define CO_CONFIG_PDO_SYNC_ENABLE 0x00
301/CO_TIME.h | 33 | #define CO_CONFIG_TIME (0) | #define CO_CONFIG_TIME (1)
301/crc16-ccitt.h | 34 | #define CO_CONFIG_CRC16 (0) | #define CO_CONFIG_CRC16 (1)
CANopen.h | 244 | #define CO_NO_LSS_SLAVE 1 | #define CO_NO_LSS_SLAVE 0

I also added this define `CO_VERSION_MAJOR=2` on compiler options (`Define C macro` on XC16 compiler). The default configuration without this define is CANopenNode V1.3.

You will find attached an OD example : 
[OD.zip](https://github.com/CANopenNode/CANopenNode/files/5746562/OD.zip)

And the driver I am using (dsPIC33) if your target driver is incomplete you may find how to add the missing declarations from this one : 
[dsPIC33_Driver.zip](https://github.com/CANopenNode/CANopenNode/files/5746574/dsPIC33_Driver.zip)

Please note on Libedssharp you have to configure : `Tools `/ `Preferences `/ And select `CANOPENNODE_LEGACY` as exporter (for V1.3 and V2 CANopenNode) 

Hope this help !

CANopenNode, 2020-12-28:
see https://github.com/CANopenNode/CANopenNode/issues/247#issuecomment-751703054

fenugrec, 2020-12-30:
@JuPrgn thanks for that clear summary, much appreciated.

JuPrgn, 2020-12-30:
You are welcome ! 
The Libedssharp version I mentionned had a problem with the combobox to set pdo map as optionnal it was automatically returning off. I had to set it manually on the project file. This was not the case on the previous one and should be fixed on the new one that is currently available. Please let me know if you need some more details or if this work for you too.

CANopenNode, 2020-12-31:
Latest libedssharp is fixed now, CANopenNode_Legacy works well and XDD(v1.1) may be used for storing any project. It is not necessary to use old xml format for storage.

`example/_project.xml` should be used as a template for projects based on CANopenNode (<= v2). Libedssharp produces equivalent CO_OD.h/.c files as in example directory. 
`301-profile.xpd` is a template for CANopenNode (>= v4)

Readme.md files are updated. Please use latest versions of libedssharp and CANopenNode (master or newOD branch)

There was some mess since last OD updates. I think, it is fixed now.

fenugrec, 2021-01-02:
Thanks. I think I've found a working combination for the moment. I'll reopen if I run into more issues.

closed 2021-01-02 by fenugrec