CANopenNode/CANopenNode #259: CO_fifo CO_fifo_cpyTok2B64() : comparison is always false (unsigned char < 0)

fenugrec, 2020-12-30:
Two warnings near 
```
../CANopenNode/301/CO_fifo.c:1329:15: warning: comparison is always false due to limited range of data type [-Wtype-limits]
         if (c < 0 || code < 0) {
```

c and code are `char`, which is unsigned on gcc-arm. Not sure of the implications of this, (also not sure why we have base64 code but  that's another subject)

CANopenNode, 2020-12-31:
base64 is used by CiA309, see usage in https://github.com/CANopenNode/CANopenSocket/tree/master/examples/basicDevice

CANopenNode, 2020-12-31:
> char is unsigned on gcc-arm.

Thank you for this notice. I wasn't sure about chars, but I still used them widely as type for byte buffers. Mainly inside gateway stuff. It seems this wasn't a good idea. In gcc-arm that line has a bug.

I'm asking here for opinions:
1. should it stay type char and just fix problematic line(s)
2. replace `char buf[..._BUF_SIZE]` and related with some other type than char in all places:
   a. with int8_t
   b. with uint8_t
   c. with own type, defined inside CO_driver_target.h

Different systems treats different types differently. That's why I like to use own, more precisely defined data types. I'm asking here, which type is most commonly used for binary data buffers with granularity of one byte.

This may also be related to #239

fenugrec, 2020-12-31:
`char` should only be used where it never matters if the underlying type is signed or unsigned. This means avoiding to do math on them as well as comparisons for < 0 like here. I personnaly always use uint8_t to guarantee unsigned 8-bit bytes, but I don't work with "weird" platforms like C2000 that have `CHAR_BIT != 8`

Lely's canopen stack have recently addressed this problem , and it's not trivial : 
https://gitlab.com/lely_industries/lely-core/-/merge_requests/25

in some places using `uint_least8_t` was sufficient but there's also some c++ devilry in other places.

geoffrey-vl, 2020-12-31:
There is a rather interesting topic about uint8_t vs unsigned char on SO, see https://stackoverflow.com/questions/1725855/uint8-t-vs-unsigned-char. The (u)int8_t is a typedef already, so actually it is the kind of abstraction you more or less need. For linux, see https://elixir.bootlin.com/linux/v5.10.4/source/include/linux/types.h.
As pointed out on SO, it won't cover all targets, but at least most "modern" systems.
As @fenugrec I'm also a fan of using (u)int_t types.

CANopenNode, 2021-01-02:
Then I will just systematically and carefully replace chars with uint8_t. This is mostly necessary in newer code, related to gateway, which is the same in v2 and v4.

For now I won't care about platforms "CHAR_BIT != 8". This will be later issue, if.

CANopenNode, 2021-01-05:
done in v2 and v4, commits 689572af and 267a2daf

closed 2021-02-01 by CANopenNode