CANopenNode/CANopenNode #260: SDO_server needs CRC16

fenugrec, 2021-01-02:
Hi, 
I was trying to see if I can eliminate crc16 code but it's apparently needed by the SDO_server code for the block transfers.

Easy to figure out, but I suggest adding some # ifdef magic to ensure CO_CONFIG_CRC16 is defined if SDO_server is enabled.

CANopenNode, 2021-01-02:
I can't do much in current master branch, because SDO server is not configurable.
But in newOD branch CRC16 can be disabled, if SDO block transfer is not used. CRC16 can be used also for storing object dictionary.

fenugrec, 2021-01-02:
Yes - what I meant was , on current master, since SDO_server is mandatory, have something like
```
#ifndef CO_CONFIG_CRC16
#error CRC16 is required for sdo server etc
```
Not a big deal as I said, it's pretty obvious when compiling.

closed 2021-01-02 by fenugrec