CANopenNode/CANopenNode #261: PDO receive and transmit

edesta-be, 2021-01-14:
Hello!
So i have been working really heavily in your stack through the Zephyr OS and I have loved it (so thank you firstly). So far I have been able to get a board up and running sending messages from the MCU and controlling different parts of the system via CANopen. But to further my understanding I started dabbling in PDOs and what they have to offer with their mapping capabilities and just larger data size they can transmit. 
In an effort to confuse myself less, i used the canopen through Python to map and command my node as a "master". I was attempting to do that through your stack on an MCU instead of a Linux PC that I am currently using and I was a little lost by the PDO.c/.h files and their wording. 

For reference here are the steps straight from my python script so I can describe exactly what I am doing:

`network = canopen.Network()

network.connect()

node = network.add_node(NODEID, EDS)

node.nmt.state = 'PRE-OPERATIONAL' # set controller into pre operational
time.sleep(1)

node.tpdo.read()
node.rpdo.read()

# Re-map TPDO[1]
node.tpdo[3].clear()
node.tpdo[3].add_variable(0x6041)
node.tpdo[3].add_variable(0x606C)
node.tpdo[3].trans_type = 1
node.tpdo[3].enabled = True
# Save new PDO configuration to node
node.tpdo.save()

node.rpdo[3].clear()
node.rpdo[3].add_variable(0x6040)
node.rpdo[3].add_variable(0x60FF)
node.rpdo[3].enabled = True
node.rpdo.save()

n = 6000

node.nmt.state = 'OPERATIONAL' # put the controller in operational

node.sdo[0x6060].raw = 3
time.sleep(2)

node.rpdo[3][0x60FF].raw = n 
node.rpdo[3].transmit()
time.sleep(1)

node.rpdo[3][0x6040].raw = 6 # ready to switch on
node.rpdo[3].transmit()
time.sleep(1)

node.rpdo[3][0x6040].raw = 271 # operation enabled state with motion halted
node.rpdo[3].transmit()
time.sleep(1)

node.rpdo[3][0x6040].raw = 15
node.rpdo[3].transmit()
time.sleep(1)

while True:
	node.tpdo[3].wait_for_reception()
	print(node.tpdo[0x6041].raw)
	print(node.tpdo[0x606C].raw)
	node.rpdo[3][0x6040].raw = 7 # setting bit 4 and 5 to 0
	node.rpdo[3][0x60FF].raw = n
	node.rpdo[3].transmit()
	
	node.rpdo[3][0x6040].raw = 31 # ready to switch on
	node.rpdo[3].transmit()
	time.sleep(1)
	
	print("")
	n = int(n*0.8)
	
	if n <= 100:
		break

node.rpdo[3][0x6040].raw = 271 # operation enabled state with motion halted
node.rpdo[3].transmit()

network.disconnect()`

Ignoring the switch of NMT states (I am able to do that) and sending the SDO (i can also do that) I was wondering how to go about calling your functions to achieve this functionality. I know once I get the general order of your particular function calls this will be very easy to implement.

So in short, I initially map the TPDO and RPDO of the node at NODEID (for your reference this current setup has 8 TPDOs and 8 RPDOs). Once mapped/saved during PRE-OPERATIONAL state. I transition to OPERATIONAL state begin update the RPDO values of the node that I had just stored. Once i affect it, I just transmit it and move on to the next message i want to transmit. 

Once i have successfully configured the 0x6040 to the state i need it to be in, I begin sending 60FF messages through that same transmit process. So very very basic (I really just want to understand how to imitate this here).

I look forward to your  response and thank you for your time I appreciate it!

edesta-be, 2021-01-15:
To further add, just looking through the functions that are defined for the system, it seems as though it is handling the RPDO and TPDO of its own OD and not able to access or request to change the PDOs of the slave nodes. Maybe i am reading the docs and the code wrong but it seems that way to me, so those are my more major questions.

Ofcourse a method to do this is to map the RPDO of the Master to the TPDO of the Slave and the TPDO of the Master to the RPDO of the Slave but I want to also understand if there is a way to Map the PDOs on the slave prior to trying to access it. 

Lastly, with the sync messages, there is a SYNC generator boolean that I cannot determine how to set on, is the 1005 value meant to be anything beyond the 0x40000000 value

CANopenNode, 2021-01-18:
Thank you too.
You really experiment quite a lot. CANopen may be hard to start, but once you understand the details, I think, it is easy. I will prepare some more examples, including PDO, in https://github.com/CANopenNode/CANopenSocket. There is something in "basicDevice", but no PDO yet.

First I would like to suggest you, to monitor raw CAN messages:

    candump can0 #or something

Until you have heavy PDO communication, you can easily see, what is going on. You should understand each message:

- bootup or heartbeat message with info about current NMT state
- emergency message (some errors may prohibit NMT operational)
- SDO messages are in pair(s)
- PDO messages works only in NMT operational, contains data only, configured are by communication and mapping parameters,
- Sync, etc.

Maybe I'm explaining things, you already know. I think, we should go step-by-step. And don't think about CANopen as master-slave protocol. More like producer-consumer(s). There are combinations.

These are some hints. Please include some candump, if you have questions, so I could easily see the problem.

edesta-be, 2021-01-18:
Yes! I am glad you re-clarified because it never hurts to have some extra confirmation :) 
I will send a candump from my setup in a few hours and I will explain the respective nodes and the ones you should pay attention to etc. 

But I look forward to your PDO example as it really has me scratching me head at what to do. 
There was also something that I noticed with the COB-ID that you set has the default value instead of the value that it reads in from the EDS file that I thought was interesting because i was trying to force setting the COB-IDs to equate to one another from the perspective of the TPDO matching the RPDO on the producer and vice versa on the consumer. 

Ofcourse naturally, at least my understanding is that we "map" the PDO through a set of SDO commands very similar to how the SDO is intended to work.  So with that I would be interested to understand what that looks like to process the received data etc. 

But I think everything will make more sense once i send over the candump! 

Thank you!

CANopenNode, 2021-01-18:
I will try to explain PDO a little:

It is basically a raw CAN message with CAN-ID and up to 8 bytes of data. We have somewhere documented the meaning of data in specific PDO. So PDO is no more than CAN message, transmitted by one CAN(open) device and received by none, one or many CAN(open) devices.

CANopen gives us the rules, how we can pick some data, which are already inside (large) CANopen object dictionary (mapping parameters) and how to send or receive the PDO (communication parameters).

Pre-defined COB-IDs are often confusing. They are just a recommendation. Usually we use default value for TPDO. And for RPDO we use the COB-ID which matches the TPDO in interest. This is producer-consumers model. In master-slaves model pre-defined COB-IDs are more useful.

Example:

- first device: our application writes data to OD (object dictionary) variable. Configure TPDO and map to that OD variable.
- second device: configure RPDO to match above TPDO. Map data to custom OD variable, named `valueFromFirstDevice`. Application on second device reads that OD variable.

edesta-be, 2021-01-19:
Thank you! That makes a lot more sense in terms of the COB-ID i was overcompensating but in that case that clarifies things a lot more. 

This is the candump form my terminal of running the code above. 701 is the CANnode running your stack, you can ignore it for now as it does nothing exciting in this demonstration. 719 is the node that gets mapped and sends messages one change. So from the code above i changed the PDO transmit type from 1 (which was every SYNC message) to 255 (to transmit on change of value which i figured is easier to track) 

Let me know what you think!
[candump.txt](https://github.com/CANopenNode/CANopenNode/files/5833087/candump.txt)

CANopenNode, 2021-01-19:
It seems everything is OK. 719 sends no emergency message after bootup. In operational TPDOs seems to work.

edesta-be, 2021-01-19:
Yes so this is what the python script at the top sends. I am wondering if this is imitable through your stack via the 701 node mapping the 719 node.

CANopenNode, 2021-01-20:
I don't have much experience with CANopen for Python, but it enables nice interaction with CANopen network, as shown in your example. My stack have so called gateway interface with ascii commands, as shown in readme.md and in basicDevice example. These are two alternatives. To configure PDO for example, Python uses nice commands, without care what is happening in background. With ascii commands in my stack it is necessary to use raw SDO read/write commands.

There is also a third type of tools for CANopen: a GUI, where you can configure CANopen network with drag and drop, etc.

There are a lot of options around CANopen.

One of the advantages of CANopen network is, it can run without a master, once it is configured. Just exchanging PDOs and heartbeats.
Another advantage is, it can be very deterministic and real-time, better than 1 millisecond. Python may be slower at this point, but it depends on usage.

Do you like me to comment your candump.txt a little? If so, can you provide candump with timestamps:

    candump -td -a can0 | tee candump2.txt

edesta-be, 2021-01-21:
Hello!
Apologies for the delay could not carve out enough time to get back to my setup!
I have attached the candump2.txt formed by running the python script and your above command.

To comment further, I completely agree! That is where i want to reach in terms of how I set up the system, once i send a set of SDOs to map a PDO, I just want to receive data from the node as it changes. I am interested to see what I need to do on your end to receive these mapped PDOs and retrieve  their data. But once you have time to generate an example that will help. 

Thank you again, I look forward to your response and let me know your thoughts and if there is something I should consider as I transition this logic to your stack!

[candump2.txt](https://github.com/CANopenNode/CANopenNode/files/5852171/candump2.txt)

CANopenNode, 2021-01-25:
I will try to explain candump2.txt. It shows actual CAN communication between two devices. This is example of master-slave type of communication.

    line 2: (001.999987)  can0  719   [1]  7F                        '.'

Line 2 shows CAN message, occurred 2 seconds after previous CAN message. It is from can0 interface, which is probably USB to CAN interface connected to Linux computer, which behaves like a CANopen master in this example. Another device on CAN bus is microcontroller, running CANopenNode, I guess. It could be any CANopen device.

However, this second device has Node-Id = 0x19. Message above is Heartbeat message from this second device. CANopen heartbeat message is CAN message, with 11-bit identifier (0x700 + node-id) and has one data byte. For default 11-bit CAN identifiers used by CAN open see standard CiA301 (7.3.3 Generic pre-defined connection set) or [CO_Default_CAN_ID_t](https://canopennode.github.io/CANopenSocket/group__CO__driver.html#ga01dd35ae53fd2209ceccabdc8bf8dd06) from CANopenNode documentation.

Standard CAN messages all have 11-bit identifier and 0 to 8 data bytes. CAN message with lower identifier has higher priority on the bus. There are other types of CAN messages, not used here: 29-bit extended identifier; CAN FD, newer with more data and higher data rates, etc.

Heartbeat message contains one data byte, 0x7F in our case. It shows internal NMT state of the device, NMT pre-operational in our case. See NMT states in standard or [CO_NMT_internalState_t](https://canopennode.github.io/CANopenSocket/group__CO__NMT__Heartbeat.html#ga1e8c2a6c0fd4a33183503d25a7c6d744)

    line 6: (000.248715)  can0  000   [2]  82 19                     '..'
    line 7: (000.007631)  can0  719   [1]  00                        '.'
    line 8: (001.999969)  can0  719   [1]  7F                        '.'

Line 6 is NMT command from master: CAN-id = 0 - highest priority message. [CO_NMT_command_t](https://canopennode.github.io/CANopenSocket/group__CO__NMT__Heartbeat.html#gac396242e2e12ef6b0b22ff48636bc4eb) 0x82 (reset communication) is sent to node-id=0x19. Node 19 responded with boot-up message and after two seconds with heartbeat. CANopenNode gateway (CiA309-3) produces the same NMT master message with command `[1] 0x19 reset communication`.

    line 21: (000.481348)  can0  619   [8]  40 00 18 01 00 00 00 00   '@.......'
    line 22: (000.002445)  can0  599   [8]  43 00 18 01 99 01 00 00   'C.......'

This is SDO. Master (SDO client) sends read or write request for variable form Object Dictionary (CAN-ID = 0x600 + remote-node-id). SDO server responses (CAN-ID = 0x580 + node-id). SDO messages are harder to read, usually it is not necessary to know the details. For description see CiA301 or [CO_SDO_state_t](https://canopennode.github.io/CANopenSocket/group__CO__SDOserver.html#ga0b0e614dadcc1c005185b8bc9a7fec11).

Line 21 is CO_SDO_ST_UPLOAD_INITIATE_REQ, index=0x1800, subindex=0x01
Line 22 is CO_SDO_ST_UPLOAD_INITIATE_RSP, same index and subindex, data=0x00000199

Equivalent CANopenNode gateway command: `[1] 0x19 read 0x1800 1 x32`.

    line 25: (000.000710)  can0  619   [8]  40 00 18 03 00 00 00 00   '@.......'
    line 26: (000.002455)  can0  599   [8]  80 00 18 03 11 00 09 06   '........'

Line 25 is CO_SDO_ST_UPLOAD_INITIATE_REQ, index=0x1800, subindex=0x03
Line 26 is CO_SDO_ST_ABORT, same index and subindex, [CO_SDO_abortCode_t](https://canopennode.github.io/CANopenSocket/group__CO__SDOserver.html#ga7587ddcf798747fe6d97d03bf1bf2979) = 0x06090011 #Sub index does not exist (this is strange, TPDO Inhibit time, Event timer and SYNC start value are missing)

This pattern then repeats and reads all TPDO and RPDO communication and mapping parameters. As in Python program.

    line 183: (000.001001)  can0  619   [8]  23 00 18 01 99 01 00 80   '#.......'
    line 184: (000.002439)  can0  599   [8]  60 00 18 01 00 00 00 00   '`.......'

Line 183 is CO_SDO_ST_DOWNLOAD_INITIATE_REQ, index=0x1800, subindex=0x01, data=0x80000199
Line 184 is CO_SDO_ST_DOWNLOAD_INITIATE_RSP, same index and subindex, OK

Equivalent CANopenNode gateway command: `[1] 0x19 write 0x1800 1 x32 0x80000199` - disable PDO.
Then follows several similar commands configuring TPDOs and RPDOs. 

After that NMT command sets node 0x19 to operational and then appears four different PDO messages on the bus.

Python is certainly most clear. More raw approach may also be with several CANopen gateway commands. Observing candump may be useful for understanding the details or for debugging.

I didn't decode all SDO commands and the rest of the program, but I hope I explained a little. It is important to know, where does TPDOs came from and what are the contents. See also some explanation of [PDO re-mapping procedure](http://can-newsletter.org/uploads/media/raw/e522172be523845be799af1e5d6dd1f7.pdf).

edesta-be, 2021-02-11:
Hello!
I wanted to check in on this issue, and see what the status was on potentially an example of this system transmitting and receiving pdos from another node!

Let me know how I can help as well would be happy to because this would be very useful! 

Thank you!!

CANopenNode, 2021-05-17:
PDO tutorial is here now:
https://github.com/CANopenNode/CANopenDemo

CANopenNode, 2022-04-20:
His example is for [CANopen for Python](https://github.com/christiansandberg/canopen). This is another project.

closed 2023-03-28 by CANopenNode