CANopenNode/CANopenNode #262: SDOclientUpload

edesta-be, 2021-01-31:
Hello again! :) 

As a side bar, I was taking a look at SDOmaster.c/.h and I have previously used the SDOclientDownloadInitiate and SDOclientDownload functions to send messages to other nodes on the system. (These are one off messages and always proved to be successful)

My question comes from the Upload side of the same set of functions:
```
int32_t actual_velocity;
uint32_t size_for_velocity;
uint32_t abort_code;
CO_SDOclientUploadInitiate(CO->SDOclient[0], 0x606C, 0,(uint8_t *) &actual_velocity, sizeof(actual_velocity), 0);
CO_SDOclientUpload(CO->SDOclient[0], (uint16_t)elapsed, timeout, &size_for_velocity, &abort_code);
```
I call the second function cyclically while waiting for it to return 0. However, i do not get the data i expect. Perhaps I am missing something when I am commanding these two functions. The first call always returns 0 prior to entering the loop call for the second function. 

The second function eventually breaks out after it equates to 0, however the data returned is not the expected value. When i have previously used the download functions, to write to another nodes OD. The data value of actual_velocity is always equal to 1 no matter the actual value that i am expecting (it also does not seem to go through the loop for the second function as it immediately returns a 0). However, if i am calling it from startup without calling the download client, the system then decides to equate the actual_velocity value to a 0. 

I might just be missing something very basic. 

Thank you in advance!

CANopenNode, 2021-02-01:
SDOmaster.c/.h is part of the old version v1.3. In new version it has been renamed to SDOclient.c/.h and is actually rewritten.

SDO client is quite advanced type of communicatio, if communication is possible via PDOs, they are much easier to use and much more efficient.

You can find example for usage of the old "SDO master" here:
https://github.com/CANopenNode/CANopenSocket/blob/7b8b45fcd2e2d6fa5a9a55d8f3aa8b54ce81a98b/canopend/src/CO_master.c

edesta-be, 2021-02-02:
Perfect thank you! I will take a look at this!

edesta-be, 2021-02-08:
Hello!
So i used the master that you showed. 
When I run it via Python 
the candump terminal shows the request and I get the right response back
```
  can0  61A   [8]  40 61 60 00 00 00 00 00
  can0  59A   [8]  4F 61 60 00 03 00 00 00
```

But when i run it using the master.c file
```
can0  081   [8]  00 82 10 1A 48 76 A6 09
can0  081   [8]  10 81 10 14 1A 06 00 00
```

This is clearly wrong but I am not sure what I missing. When i print out the values that I pass in, they do not update with the expected response.

CANopenNode, 2021-02-08:
These are just two emergency messages, CO_EM_PDO_WRONG_MAPPING and CO_EM_CAN_TX_OVERFLOW

SDO client is hard, I don't have information.

edesta-be, 2021-02-08:
OKay thank you!  I'll focus back in on trying to get this to work with PDOs then!

closed 2021-02-08 by edesta-be