CANopenNode/CANopenNode #264: Support question, possible to implement master on embedded system, i.e. not on Linux

wolfc01, 2021-02-03:
Hi,

I have a question, what are the steps to perform to run the CANopen master on an embedded CPU (STM32 in my case). Is that easily feasible: has somebody realized that before?

We have 4 STM32 based processor boards of which one should act as the CANopen master node. We have requirements on having LSS, futher some low speed request responses between the master and slave nodes (all initiated by the master), without any realtime or throughput requirements.

Br
Carl.

robincornelius, 2021-02-03:
> by the master), without any realtime or throughput requirements. Further
> the required messages sized are small (<= 8 bytes) .

Hi,

I don't use the LSS features but i'm always using one of my embedded nodes
as a master, when used with my own nodes and other 3rd party can nodes.
Think i've been up to about 30 nodes on the biggest systems. In the
simplest terms to get it to work every time the process i always follow is
:-

1. Set all slave nodes to start in pre-operational mode, and save this to
their eeprom or how ever you set them up
2. Switch on the system. in the master node you may need a delay here to
allow time for every slave to boot
3. From the master send a NMT_CommunicationsReset() (you may need to be
careful you don't catch this signal on the master as well and reset
yourself!)
4. Have a small delay to allow all nodes to reset.
5. Send a NMT_Start() to all nodes

This procedure will ensure all PDOs are only sent when the NMT_Start is
issued so everything is synchronized at start up. The above is very simple
to be implemented on embedded system, just ensure the master bit on object
0x1F80 is set in the object dictionary and if using the EDS editor this
will ensure the NMT_MASTER functions are available to send the
communications reset/start bus etc,. Failing to do the CommunicationsReset
will result in possibly unpleasant error messages related to Bus Off / TX
Passive so if you also monitor emergency messages, this is a signal to
clear everything after the typically unclean cold boot where there is
always one node who sees a passive bus as they won the power on race
first.

There are a number of other choices for the startup where you may wish to
ping each node via SDO to ensure its alive before issuing the NMT_Start()
or you may wish to start certain nodes before others, but if you do this be
very careful of who is sending and listening to PDOs as you could miss the
initial messages which may or may not affect your system.

LSS is just an earlier stage in the boot sequence, but i've never even
tried to use that bit so can offer no advice on that.

Robin

wolfc01, 2021-02-04:
Hi Robin,

thanks for your valuable input,

Br
Carl

CANopenNode, 2021-02-08:
There is 309/CO_gateway_ascii, which should work on microcontroller with ascii interface. This could be used for rich service access with LSS, NMT, SDO, etc., as described in gettingStarted.md or LSSusage.md.

Othervise modules can be used separately LSS master, NMT master (quite simple) or SDO master (old example for blocking usage is here: https://github.com/CANopenNode/CANopenSocket/blob/7b8b45fcd2e2d6fa5a9a55d8f3aa8b54ce81a98b/canopend/src/CO_master.c)

closed 2021-03-18 by CANopenNode