CANopenNode/CANopenNode #267: Outdated Getting Started?

alde103, 2021-02-22:
Hi,

I'm following the [getting started](https://github.com/CANopenNode/CANopenNode/blob/master/doc/gettingStarted.md) guide in my rpi0w; I have installed `can-utils` and it seems that the building process was successful, but when executing the following instruction 
`./canopend vcan0 -i 4 -s od4_storage -a od4_storage_auto`
it prompts the following:
```
./canopend: invalid option -- 'a'
Usage: ./canopend <CAN device name> [options]

Options:
  -i <Node ID>        CANopen Node-id (1..127) or 0xFF (LSS unconfigured).
  -r                  Enable reboot on CANopen NMT reset_node command. 
  -s <storage path>   Path and filename prefix for data storage files.
                      By default files are stored in current dictionary.
  -c <interface>      Enable command interface for master functionality.
                      One of three types of interfaces can be specified as:
                   1. "stdio" - Standard IO of a program (terminal).
                   2. "local-<file path>" - Local socket interface on file
                      path, for example "local-/tmp/CO_command_socket".
                   3. "tcp-<port>" - Tcp socket interface on specified 
                      port, for example "tcp-60000".
                      Note that this option may affect security of the CAN.
  -T <timeout_time>   If -c is specified as local or tcp socket, then this
                      parameter specifies socket timeout time in milliseconds.
                      Default is 0 - no timeout on established connection.

See also: https://github.com/CANopenNode/CANopenNode
```
I don't know if I missed to add something before building the package.

By removing `-a od4_storage_auto` from the instruction, the command seems to work, it just doesn't do the `heartbeat` function.

```
  vcan0  704   [1]  00
  vcan0  084   [8]  00 50 01 2F 04 00 00 00
```
Help would be highly appreciated. Thank you very much.

CANopenNode, 2021-02-25:
Please note, there are two branches: master and newOD (use `git checkout master` or `git checkout newOD`, latter is also part of CANopenSocket)
You are probably compiling with newOD branch and reading getting started from master branch or online, which is not fully up to date. You should read `CANopenNode/doc/gettingStarted.md` directly.

in newOD there is different usage of storage and heartbeat producer is disabled by default.

There is currently some mess, newOD will become master soon.

alde103, 2021-02-25:
Yes you are right, at the time I was working with CANOpenSocket, thank you very much for your support.

closed 2021-02-25 by alde103