CANopenNode/CANopenNode #268: Locking of CAN in CO_SDOserver.c results error in ARM based systems , needs refactoring

MuhammadTharwat, 2021-02-24:


geoffrey-vl, 2021-02-24:
I believe more context is required before anything can be discussed or solved if there would be any need.

What bug are you seeing? Do you have debug output? Can you share your code? What is the expected behavior? etc...

MuhammadTharwat, 2021-02-24:
@geoffrey-vl  
It is not design issue , its implementation issue. 
On ARM controllers atomic operations are defined as follows : 
#define CO_LOCK_OD()   { 							                  \
		uint32_t PrevPrimask= __get_PRIMASK();   \
		__disable_irq();

#define CO_UNLOCK_OD()     __set_PRIMASK(PrevPrimask);              \
		}

In that case , if u try compile the stack as is , it would boom with 
Error: identifier "PrevPrimask" is undefined. 

That was hidden because the examples provided are based on PIC controllers , in which the locking is done via single statement. 
Also using "Unlock" multiple times in the same API Vs single Lock is not good approach even if this usage is precise. 
So I would just reorder the lock & unlock in two APIs : 
-CO_SDO_readOD
-CO_SDO_writeOD
So that there is single Lock Vs single Unlock.

geoffrey-vl, 2021-02-24:
What branch are you using?

Because for V1.3 branch, the CO_UNLOCK_OD is defined within one of your target drivers, and you may have to add your own driver depending on which hardware you use.
For master (V2), drivers where removed from the repo since basically maintaining them was a burden. Only generic linux (socketcan) en example driver are supplied. You need to bring your own driver.

See https://github.com/CANopenNode/CANopenNode/blob/master/doc/deviceSupport.md

MuhammadTharwat, 2021-02-24:
@geoffrey-vl 
I'm using the latest on the "master" branch SHA-1: acd719e27a1355470c90662c78c235b153a53dce
That's after the V2 release (depending on my final target time I may go back to V2 release or to any newer releases on the horizone) 
Indeed the LOCK & UNLOCK are tareget dependent and I'm using my own HW-specific code. 
But again, these 2 APIs need to reorder (refactor) the Lock & Unlock calls or else any ARM controller (or even OS integration) would impose the refactor locally. 
If you look at the third-party demo on ST referred to from ur page, you would find that the stack is customized !
The examples on PIC are the ideal ones as they do not have any customizations.

geoffrey-vl, 2021-02-24:
Can you share your platform and RTOS please?
For example, my driver for STM32 and ARM Cortex-M4 and mbed-os locks as following:

header:
```
#define CO_UNLOCK_OD()              co_unlock_od();
```
source:
```
void co_lock_emcy()
{
    co_emcy_mutux.lock();
}
```
So in my case it uses a mbed-os PlatformMutex object for locking and unlocking.
You need your own implementation header and source files for platform agnostic code such as locking.
When you get that right, CANopenNode will not require any changes.

`Error: identifier "PrevPrimask" is undefined.`
It seems to me that you have not included the header defines PrevPrimask. Or you are using a driver that is not compatible with your OS.

When compiling, make sure that the build tool doesn't compile the example directories. Some build tools allow you to select your sources, other allow you to blacklist files or folder. You need to figure that out.

Also the documentation I'm referring to mostly has examples that are out of date, and only work for the V1.3 release. I'm not aware of updated drivers out there, at least the docs were never updated to link no newer released driver. I guess this part hasn't particularly improved either when they were removed from the stack.

MuhammadTharwat, 2021-02-24:
@geoffrey-vl 
I'm using the next definitions for the bare-metal application
#define CO_LOCK_OD()   { 							                  \
		uint32_t PrevPrimask= __get_PRIMASK();   \
		__disable_irq();

#define CO_UNLOCK_OD()     __set_PRIMASK(PrevPrimask);              \
		}
	
Here as u know PrevPrimask is just a local variable inside the block made by Lock & Unlock to store the status register (in case interrupt was already disabled) 
So it's a variable not a macro ! 
I have done the refactor and it is working fine, but if u try define the macros as my definition (which is the common one for bare-metal and some RTOS like FreeRTOS) the compilation would boom !

martinwag, 2021-02-24:
Isn't that stuff regarding saving and restoring IRQ context done by the compiler/toolchain on ARM? Otherwise maybe change this
```
#define CO_LOCK_OD() {
uint32_t PrevPrimask= __get_PRIMASK();
__disable_irq();

#define CO_UNLOCK_OD() __set_PRIMASK(PrevPrimask);
}
```
to

c file
`uint32_t PrevPrimask;`

header file
```
extern uint32_t PrevPrimask;

#define CO_LOCK_OD() \
{\
PrevPrimask= __get_PRIMASK(); \
__disable_irq(); \
}

#define CO_UNLOCK_OD() \
{\
__set_PRIMASK(PrevPrimask);\
}
```

...and as for freertos, just use the RTOS:
```
extern SemaphoreHandle_t CO_EMCY_mtx;
/** Lock critical section in CO_errorReport() or CO_errorReset() */
static inline void CO_LOCK_EMCY(void) { (void)xSemaphoreTake(CO_EMCY_mtx, portMAX_DELAY); }
/** Unlock critical section in CO_errorReport() or CO_errorReset() */
static inline void CO_UNLOCK_EMCY(void) { (void)xSemaphoreGive(CO_EMCY_mtx); }
```

MuhammadTharwat, 2021-02-24:
> Isn't that stuff regarding saving and restoring IRQ context done by the compiler/toolchain on ARM? Otherwise maybe change 

No, it needs to be done explicitly. 
I could have easily declared the global mask as extern, but this is not neat approach as the hassle of including global extern in .c and then using it in a .h is not really necessary.

martinwag, 2021-02-24:
Had a short look at this (so I don't get too rusty... :-) ). You only need to evaluate PRIMASK if you want to do nested irq calls [source](https://stm32f4-discovery.net/2015/06/how-to-properly-enabledisable-interrupts-in-arm-cortex-m/) This is not the case for V1.3 stack, but @CANopenNode what about current versions?

You need to place variable definition in C file, as otherwise the compiler might place **multiple indipendent** of them in the binary. This will give you a big mess. This is callled _one definition rule_.

MuhammadTharwat, 2021-02-24:
Well @martinwag  , Let's forget about the definition of this dependency for a moment ! 
Do u see that using single Lock and multiple Unlock is a good approach ? 
Especially when u can see the library does not have a protection against incosistent number of calls ! 
What I propose is to reorder the locking in the two mentioned APIs so that there is a single Lock & single Unlock .
Using externs should be "Silver Bullet" and even if this is a temporary work-around, the library still needs that change.

martinwag, 2021-02-24:
OK, now I understand what you want to do :-).

Essentially, best would be something that ensures unlocking, like C++ std::lock_guard (unlocks when going out of scope).

CANopenNode, 2021-02-25:
This is related comment from two weeks ago: https://github.com/CANopenNode/CANopenNode/issues/162#issuecomment-777438875

Passing some argument (pointer to CO_CANmodule) to all the CO_LOCK_...() and CO_UNLOCK_...() should be a good solution. What do you think?

I will certainly use it with newOD branch. But don't want to change v2, if not really necessary.

MuhammadTharwat, 2021-02-25:
@CANopenNode  
The stack currently wont compile for baremetal application on ARM controller without the work around of extern ! 
The change would be really minor ! , nothing in design , just implementation !

CANopenNode, 2021-03-02:
> So I would just reorder the lock & unlock in two APIs :
> -CO_SDO_readOD
> -CO_SDO_writeOD
> So that there is single Lock Vs single Unlock.

This is very old part of code, the same in current master (v2.0) and v1.3. I don't like to change that.
But in newOD branch that part is completely renewed. Also there are only single lock/unlock calls, no more "magic" like in CO_SDO_readOD.

closed 2021-03-18 by CANopenNode