CANopenNode/CANopenNode #269: latest release branch

vvkaggarwal12, 2021-03-02:
Hi,

I am quite new to CANOpenNode. I would like to use canopennode in my project. 
I would like to know which latest release i would use for my project as i am not able to understand which is the latest stable release version.
And also i was trying to use v4.0 in my project, there are few changes made in CO_SDOserver.h file because of which it is not compiling for example CO_SDO_t structure definition had been removed from the .h file.

geoffrey-vl, 2021-03-02:
See https://github.com/CANopenNode/CANopenNode/issues/245 and https://github.com/CANopenNode/CANopenNode/issues/162

vvkaggarwal12, 2021-03-02:
Hello,
Is v2.0 uses new OD or Old OD?

CANopenNode, 2021-03-02:
v2.0 (current master branch) uses old OD.
v4.0 (newOD branch) should be very stable and only PDO is still old for now. CO_SDO_t dependencies has been replaced by  functions from CO_ODinterface. There may be some minor differences in drivers between v2.0 and v4.0.

closed 2021-03-18 by CANopenNode